moveit/moveit #417: Environment changed during execution - Moveit Commander Pick

pet1330, 2017-01-16:
### Description

Hi, I'm trying to use the [movit_commander](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_commander/src/moveit_commander/move_group.py#L480-L485) to pick up an object using a Baxter robot, however, I'm having an issue making it work. I have managed to get the robot to move to the correct location for the `pre_grasp_approach` by supplying a Grasp position directly above the object, however, after it has moved to this location, and moved down to pick up the object, I get the following error:

`ABORTED: Solution found but the environment changed during execution and the path was aborted`

This Issue seems to be due to the object that is attached during the pick operation is colliding with its unattached version, as discussed in #323  and mentioned that the `planning_scene_diff.is_diff` should be set to false, however, I'm unsure how to achieve this.

What is also odd about this Issue is that if I try to move the robot using the RVIZ plugin, there is no issue, but if I try to run the pick operation again, the planner fails due to the above collision constraints and I am required to first detach the object from the robot, before I can run it again.

[Screen Recording of the Issue](https://youtu.be/_FmgNESuGdk)

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Latest available binary build

### Steps to reproduce

* [Set the Scene](https://github.com/pet1330/FInCoR/blob/pickandplace/locomotion/src/set_scene.py)  => [Scene configuration](https://github.com/pet1330/FInCoR/blob/pickandplace/locomotion/config/object_list.yaml)
* [Pick the Object](https://github.com/pet1330/FInCoR/blob/pickandplace/locomotion/src/pick_example.py)

### Expected behaviour
* I would expect the Baxter robot to pick up the item by 
* First moving to the `pre_grasp_approach` pose
* then Moving down to the object and close the grippers
* then move to the `post_grasp_retreat` pose

### Actual behaviour

* The Baxter robot moves to the `pre_grasp_approach` pose
* then moves down towards the object's position
* then attaches the object to the robot before failing with the error: `ABORTED: Solution found but the environment changed during execution and the path was aborted`

### Backtrace or Console output
[Please see video for console output.](https://youtu.be/_FmgNESuGdk)

v4hn, 2017-01-16:
The linked issue probably has nothing to do with this.
Why don't you visualize the monitored planning scene of your move_group node instead of /planning_scene?
This is missing octomap information and information on objects within the move_group, so the rviz video is a lot less helpful...

The bug here is that execution of the liftup-motion at the end of the pickup is aborted because an unforeseen collision is predicted for this trajectory. The question is: which and why.

The second failure is no bug: if you grabbed an object (only liftup failed, grabbing succeeded), you can't grab a second one without removing the first one from the gripper again.

Does the collision checking publish any collision markers for the invalidated plan of the first execution?

pet1330, 2017-01-16:
@v4hn, thank you for the quick reply! 

Sorry! That was my mistake, I was meant to [link to issue #323 on the moveit_ros repository](https://github.com/ros-planning/moveit_ros/issues/323). (I've updated the issue above to reflect this)

The visualisation in RVIZ is `/move_group/monitored_planning_scene`, is that correct?

[image]

How do I display the octomap information and information on the objects? As far as I'm aware, no collision markers are published. I checked `/move_group/display_contacts` topic, but nothing is being published on it, I also checked the collision status of Baxter on topics under `robot/limb/right/` but to no avail.

I understand why planner is failing when there is an attached object, but I found it weird that this does not affect the RVIZ planner as well. Should the attach object not also prevent RVIZ from moving the arm?

Below is the planner output, which states that a contact was found between `tag_1` that is attached and in the scene. But, I'm not sure how to fix this...

```
[ INFO] [1484599462.597802759]: Solution found in 0.033842 seconds
[ INFO] [1484599462.617272412]: SimpleSetup: Path simplification took 0.019406 seconds and changed from 26 to 17 states
[ INFO] [1484599462.622662784]: Found successful manipulation plan!
[ INFO] [1484599462.622741107]: Pickup planning completed after 0.198079 seconds
[ INFO] [1484599469.357821105]: Trajectory component 'retreat' is invalid
[ INFO] [1484599469.357900454]: Found a contact between 'tag_1' (type 'Object') and 'tag_1' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [1484599469.357917702]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1484599469.363083419]: Stopping execution because the path to execute became invalid (probably the environment changed)
[ INFO] [1484599469.363116333]: MoveitSimpleControllerManager: Cancelling execution for /robot/limb/right
[ INFO] [1484599469.363148106]: Stopped trajectory execution.
[ INFO] [1484599469.623183408]: Waiting for a 2.000000 seconds before attempting a new plan ...
[ INFO] [1484599471.623302731]: Done waiting
```

pet1330, 2017-01-30:
@marc-hanheide is also interested in this

cangjiaxaun, 2017-04-04:
Hi peter @pet1330 have you solved this issue? I meet similar problem as you were.

cangjiaxaun, 2017-04-06:
@pet1330 I find where the is_diff parameter was set, it's at the move_group_interface.cpp file 
[](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L727)

As the pr2 robot can perfectly perform the pick and place mission, I think this error may not caused by the source code but some setting in our moveit! config. However I tried another model but meet same problem.

pet1330, 2017-05-17:
@v4hn Any update on this issue?

hjzh4, 2018-03-09:
Hey guys, @pet1330 @cangjiaxaun , have you solved this issue. I meet similar problem.

pet1330, 2018-03-09:
Nope, I could not solve this, so ended up not using MoveIt.

hjzh4, 2018-03-09:
Hi Peter,

What did u use after Moveit? I also want to use Baxter to do a pick and
place task.

Thanks!

rhaschke, 2018-03-09:
Hi guys,

we are currently developing a more flexible task planning pipeline: 
https://github.com/ros-planning/moveit_task_constructor
There are also some examples available for pick-and-place with various robots (PR2, Baxter, Pepper):
https://github.com/ubi-agni/mtc_demos

Please be aware, that the API is still under heavy development and there is no documentationn available yet (except some source-code comments). However, this might be an interesting alternative to you, as it allows to plan rather complex _action sequences_ in a coherent fashion.

v4hn, 2018-03-10:
If @rhaschke makes advertisement for MTC here, I'll add to it. :)
The examples he give only plan and don't execute yet, kind of contrary to the problem discussed in this issue.
For a first example on execution, you can have a look [here](https://github.com/TAMS-Group/mtc_pour/blob/master/src/mtc_pour_tams_ur5_demo.cpp) too.

MTC is meant to replace the Pick Place capability you (try to) use.
Feedback and ideas are very welcome!

felixvd, 2018-07-23:
I am reproducing the new pick and place tutorial with another robot and am getting a similar error. 

```
[ INFO] [1532350772.240666561]: Planning attempt 1 of at most 1
[ INFO] [1532350772.264647874]: Added plan for pipeline 'pick'. Queue is now of size 1
[ WARN] [1532350772.288972916]: No joint states of grasp postures have been defined in the pick place action.
[ WARN] [1532350772.289970969]: Cannot find planning configuration for group 'a_bot' using planner 'RRTConnectkConfigDefault'. Will use defaults instead.
[ INFO] [1532350772.290845382]: Planner configuration 'a_bot' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1532350772.291515134]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1532350772.311104549]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1532350772.311141696]: Solution found in 0.019956 seconds
[ INFO] [1532350772.350087444]: SimpleSetup: Path simplification took 0.038874 seconds and changed from 3 to 2 states
[ INFO] [1532350772.354791536]: Found successful manipulation plan!
[ INFO] [1532350772.354873577]: Pickup planning completed after 0.114033 seconds
[ INFO] [1532350772.439739442]: Fake execution of trajectory
[ INFO] [1532350775.742091952]: Fake execution of trajectory
[ INFO] [1532350776.840039110]: Trajectory component 'retreat' is invalid
**[ INFO] [1532350776.840133610]: Found a contact between 'object' (type 'Object') and 'object' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.**
[ INFO] [1532350776.840153093]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1532350776.840231149]: Fake execution of trajectory
[ INFO] [1532350776.849711496]: Stopping execution because the path to execute became invalid(probably the environment changed)
[ INFO] [1532350776.939943813]: Fake trajectory execution cancelled
[ INFO] [1532350776.939994697]: Stopped trajectory execution.
[ INFO] [1532350777.239455010]: Completed trajectory execution with status PREEMPTED ...
[ INFO] [1532350777.355502973]: Waiting for a 2.000000 seconds before attempting a new plan ...
[ INFO] [1532350779.355674417]: Done waiting
```

[image]

The system looks very similar, and the code is almost identical to the tutorial. I only changed the frames and move group name, essentially. The error seems like there is an issue in the pipeline. Maybe @Ridhwanluthra who just wrote the tutorial has some insights.

v4hn, 2018-07-24:
Hey Felix,

your problem seems independent of the one in this issue.
It seems like the pipeline fails to correctly remove the object from the collision world when it attached it to the gripper. As a result the object collides with the attached version of itself, I guess.
I have no idea how you triggered this. I don't see it in everyday usage of the pipeline.

felixvd, 2018-07-24:
As a quick update: I put 10 Grasp messages instead of 1 into the vector that's sent to the pick function, so that the planning would be reattempted more often. Each attempt seems to succeed through all stages about 5% of the time, and fails in the "reachable & valid pose filter" stage about 70% of the time.

I was not able to reproduce the error about the contact between the two "object"s today, even when publishing the same object id twice. I was pretty sure that that was not the original problem anyway, and from what I understand only one object of each id exists in the planning scene, and the newer ADD replaces the old object.

I will look into it some more, but I am already surprised by the low rate of success for the grasp approaches. The grasp pose is not even close to the object.

[image]

Primula-a, 2020-06-18:
> Hi guys,
>
> we are currently developing a more flexible task planning pipeline:
> https://github.com/ros-planning/moveit_task_constructor
> There are also some examples available for pick-and-place with [...]

Hi, I'm new to moveit and ros. In particular, I'm working with Pepper (Ros Kinetic). This pack seems very interesting for me, but I need a little tutorial about installation and use. Can you help me?

felixvd, 2020-06-19:
Yes, look here: https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html

As an update to my post further up, I couldn't reproduce the issue reliably and we didn't run into the problem in our usage either. I think this thread could be closed.

closed 2020-06-19 by v4hn