moveit/moveit #424: move_group segfaults in moveit::core::RobotModel::constructShape

vmatare, 2017-01-24:
I'm on the kinetic-devel branch (`#f63872959`), compiling against `geometric_shapes 0.5.2` and `Eigen 3.3.1`. For me, move_group just segfaults on startup:
<script src="https://gist.github.com/vmatare/c6cb12f837a6863bcffd3800e2e20319.js"></script>
Valgrind has the following comment on the matter:
<script src="https://gist.github.com/vmatare/591412690a9d8f9315d2c578747dd2a8.js"></script>

v4hn, 2017-01-24:
Hey @vmatare , could you/are you allowed to share the mesh that triggers the segfault?

vmatare, 2017-01-24:
Sure, it's the publicly available mesh for the Aldebaran Pepper robot from [https://github.com/ros-naoqi/pepper_meshes]. However their `CMakeLists.txt` invokes some weird binary installer that somehow downloads and/or extracts (wtf?) the STL files. In my case, `constructShape()` is being called on `Torso_0.10.stl` from these files. I'd attach it here, but that's not allowed either. I'll try to put them on dropbox...

vmatare, 2017-01-25:
Alright, I did some further digging into this, and it seems that somehow it manages to use two different values for the `EIGEN_DEFAULT_ALIGN_BYTES` macro in a single binary. I put some debug output in `eigen3/Eigen/src/Core/Memory.h`, and behold:
```
[ INFO] [1485357250.325838043]: Loading robot model 'JulietteY20MP'...
EIGEN handmade_aligned_malloc 32 0: 0x1c2f5a0
EIGEN handmade_aligned_free 20 0: 0x1c2f5a0
Segmentation fault (core dumped)
```
So the `malloc` is done with an alignment to 32 bytes, while the `free` is done with 20-byte alignment. I'm not even sure any more whether this is actually caused by the moveit build :-/

de-vri-es, 2017-01-25:
Interesting. A stack trace at both locations would also be useful. It should tell us which libraries use conflicting values for `EIGEN_DEFAULT_ALIGN_BYTES`. Then we can see if there is anything we can do about it.

vmatare, 2017-01-25:
Yeah, that's pretty much what I'm doing right now. I think you guys can mostly disregard this issue since I think the reason for the whole mess is inconsistent `-march=` compiler flags. I installed `geometric_shapes` via distro packages with `-march=broadwell`, while the whole moveit build was done without any `-march` flag. Will report back tomorrow, but right now it looks like a case of PEBKAC^^

vmatare, 2017-01-26:
Okay, this segfault was indeed caused by inconsistent `-march=` flags which broke Eigen memory alignment. So I'm closing this since the issue wasn't really moveit related at all. Sorry about the noise ;-)

closed 2017-01-26 by vmatare

de-vri-es, 2017-01-26:
Thank for looking in to it :)