moveit/moveit #425: Creating moveit::planning_interface::MoveGroup crashes when running in nodelet

gaspers, 2017-01-26:
### Description

After updateing to the current ubuntu packages, we discovered a problem with creating an object instance of moveit::planning_interface::MoveGroup in a nodelet. This seems to be an issue with the changes in
moveit_ros/planning_interface/move_group_interface/src/move_group.cpp
in
template <typename T> void waitForAction(..)
and
void waitForExecuteActionOrService(..),
where ros::spinOnce() was exchanged with a call to the callbackQueue (https://github.com/ros-planning/moveit_ros/pull/701). This does not work for nodelet-NodeHandles.

In detail:
In commit
https://github.com/ros-planning/moveit_ros/commit/9479afcfd6bd05b3580d4bfb87cb773f29a905d4
in lines like
( ( ros::CallbackQueue * ) node_handle_.getCallbackQueue())->callAvailable();
the NodeHandle 
node_handle_.getCallbackQueue()
returns CallbackQueueInterface * and not ros::CallbackQueue *

The cast seems to break the code.

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Source or Binary build? Both
* If binary, which release version? 0.7.6-0trusty-20161230-092613-0800 (bin)
* If source, which git commit or tag? 0.7.6

### Steps to reproduce
Create an object instance of moveit::planning_interface::MoveGroup in a _nodelet_:

using namespace moveit::planning_interface;
MoveGroup::Options options(group_name_);
options.node_handle_ = nh_;
boost::make_shared<MoveGroup>(options, tf_listener_, ros::Duration(0.3));

### Expected behaviour
Object instance is created and works properly.

### Actual behaviour
Segmentation fault occurs.

### Backtrace or Console output
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc506f700 (LWP 20788)]
0x00007ffff74b49f4 in std::_Rb_tree<unsigned long, std::pair<unsigned long const, boost::shared_ptr<ros::CallbackQueue::IDInfo> >, std::_Select1st<std::pair<unsigned long const, boost::shared_ptr<ros::CallbackQueue::IDInfo> > >, std::less<unsigned long>, std::allocator<std::pair<unsigned long const, boost::shared_ptr<ros::CallbackQueue::IDInfo> > > >::find(unsigned long const&) ()
   from /opt/ros/indigo/lib/libroscpp.so
(gdb) bt
#0  0x00007ffff74b49f4 in std::_Rb_tree<unsigned long, std::pair<unsigned long const, boost::shared_ptr<ros::CallbackQueue::IDInfo> >, std::_Select1st<std::pair<unsigned long const, boost::shared_ptr<ros::CallbackQueue::IDInfo> > >, std::less<unsigned long>, std::allocator<std::pair<unsigned long const, boost::shared_ptr<ros::CallbackQueue::IDInfo> > > >::find(unsigned long const&) ()
   from /opt/ros/indigo/lib/libroscpp.so
#1  0x00007ffff74b2a3f in ros::CallbackQueue::removeByID(unsigned long) ()
   from /opt/ros/indigo/lib/libroscpp.so
#2  0x00007ffff74a8c2f in ros::Publication::removeCallbacks(boost::shared_ptr<ros::SubscriberCallbacks> const&) () from /opt/ros/indigo/lib/libroscpp.so
#3  0x00007ffff749b9ba in ros::TopicManager::unadvertise(std::string const&, boost::shared_ptr<ros::SubscriberCallbacks> const&) ()
   from /opt/ros/indigo/lib/libroscpp.so
#4  0x00007ffff748f698 in ros::Publisher::Impl::unadvertise() ()
   from /opt/ros/indigo/lib/libroscpp.so
#5  0x00007ffff748f707 in ros::Publisher::Impl::~Impl() ()
   from /opt/ros/indigo/lib/libroscpp.so
#6  0x00007ffff748fe39 in boost::detail::sp_counted_impl_pd<ros::Publisher::Impl*, boost::detail::sp_ms_deleter<ros::Publisher::Impl> >::dispose() ()
   from /opt/ros/indigo/lib/libroscpp.so
#7  0x000000000040624e in boost::detail::sp_counted_base::release() ()
#8  0x00007fffe615407e in moveit::planning_interface::MoveGroup::MoveGroupImpl::MoveGroupImpl (this=0x7fffb8000da0, opt=..., tf=..., wait_for_servers=...)
---Type <return> to continue, or q <return> to quit---
    at /home/illing/ros/src/moveit/moveit_ros/planning_interface/move_group_interface/src/move_group.cpp:95
#9  0x00007fffe60be097 in moveit::planning_interface::MoveGroup::MoveGroup (
    this=0x7fffb8000d60, opt=..., tf=..., wait_for_servers=...)
    at /home/illing/ros/src/moveit/moveit_ros/planning_interface/move_group_interface/src/move_group.cpp:1261
#10 0x00007fffec285a74 in boost::make_shared<moveit::planning_interface::MoveGroup, moveit::planning_interface::MoveGroup::Options&, boost::shared_ptr<tf::TransformListener> const&, ros::WallDuration>(moveit::planning_interface::MoveGroup::Options&, boost::shared_ptr<tf::TransformListener> const&, ros::WallDuration&&)
    (arg1=...) at /usr/include/boost/smart_ptr/make_shared_object.hpp:218
#11 0x00007fffec267e32 in simple_pick_place_fkie::GraspAtPointExecution::createMoveGroup (this=this@entry=0xb90ba0)
    
    ...

gaspers, 2017-01-26:
We implemented this fix and tried to not break the behaviour introduced by https://github.com/ros-planning/moveit_ros/commit/9479afcfd6bd05b3580d4bfb87cb773f29a905d4

For us, this fix works fine in the described setup.

v4hn, 2017-02-01:
Resolved in #426

closed 2017-02-01 by v4hn