moveit/moveit #427: Installing from source on kinetic

mcevoyandy, 2017-01-26:
### Description

Attempting to install from source on kinetic. Following the install steps, `catkin build` errors out saying `ModelInterfaceSharedPtr` in namespace `urdf` does not name a type.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source 

### Steps to reproduce
Followed steps for installing from source for kinetic.

### Backtrace or Console output

https://gist.github.com/mcevoyandy/139f82fc0ed319bd21158f5da2531e14

mcevoyandy, 2017-01-26:
did `robot_model` from source as well on the `kinetic_devel` branch. This seems to have fixed the problem.  FYI, I started installing moveit from source because of C++11 error from `geometric_shapes` package. The issue regarding that was closed, but it seems only if you install from source?

v4hn, 2017-01-30:
Can anyone validate this? I don't see why this should/would happen. The urdf included and released these typedefs.

de-vri-es, 2017-01-30:
@mcevoyandy: Is it possible that catkin was finding an old version of `robot_model` that was causing problems? Perhaps you hadn't upgraded your system packages in a while?

mcevoyandy, 2017-02-01:
@de-vri-es: possibly. This was a new install of 16.04 and kinetic though. I got the C++11 error with the debian version I installed following the binary install  tutorial. I tried upgrading my packages to see if this helped, but still kept getting the error so tried to install from source and got the urdf namespace error... I don't know. If I'm the only one who's seen this then it must be something on my end. It's all working now though...

v4hn, 2017-02-01:
Closing as resolved/not reproducible

v4hn, 2017-02-01:
Actually, @ceesietopc just brought up this exact problem in the #moveit IRC channel.
If I'm correct this problem exists because OSRF still serves robot_model version 1.12.5, but there was a missing namespace that got fixed only later in https://github.com/ros/robot_model/commit/409c4b923c38582ba595c7fc2a0536f0ee60045c#diff-e142433069a586e135e63a432e30b99dR86 .
This is already released in 1.12.7, but not yet synced.

@davetcoleman Do you have any idea why this does not break in moveit_ci? As far as I can see it *should*.

davetcoleman, 2017-03-11:
Yes, moveit_ci should have caught this error using the non-shadow-fixed build, this is concerning. I've just looked into the scripts and I can't pinpoint the error without further testing. Unfortunately I'm late in responding to this question, and robot_model [has since been released](https://github.com/ros/robot_model/releases). Not sure I can debug it easily until a similar problem re-emerged.

v4hn, 2017-03-20:
As @davetcoleman stated above, the package is synced in OSRF's deb-repository.
Closing as resolved.

closed 2017-03-20 by v4hn

pro100kot, 2018-02-20:
```
sudo apt-get install ros-kinetic-urdf
```
Helped for me