moveit/moveit #430: DistancetoCollision within a Planner

cgeckeler, 2017-01-27:
### Description
While attempting to add a Planner to MoveIt, I ran across a problem: it does not seem possible to call DistancetoCollision from within the Planner. For the interface, etc I looked to the already implemented CHOMP planner. In order for distanceToCollision to recognize collision objects `planning_scene_interface_.getObjects()` is used, this causes an infinite loop (possibly because the planner is called within move_group , and the PlanningSceneInterface also needs it). The exact same code(see below) used outside of the planner (eg. with action server) functions without a problem.

If there is another way of going about this, please let me know, otherwise adding the ability to get distance to environment calls to be used within planners would be nice enhancement feature. Any help would be greatly appreciated , thank you!

See also:
http://answers.ros.org/question/252110/distancetocollision-within-a-moveit-planner/

### Your environment
* ROS Distro: Jade
* OS Version: Ubuntu 14.04
* Source build

### Steps to reproduce
Building moveit from source add 
```
  planning_scene_monitor::PlanningSceneMonitorPtr planningMonitor_;
  planningMonitor_ = boost::make_shared<planning_scene_monitor::PlanningSceneMonitor>("robot_description");

  planning_scene_monitor::LockedPlanningSceneRW ps(planningMonitor_);
  ps->getCurrentStateNonConst().update();
  planning_scene::PlanningScenePtr scene = ps->diff();
  scene->decoupleParent();

  //get state
  robot_state::RobotState state = scene->getCurrentStateNonConst();

  //process collision objects in scene
  moveit::planning_interface::PlanningSceneInterface planning_scene_interface_;
  std::map<std::string, moveit_msgs::CollisionObject> c_objects_map = planning_scene_interface_.getObjects(); //error here
  for(auto& kv : c_objects_map){
      scene->processCollisionObjectMsg(kv.second);
  }

  //print distance
  ROS_INFO_STREAM("Distance to Collision: " << scene->distanceToCollision(state));
```

to the solve function of chomp_planner.cpp , as well as moveit_ros_planning_interface to the package.xml and CMakeLists.txt. 

### Expected behaviour
Get the distance to nearest collision.

### Actual behaviour
Infinite Loop.

avoelz, 2017-01-30:
If you write your own planner for MoveIt, then you should implement planning_interface::PlannerManager and planning_interface::PlanningContext. In PlannerManager you have to implement the method getPlanningContext which has a PlanningSceneConstPtr as argument. You should pass this planning scene to your own PlanningContext, for example in the constructor. Use this planning scene for all your collision and collision distance queries. It is not necessary to use a PlanningSceneMonitor within your planner plugin.

cgeckeler, 2017-01-30:
Thank you for your answer, that is good to know and the CHOMP planner is constructed as you describe. However, calling distanceToCollision even with the PlannerSceneConstPtr still does not return a meaningful distance (gives the maximum value). To recognize obstacles that have been added to the scene, `planning_scene_interface_.getObjects()` is necessary, for which the  PlanningSceneInterface is required, which is where my problem lies. If this is incorrect or there is another way to perform a distance call, I would be much obliged.

avoelz, 2017-01-30:
As fas as I understand the MoveIt code, planning_scene_interface_.getObjects() does nothing else than calling the GetPlanningScene-Service and returning the list of collision objects. I would assume that the scene returned by GetPlanningScene is the same that is passed to the planner plugin in getPlanningContext. To check this, you could print the ids of the collision objects in the scene. I suppose that you are searching the error in the wrong place.

cgeckeler, 2017-01-30:
Indeed, the getObjects function just returns a list of collision objects which are then processed with scene.processCollisionObjectMsg, from the above code: 
 ```
std::map<std::string, moveit_msgs::CollisionObject> c_objects_map = 
     planning_scene_interface_.getObjects();

  for(auto& kv : c_objects_map){
      scene->processCollisionObjectMsg(kv.second);
  }
```
Without these steps the obstacles are not recognized and the distanceFunction returns the maximum value. This code also works without a problem when run externally. However when running it from within a planner the planning_scene_interface does not instantiate and instead goes into an infinite loop.

avoelz, 2017-01-30:
This still seems very strange to me. processCollsionObjectMsg does nothing more than adding the collision object to the scene. If you don't get collision distances without this call, then the collision objects were not yet in the scene. Which collision objects are in the scene according to the RViz motion planning plugin and which objects are in the scene that is passed to getPlanningContext in PlannerManager? This is where I would start to search.

v4hn, 2017-01-31:
Hey @cgeckeler,

the `*_interface` packages are meant to be used only from other ROS nodes to access common internal functions of the `move_group` node. They should definitely not be used from within a planner module that runs *as part* of the move_group node! This is highly inefficient and there are better alternatives.
I'm not sure whether we *could* support this usage, but it's clearly not intended. Mainly the internal locking of the planning scene intervenes there.
The planner receives one frozen version of the planning scene maintained by the node and works on it.
If the objects you added before you sent the planning request are not part of the scene your planner receives, they did not arrive at move_group node's PlanningSceneMonitor yet.
In the node that sends your planning requests, you might want to add the objects via `PlanningSceneInterface::applyCollisionObjects` to make sure they are part of `move_group`'s planning scene before you send the planning request.

@avoelz already gave a number of hints, so I'm inclined to close this as wont-fix.

cgeckeler, 2017-01-31:
Thank you for your suggestion, I tried looking there, and  according to rviz the collision objects are present, but using` planning_scene->printKnownObjects` on the planning scene  within the planner does not return any objects. This is strange considering that the planner plans around them and therefore must recognize that they exist.

avoelz, 2017-01-31:
Where did you place the call `planning_scene->printKnownObjects`? If I look at the method getPlanningContext in chomp_plugin.cpp, I see that the planning scene is not passed on to the planning context (as it should be). Instead, looking at chomp_planning_context.cpp, I see that in the constructor a new (and therefore empty) planning scene is created and this one is used in solve. If I am right, then you should get the message `ROS_INFO_STREAM("Configuring New Planning Scene.");` printed.

v4hn, 2017-01-31:
Oh. Well, the chomp interface *compiles* again, but it's definitely not ready for use at the moment.
It's a bit sad to hear it has such severe shortcomings...

cgeckeler, 2017-02-01:
@avoelz Yes, you are correct the "Configuring New Planning Scene" is printed, which would explain why the collision objects are not in the scene, and where the trouble started. I will call the planner outside of moveit with an action server which circumvents this issue. 

@v4hn That makes sense, and explains the undefined behaviour when calling these functions from within the planner.  The `PlanningSceneInterface::applyCollisionObjects` works with adding the objects to the planning scene.

Thank you very much for your replies, they have helped immensely! 

Closing this issue since this is not a problem with the `PlanningScene::distanceToCollision` function, rather it's unexpected behaviour from the CHOMP planning context.

closed 2017-02-01 by cgeckeler