moveit/moveit #431: "apt-get install ros-indigo-moveit" not installing

130s, 2017-01-30:
A few people in the comment section of [this post](http://answers.ros.org/question/251168/unable-to-locate-package-ros-indigo-moveit-full/?answer=251173#post-id-251173) reported the issue in the title.

If you see the issue, please provide sufficient information including the output of the following commands:

```
lsb_release -a
more /etc/apt/sources.list.d/ros-latest.list
sudo apt-get update
```

bhavyadoshi26, 2017-01-30:
`sudo apt-get install ros-indigo-moveit`

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> E: Unable to locate package ros-indigo-moveit

`lsb_release -a`

> No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

`more /etc/apt/sources.list.d/ros-latest.list`

> deb http://packages.ros.org/ros/ubuntu trusty main

`sudo apt-get update`

> Get:1 http://security.ubuntu.com trusty-security InRelease [65,9 kB]
> Ign http://extras.ubuntu.com trusty InRelease                                  
> Hit http://ppa.launchpad.net trusty InRelease                                  
> Ign http://de.archive.ubuntu.com trusty InRelease                              
> Hit http://ppa.launchpad.net trusty InRelease                                  
> Get:2 http://extras.ubuntu.com trusty Release.gpg [72 B]                       
> Get:3 http://de.archive.ubuntu.com trusty-updates InRelease [65,9 kB]          
> Hit http://extras.ubuntu.com trusty Release                                    
> Hit http://ppa.launchpad.net trusty/main amd64 Packages                        
> Get:4 http://security.ubuntu.com trusty-security/main Sources [124 kB]         
> Get:5 http://de.archive.ubuntu.com trusty-backports InRelease [65,9 kB]        
> Hit http://extras.ubuntu.com trusty/main Sources                               
> Hit http://ppa.launchpad.net trusty/main i386 Packages                         
> Hit http://extras.ubuntu.com trusty/main amd64 Packages                        
> Hit http://ppa.launchpad.net trusty/main Translation-en                        
> Get:6 http://security.ubuntu.com trusty-security/universe Sources [48,6 kB]    
> Hit http://de.archive.ubuntu.com trusty Release.gpg                            
> Hit http://ppa.launchpad.net trusty/main Sources                               
> Hit http://extras.ubuntu.com trusty/main i386 Packages                         
> Get:7 http://security.ubuntu.com trusty-security/multiverse Sources [3.208 B]  
> Get:8 http://de.archive.ubuntu.com trusty-updates/main Sources [389 kB]        
> Hit http://ppa.launchpad.net trusty/main amd64 Packages                        
> Get:9 http://security.ubuntu.com trusty-security/main amd64 Packages [574 kB]  
> Hit http://ppa.launchpad.net trusty/main i386 Packages                         
> Hit http://ppa.launchpad.net trusty/main Translation-en                        
> Get:10 http://packages.ros.org trusty InRelease [4.031 B]                      
> Get:11 http://security.ubuntu.com trusty-security/universe amd64 Packages [150 kB]
> Get:12 http://packages.ros.org trusty/main amd64 Packages [788 kB]             
> Get:13 http://security.ubuntu.com trusty-security/multiverse amd64 Packages [4.136 B]
> Get:14 http://security.ubuntu.com trusty-security/main i386 Packages [534 kB]  
> Get:15 http://de.archive.ubuntu.com trusty-updates/universe Sources [172 kB]   
> Ign http://extras.ubuntu.com trusty/main Translation-en_US                     
> Get:16 http://security.ubuntu.com trusty-security/universe i386 Packages [151 kB]
> Ign http://extras.ubuntu.com trusty/main Translation-en                        
> Get:17 http://de.archive.ubuntu.com trusty-updates/multiverse Sources [7.535 B]
> Get:18 http://security.ubuntu.com trusty-security/multiverse i386 Packages [4.284 B]
> Get:19 http://de.archive.ubuntu.com trusty-updates/main amd64 Packages [939 kB]
> Get:20 http://security.ubuntu.com trusty-security/main Translation-en [317 kB] 
> Get:21 http://security.ubuntu.com trusty-security/multiverse Translation-en [2.201 B]
> Get:22 http://security.ubuntu.com trusty-security/universe Translation-en [88,2 kB]
> Get:23 http://de.archive.ubuntu.com trusty-updates/universe amd64 Packages [395 kB]
> Get:24 http://de.archive.ubuntu.com trusty-updates/multiverse amd64 Packages [14,0 kB]
> Get:25 http://de.archive.ubuntu.com trusty-updates/main i386 Packages [898 kB] 
> Get:26 http://de.archive.ubuntu.com trusty-updates/universe i386 Packages [397 kB]
> Get:27 http://packages.ros.org trusty/main i386 Packages [788 kB]              
> Get:28 http://de.archive.ubuntu.com trusty-updates/multiverse i386 Packages [14,4 kB]
> Get:29 http://de.archive.ubuntu.com trusty-updates/main Translation-en [461 kB]
> Get:30 http://de.archive.ubuntu.com trusty-updates/multiverse Translation-en [7.340 B]
> Get:31 http://de.archive.ubuntu.com trusty-updates/universe Translation-en [209 kB]
[... 32 lines cut ...]

After which `sudo apt-get install ros-indigo-moveit` worked and it installed. 

> Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  bbswitch-dkms lib32gcc1 libc6-i386 libpcre++-dev libpcre++0 libvdpau1
  linux-headers-4.4.0-31 linux-headers-4.4.0-31-generic
  linux-image-4.4.0-31-generic linux-image-extra-4.4.0-31-generic
  linux-signed-image-4.4.0-31-generic ros-indigo-collada-robots
  screen-resolution-extra
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  ros-indigo-moveit
0 upgraded, 1 newly installed, 0 to remove and 123 not upgraded.
Need to get 2.360 B of archives.
After this operation, 48,1 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros/ubuntu/ trusty/main ros-indigo-moveit amd64 0.7.6-0trusty-20161230-113550-0800 [2.360 B]
Fetched 2.360 B in 0s (6.321 B/s)              
Selecting previously unselected package ros-indigo-moveit.
(Reading database ... 443070 files and directories currently installed.)
Preparing to unpack .../ros-indigo-moveit_0.7.6-0trusty-20161230-113550-0800_amd64.deb ...
Unpacking ros-indigo-moveit (0.7.6-0trusty-20161230-113550-0800) ...
Setting up ros-indigo-moveit (0.7.6-0trusty-20161230-113550-0800) ...

130s, 2017-01-30:
> sudo apt-get update
> :
> After which sudo apt-get install ros-indigo-moveit worked and it installed.

This sounds like to me that your package index was old when you were having an issue, which was updated by `apt-get update` and that's why you now are able to install the package in question. If this is correct, the issue raised has nothing to with either MoveIt! or ROS.

Closing but feel free to give additional info/question.

closed 2017-01-30 by 130s

bhavyadoshi26, 2017-01-30:
Yes, that seems to be the logical explanation. Thank you for the inputs!

atenpas, 2017-05-30:
In case someone runs into this issue: _/etc/apt/sources.list.d/ros-latest.list_ did not exist for me, only _/etc/apt/sources.list.d/ros-latest.list.save_. The solution is:
`sudo mv /etc/apt/sources.list.d/ros-latest.list.save /etc/apt/sources.list.d/ros-latest.list
sudo apt-get update && sudo apt-get install ros-indigo-moveit `

manacoa, 2017-06-22:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
```

```
more /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros/ubuntu xenial main
```

```
sudo apt-get update
Get:1 file:/var/cuda-repo-8-0-local-ga2  InRelease
Ign:1 file:/var/cuda-repo-8-0-local-ga2  InRelease
Get:2 file:/var/nv-deep-learning-repo-ga-cuda8.0-digits5.0  InRelease
Ign:2 file:/var/nv-deep-learning-repo-ga-cuda8.0-digits5.0  InRelease
Get:3 file:/var/cuda-repo-8-0-local-ga2  Release [574 B]
Get:4 file:/var/nv-deep-learning-repo-ga-cuda8.0-digits5.0  Release [574 B]
Get:3 file:/var/cuda-repo-8-0-local-ga2  Release [574 B]
Get:4 file:/var/nv-deep-learning-repo-ga-cuda8.0-digits5.0  Release [574 B]
Hit:7 http://gb.archive.ubuntu.com/ubuntu xenial InRelease                     
Get:8 http://gb.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Get:9 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]     
Get:10 http://gb.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB] 
Hit:11 http://packages.ros.org/ros/ubuntu xenial InRelease                     
Hit:12 https://download.sublimetext.com apt/stable/ InRelease  
Fetched 306 kB in 0s (341 kB/s)
Reading package lists... Done
W: Invalid 'Date' entry in Release file /var/lib/apt/lists/_var_nv-deep-learning-repo-ga-cuda8.0-digits5.0_Release
```

```
sudo apt-get install ros-indigo-moveitReading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-indigo-moveit
```

This is after an uneventful fresh install of Kinetic Desktop Full on Ubuntu 16.04.

gavanderhoorn, 2017-06-22:
```
Codename:	xenial
```

and

```
sudo apt-get install ros-indigo-moveit
```

Are you trying to install Indigo MoveIt on Kinetic?

manacoa, 2017-06-22:
Yes!

manacoa, 2017-06-22:
Oops

gavanderhoorn, 2017-06-22:
Well that won't work: the Indigo binaries are not released on Kinetic. That is the reason why `apt-get` fails.

manacoa, 2017-06-22:
My apologies! Frazzled brain. Sorry for wasting your time.