moveit/moveit #449: Octomap is ignored during planning

cristipiticul, 2017-02-21:
### Description

I am working with the Baxter robot and an Asus Xtion 3D camera for detection. I am using the PointCloudUpdater plugin to create an octomap, which is used for collision detection. The octomap is built successfully and I can see it in RViz.

If I try to make a plan, and there is an obstacle between the start pose and the goal pose, the trajectory goes **through** the obstacle (the arm does not avoid it). Still, after finding the trajectory, the planner checks the trajectory and find that it collides with the octomap. So the planning fails.

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Binary build
* Version 0.7.2

### Steps to reproduce
If there is one box in the scene:
1. Move the arm to the left of the box
2. Set the goal state to a pose on the right of the box
3. Plan

### Expected behaviour
The planner returns a trajectory which avoids the box.

### Actual behaviour
The planner fails, because the trajectory (probably generated ignoring the octomap) goes through the box.

### Console output

```
[ INFO] [1487677882.633983235, 1190.572000000]: both_arms[RRTkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1487677882.634871240, 1190.572000000]: both_arms[RRTkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1487677882.684539513, 1190.584000000]: both_arms[RRTkConfigDefault]: Created 9 states
[ INFO] [1487677882.929209239, 1190.694000000]: both_arms[RRTkConfigDefault]: Created 29 states
[ INFO] [1487677882.929438338, 1190.694000000]: ParallelPlan::solve(): Solution found by one or more threads in 0.295611 seconds
[ INFO] [1487677882.929507316, 1190.694000000]: SimpleSetup: Path simplification took 0.000003 seconds and changed from 2 to 2 states
[ERROR] [1487677882.942114849, 1190.701000000]: Computed path is not valid. Invalid states at index locations: [ 5 6 7 ] out of 10. Explanations follow in command line. Contacts are published on /move_group/display_contacts
[ INFO] [1487677882.942764871, 1190.701000000]: Found a contact between '<octomap>' (type 'Object') and 'l_gripper_r_finger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1487677882.942825552, 1190.701000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1487677882.945538546, 1190.701000000]: Found a contact between '<octomap>' (type 'Object') and 'l_gripper_r_finger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1487677882.945596192, 1190.701000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1487677882.947169251, 1190.701000000]: Found a contact between '<octomap>' (type 'Object') and 'l_gripper_l_finger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1487677882.947237554, 1190.701000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1487677882.948733782, 1190.701000000]: Completed listing of explanations for invalid states.
[ WARN] [1487677882.977036152, 1190.723000000]: Fail: ABORTED: Motion plan was found but it seems to be invalid (possibly due to postprocessing). Not executing.
```

v4hn, 2017-02-21:
This sounds like the discrete collision checks during planning miss the octomap collision.
MoveIt with OMPL does not do continuous collision checks, but instead only checks discrete configurations.
You can reduce the distance between checked states in your moveit config in `config/ompl_planning.yaml`. The parameter is called [`longest_valid_segment_fraction`](https://github.com/TAMS-Group/tams_ur5_setup/blob/master/tams_ur5_setup_moveit_config/config/ompl_planning.yaml#L83) and the currently advertised value from our side is `0.005`.
If this still leads to collisions in your case, then you might want to reduce it further, but keep in mind that more collision checks also means more time is needed for planning...

cristipiticul, 2017-02-21:
You are right! It was set to 0.05. I set it to 0.005 and it worked as expected. Thank you very much!

closed 2017-02-21 by cristipiticul

davetcoleman, 2017-02-21:
note this has been documented in the [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/time_parameterization_tutorial.html)

cristipiticul, 2017-02-22:
Maybe you mean [this](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ompl_interface_tutorial.html)? You are right, it is documented. Thanks for your note!

davetcoleman, 2017-02-22:
Ah yes, thanks :)