moveit/moveit #456: [Indigo] undefined symbol in planning_scene_monitor

rbbg, 2017-03-03:
I am running into the following error when trying to call `planning_scene_monitor::getStateUpdateFrequency()`:

`/opt/ros/indigo/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/ruben/cohesion/iiwa_ws/devel/lib/libmoveit_rt_optim.so: undefined symbol: _ZN22planning_scene_monitor20PlanningSceneMonitor23getStateUpdateFrequencyEv`

This suggests a linking error, but `planning_scene_monitor::setStateUpdateFrequency(double)` _does_ work, so it seems like the symbol is missing from this version of the library.

My environment:
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Binary build -> Linked against `libmoveit_planning_scene_monitor.so.0.7.3`

v4hn, 2017-03-03:
does rebuilding your `/home/ruben/cohesion/iiwa_ws/` workspace solve the problem?

rbbg, 2017-03-03:
Found the problem: the function is only [declared but never defined](https://github.com/ros-planning/moveit/search?utf8=%E2%9C%93&q=getStateUpdateFrequency). Should be easy to fix :). I'll put in a PR.

v4hn, 2017-03-04:
Wow.

A Pull-Request for this would be great.

v4hn, 2017-03-20:
resolved in #463

closed 2017-03-20 by v4hn