moveit/moveit #457: Release Kinetic 0.9.5

130s, 2017-03-04:
https://github.com/ros-planning/moveit/issues/86 is a regression that makes hard to use Kinetic and now it's fixed already. I'd like to make a quick patch release. Let me know by 6th Monday if there's any objections.

v4hn, 2017-03-04:
Currently, we have a regression in the rviz display that #452 should fix.
With current head, the visualization of the planning scene will disappear when you hide the panel (e.g. hide the docking bar which contains the panel).

Please do not release without a fix for this.

130s, 2017-03-04:
Ok. 

I also want to make releases for IJ at the same time, simply because that makes the release job way easier. So, please list up any TODOs for IJ too.

v4hn, 2017-03-04:
On Sat, Mar 04, 2017 at 11:05:17AM -0800, Isaac I.Y. Saito wrote:
> I also want to make releases for IJ at the same time,
> simply because that makes the release job way easier.
> So, please list up any TODOs for IJ too.

In this case please give me one more day to do abi checks.
I can only run them at work, and I don't plan to go there on Sunday :)

v4hn, 2017-03-07:
Ok, apart from #452, you got green light from my perspective.

130s, 2017-03-07:
So for this quick patch release, do we just increment the patch/micro version for each distro (Unfortunately I missed the relevant part of the maintainer's meeting today)?

v4hn, 2017-03-08:
patch release -> patch version :)

With indigo and jade, we pretty much agreed not to bump anything but the patch version number.

130s, 2017-03-09:
K https://github.com/ros/rosdistro/pull/14135
J https://github.com/ros/rosdistro/pull/14132
I https://github.com/ros/rosdistro/pull/14133

Patch version incremented per each distro.

closed 2017-03-09 by 130s