moveit/moveit #460: package 'moveit_planners_ompl' depends on non-existent package 'ompl'

lkj1114889770, 2017-03-05:
### Description
I create a ros package to use move_group interface with c++,and in the package.xml, i add run depend moveit_planners_ompl,that is where error occurs.It tell me the follwing infoprmation:
[rosbuild] Building package move_group_test
  Failed to invoke /opt/ros/kinetic/bin/rospack deps-manifests move_group_test
  [rospack] Error: package 'moveit_planners_ompl' depends on non-existent package 'ompl' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
  
  
  CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:129 (message):
    
  
    Failed to invoke rospack to get compile flags for package
    'move_group_test'.  Look above for errors from rospack itself.  Aborting.
    Please fix the broken dependency!
And when i use rosfind pack ompl, it also tell me no package called ompl. so is that means there is no package called ompl, but how can i solve the problem of the dependency of  moveit_planners_ompl?
Overview of your issue here.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2017-03-05:
You didn't fill in the rest of the issue template, but from the mention of `rosbuild` in your console copy-paste I have the impression that you're using `rosmake` and not `catkin`.

First: all of MoveIt is completely Catkin based, so `rosmake` will not work. Second: `ompl` is not a catkin package but a system dependency, so `rospack find ompl` (which is the command I think you used) is not expected to find it.

Can you tell us which commands exactly you are using to create and then build your package?

lkj1114889770, 2017-03-05:
I just want to get move_group service with moveit,so i download a demo from https://github.com/jontromanab/move_group_test  ,but i got a lot of problem when i put the package i downlaod from the link to the catkin workspace, a lot of problem arise when i catkin_make the workspace. So i turned to create ros package in roscreate-pkg method whith the c++ file provided by the package i downloaded, and also according to the package.xml file the package provided, i also add the same dependency as the xml file describe whinch including moveit_planners_ompl. since ompl is not a catkin package ,how did moveit_planners_ompl works with dependency of ompl package as described in moveit_planners_ompl package's xml file.

gavanderhoorn, 2017-03-05:
> So i turned to create ros package in roscreate-pkg method

Ah, and that created a `rosbuild` package.

Catkin pkgs can depend on `rosbuild`, but not the other way around, so this will never work.

> So i turned to create ros package in roscreate-pkg method whith the c++ file provided by the package i downloaded, and also according to the package.xml file the package provided, i also add the [...]

Please first create a regular Catkin package ([catkin_create_pkg](http://wiki.ros.org/ROS/Tutorials/catkin/CreatingPackage)) and see whether that works. I don't think it makes much sense trying to diagnose what's going on with your current package.

gavanderhoorn, 2017-03-05:
Btw, please always mention that you are cross-posting your issue to other trackers / sites. jontromanab/move_group_test#1 seems like an almost verbatim copy.

lkj1114889770, 2017-03-05:
Got it. This is my first time to put my issues in github, i will do as you adviced. Thank you!

gavanderhoorn, 2017-03-06:
@lkj1114889770: can I conclude from #462 that you got things to work with Catkin?

Can we close this one?

closed 2017-03-06 by gavanderhoorn