moveit/moveit #462: /opt/ros/kinetic/include/moveit/macros/declare_ptr.h:52:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type    typedef std::shared_ptr<Type> Name##Ptr;

lkj1114889770, 2017-03-06:
### Description
I program C++ file with movegroup interface according to the moveit tutorial, the program i built is show in the follwing:

 #include <ros/ros.h>
#include <moveit/move_group_interface/move_group_interface.h>
int main(int argc, char **argv)
{
  ros::init(argc, argv, "lesson_move_group");

 
  ros::AsyncSpinner spinner(1);
  spinner.start();

  moveit::planning_interface::MoveGroup group("manipulator");
  
  std::map<std::string, double> joints;
  joints["shoulder_pan_joint"] = -0.40;
  joints["shoulder_lift_joint"] =  0.27;
  joints["elbow_joint"] =  0.48;
  joints["wrist_1_joint"] =  0.07;
  joints["wrist_2_joint"] =  0.75;
  joints["wrist_3_joint"] = -0.19;

  group.setJointValueTarget(joints);
  
  group.move();

  
}

Overview of your issue here.
After catkin_make the package, the erros are:

In file included from /home/lkj/catkin_ws/src/move_group_test/src/move_joint.cpp:2:0:
/opt/ros/kinetic/include/moveit/move_group_interface/move_group.h:43:2: warning: #warning "This header is deprecated and will go away in ROS lunar." "Please use moveit/move_group_interface/move_group_interface.h" "and the class MoveGroupInterface instead of MoveGroup" [-Wcpp]
 #warning "This header is deprecated and will go away in ROS lunar."\
  ^
In file included from /opt/ros/kinetic/include/moveit/macros/class_forward.h:40:0,
                 from /opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h:41,
                 from /opt/ros/kinetic/include/moveit/move_group_interface/move_group.h:40,
                 from /home/lkj/catkin_ws/src/move_group_test/src/move_joint.cpp:2:
/opt/ros/kinetic/include/moveit/macros/declare_ptr.h:52:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<Type> Name##Ptr;                                                                             \
                ^
/opt/ros/kinetic/include/moveit/macros/class_forward.h:51:3: note: in expansion of macro ‘MOVEIT_DECLARE_PTR’
   MOVEIT_DECLARE_PTR(C, C);
   ^
/opt/ros/kinetic/include/moveit/robot_model/link_model.h:50:1: note: in expansion of macro ‘MOVEIT_CLASS_FORWARD’
 MOVEIT_CLASS_FORWARD(Shape);
 ^
/opt/ros/kinetic/include/moveit/macros/declare_ptr.h:53:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<const Type> Name##ConstPtr;
                ^
/opt/ros/kinetic/include/moveit/macros/class_forward.h:51:3: note: in expansion of macro ‘MOVEIT_DECLARE_PTR’
   MOVEIT_DECLARE_PTR(C, C);
   ^
/opt/ros/kinetic/include/moveit/robot_model/link_model.h:50:1: note: in expansion of macro ‘MOVEIT_CLASS_FORWARD’
 MOVEIT_CLASS_FORWARD(Shape);
....
....

move_group_test/CMakeFiles/move_joint.dir/build.make:62: recipe for target 'move_group_test/CMakeFiles/move_joint.dir/src/move_joint.cpp.o' failed
make[2]: *** [move_group_test/CMakeFiles/move_joint.dir/src/move_joint.cpp.o] Error 1
CMakeFiles/Makefile2:4223: recipe for target 'move_group_test/CMakeFiles/move_joint.dir/all' failed
make[1]: *** [move_group_test/CMakeFiles/move_joint.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

It seems that the problem comes from the library of moveit because of  Compatibility or for the other reason. Can yo tell me how to resolve the problem
### Your environment
[... 174 lines cut ...]
   shapes::ShapePtr constructShape(const urdf::Geometry* geom);
           ^
In file included from /opt/ros/kinetic/include/moveit/robot_state/robot_state.h:42:0,
                 from /opt/ros/kinetic/include/moveit/move_group_interface/move_group_interface.h:43,
                 from /opt/ros/kinetic/include/moveit/move_group_interface/move_group.h:40,
                 from /home/lkj/catkin_ws/src/move_group_test/src/move_joint.cpp:2:
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:63:80: error: ‘ShapeConstPtr’ is not a member of ‘shapes’
   AttachedBody(const LinkModel* link, const std::string& id, const std::vector<shapes::ShapeConstPtr>& shapes,
                                                                                ^
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:63:80: error: ‘ShapeConstPtr’ is not a member of ‘shapes’
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:63:101: error: template argument 1 is invalid
   AttachedBody(const LinkModel* link, const std::string& id, const std::vector<shapes::ShapeConstPtr>& shapes,
                                                                                                     ^
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:63:101: error: template argument 2 is invalid
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:88:21: error: ‘ShapeConstPtr’ is not a member of ‘shapes’
   const std::vector<shapes::ShapeConstPtr>& getShapes() const
                     ^
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:88:21: error: ‘ShapeConstPtr’ is not a member of ‘shapes’
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:88:42: error: template argument 1 is invalid
   const std::vector<shapes::ShapeConstPtr>& getShapes() const
                                          ^
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:88:42: error: template argument 2 is invalid
/opt/ros/kinetic/include/moveit/robot_state/attached_body.h:141:15: error: ‘ShapeConstPtr’ is not a member of ‘shapes’

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gavanderhoorn, 2017-03-06:
```
/opt/ros/kinetic/include/moveit/macros/declare_ptr.h:52:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<Type> Name##Ptr;
```

Have you enabled C++11 in your `CMakeLists.txt`? If not, that is most likely the cause of your issue.

[moveit_core/CMakeLists.txt](https://github.com/ros-planning/moveit/blob/7dd17dd73ac53251b6e2560949944bcf3ed40f3a/moveit_core/CMakeLists.txt#L4) shows one way to do that.

lkj1114889770, 2017-03-06:
YES! It works to add C++11 in CMakeList.txt.

closed 2017-03-06 by gavanderhoorn

YuehChuan, 2017-10-28:
@lkj1114889770  cool  same issue,  and thanks @gavanderhoorn . Did you rebuild moveIt with `C++11` from source ?  I'm migrating code from indigo to kinectic, and  add  `add_compile_options(-std=c++11)` in every custom ros package CMakeLists.txt. But not yet rebuild moveit from source.

v4hn, 2017-11-17:
@YuehChuan You don't have to build MoveIt yourself, but all packages that include MoveIt headers have to be compiles with `-std=c++11` which is what adding `add_compile_options(-std=c++11)` to the beginning of all the respective `package/CMakeLists.txt` does.

xixiaoliang, 2020-06-22:
> ```
> /opt/ros/kinetic/include/moveit/macros/declare_ptr.h:52:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
>    typedef std::shared_ptr<Type> Name##Ptr;
> ```
>
> Have you enabled [...]

yes,good job.Thanks