moveit/moveit #469: [source building erro-moveIt 0.7.8 indigo]catkin_package() include dir 'include/eigen3'

YuehChuan, 2017-03-12:
### Description
moveIt indigo 0.7.8 build from source eigen erro
catkin_package() include dir 'include/eigen3' does not exist relative to
### Your environmentckage.xml  CMakeLists.txt
* ROS Distro: Indigo
* OS Version: e.g. Ubuntu 14.04
* Source build
* If source, which git commit or tag?
0.7.8 moveIt

### Steps to reproduce

Already fixed by myself just in case someone get same issue

under catkin_ws/src
```bash=
wstool init .
wstool merge https://raw.githubusercontent.com/ros-planning/moveit/indigo-devel/moveit.rosinstall
wstool update
rosdep install --from-paths . --ignore-src --rosdistro indigo
cd ..
catkin config --extend /opt/ros/indigo --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build
```
the erro showes:
catkin_package() include dir 'include/eigen3' 

http://answers.ros.org/question/256794/moveit-catkin_package-include-dir-includeeigen3-does-not-exist-relative-to/

### Expected behaviour
catkin build success
 

### Actual behaviour
Tell us what happens instead
catkin_package() include dir 'include/eigen3' does not exist relative to

```bash=
 Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:296 (message): catkin_package() include dir 'include/eigen3' does not exist relative to '/home/<user>/moveIt_ws/src/moveit/moveit_kinematics'
```
### Backtrace or Console output

https://gist.github.com/YuehChuan/35ab67d09415e0b9a9077a3b218b593d

solution: edit package.xml  CMakeLists.txt under moveit/moveit_kinematics/
https://github.com/YuehChuan/moveit/blob/indigo-devel/moveit_kinematics/CMakeLists.txt#L35

YuehChuan, 2017-03-12:
solution: edit package.xml CMakeLists.txt under moveit/moveit_kinematics/
https://github.com/YuehChuan/moveit/blob/indigo-devel/moveit_kinematics/CMakeLists.txt#L35

closed 2017-03-12 by YuehChuan

davetcoleman, 2017-03-12:
Do you mind opening a pull request for this so that future MoveIt! users of the Indigo branch do not have the same problem? Thanks!

YuehChuan, 2017-03-13:
Wow, thanks for reviewing this ! Already opened. 
Not sure whether is because of my computer environment cause the issue.  
I would like to try another fresh-install indigo again. Thanks!