moveit/moveit #473: Pyassimp workaround

jlack1987, 2017-03-17:
### Description

I am having a problem with pyassimp and am not sure how to workaround it. The issue is posted [here](https://bugs.launchpad.net/ubuntu/+source/assimp/+bug/1657374?comments=all) and I saw the other issue posted to moveit issue board that kinda talked about it, but it was closed and I am still having the problem and didn't see a workaround posted. 

I know the problem is really with pyassimp and not MoveIt, so i'm just wondering how others deal with it since I don't see any movement or anything from the maintainers of pyassimp and I feel like everybody on Ubuntu 16.04 has to be having the same problem.

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build? Binary
* If binary, which release version? 0.9.4

### Steps to reproduce
If pyassimp is in the import path anywhere it will fail. 

### Backtrace or Console output
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-1-652a73ea0321> in <module>()
----> 1 import pyassimp

/usr/lib/python2.7/dist-packages/pyassimp/__init__.py in <module>()
----> 1 from .core import *

/usr/lib/python2.7/dist-packages/pyassimp/core.py in <module>()
     27 from .formats import available_formats
     28 
---> 29 class AssimpLib(object):
     30     """
     31     Assimp-Singleton

/usr/lib/python2.7/dist-packages/pyassimp/core.py in AssimpLib()
     31     Assimp-Singleton
     32     """
---> 33     load, load_mem, release, dll = helper.search_library()
     34 _assimp_lib = AssimpLib()
     35 

ValueError: need more than 3 values to unpack
```

130s, 2017-03-17:
It's actually already worked around in MoveIt! version `0.9.5` for Kinetic (see [changelog](http://docs.ros.org/kinetic/changelogs/moveit/changelog.html), or [this ticket](https://github.com/ros-planning/moveit/issues/86)), and binary of new packages are already ready for the public release. I've [requested for the release](https://discourse.ros.org/t/new-packages-for-kinetic-2017-03-05/1430/3?u=130s).

If you're in rush, you can download binaries from the testing repository called ["shadow-fixed"](http://wiki.ros.org/ShadowRepository).

I'll close this but i there are more things we need to discuss for this feel free to post comments and reopen.

closed 2017-03-17 by 130s

130s, 2017-03-17:
> I'll close this but i there are more things we need to discuss for this feel free to post comments and reopen.

Actually it'd be better if we won't distribute info on the same issue on multiple tickets. So please use https://github.com/ros-planning/moveit/issues/86 for further discussion if needed.