moveit/moveit #475: moveit rviz plugin error

sewawa, 2017-03-22:
ubuntu-16.04  ROS kinetic

when i launch:     roslaunch lwr_moveit_config demo.launch
i get :
[ERROR] [1490149701.607062981]: PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load.  Error: Failed to load library /opt/ros/kinetic/lib/libmoveit_motion_planning_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.2: Unable to open shared object file: no file or directory)
I look for opt/kinetic/lib, the file  (libopencv_imgproc3.so.3.2) isn't exist. Where i can find this file, or i need to install something, apt-get install [what?]

davetcoleman, 2017-03-23:
libopencv is outside the domain of moveit. this appears to be an issue with your particular installation or catkin workspace setup

gortium, 2017-03-24:
My moveit setup used to work, now i got this:

[ERROR] [1490316164.810097258, 4.273000000]: Exception while loading planner 'ompl_interface/OMPLPlanner': Failed to load library /opt/ros/kinetic/lib/libmoveit_ompl_planner_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libompl.so.12: cannot open shared object file: No such file or directory)
Available plugins: ompl_interface/OMPLPlanner

Maybe it is linked to the same problem ?

v4hn, 2017-03-24:
On Thu, Mar 23, 2017 at 05:54:52PM -0700, Thierry Pouplier wrote:
> Maybe it is linked to the same problem ?

I believe both errors appear because of either incomplete updates
of your system or because of broken local workspaces that have to be rebuild.
Updating your linux distribution and/or doing a complete rebuild of your relevant
workspaces usually fixes such problems.
If you updated ompl/opencv but not the moveit packages or the other way around,
or if you updated your distribution packages and they changed ABI then you get such linking errors.

gortium, 2017-03-25:
purge ros*
update, upgrade, distro-upgrade, install ros-kinetic-desktop-full, install ros-kinetic-moveit. 
build my repository, launch move_group.launch.
Still get this error. 
Something most be wrong..

v4hn, 2017-03-27:
I just did a fresh installation of ubuntu 16.04 and kinetic and do not encounter the problem there.
This is clearly a bug at your end.

`libompl.so` is (ttbomk) the only library installed by osrf that is *not* directly in `/opt/ros/kinetic/lib` but instead in `/opt/ros/kinetic/lib/x86_64-linux-gnu`.
So make sure you have the library in that path and `echo $LD_LIBRARY_PATH` includes `/opt/ros/kinetic/lib/x86_64-linux-gnu` after you sourced your workspace.

closed 2017-05-15 by v4hn

wkoziej, 2017-06-06:
I'm not sure this issue is solved.
I have similar problem on ubuntu xenial (16.04). When I'm trying to launch generated demo I get similar error 

`[ERROR] [1496781514.471921465]: Exception while loading planner 'ompl_interface/OMPLPlanner': Failed to load library /opt/ros/kinetic/lib/libmoveit_ompl_planner_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libompl.so.12: cannot open shared object file: No such file or directory)
Available plugins: ompl_interface/OMPLPlanner
`

So I've try to resolve this:

`$ ldd /opt/ros/kinetic/lib/libmoveit_ompl_planner_plugin.so | grep not
	libompl.so.12 => not found
`

It seems to me that libompl.so.12 is only in artful version of ubuntu: https://packages.ubuntu.com/search?keywords=libompl12.

Am I wrong?

willcbaker, 2017-06-08:
@wkoziej can confirm, fresh install on 16.04 getting same error

wkoziej, 2017-06-10:
I wanted to try this problem on fresh ROS docker instalation. After that it showed up that my original instalation started working. OMPL library was found (not in the system directories but in the ROS libs):
`ldd /opt/ros/kinetic/lib/libmoveit_ompl_planner_plugin.so | grep libom
`
`
libompl.so.12 => /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12 (0x00007f8e92017000)
` 
I'm confused but happy.

xxx09, 2017-08-08:
@sewawa I have the same problem with u, I upgrade opencv (which i import cv2 in python and cv2.__version__ is 3.1.0-dev) in Synaptic to 3.2, add library path as @v4hn mentioned, and it solves the problem.

0Nel, 2019-08-02:
I had the same Error message on Ubuntu 18.04 with ros melodic and moveIt. After searching every possible post about it and almost reinstalling ros entirely I was able to fix this issue simply with:
```bash
sudo apt-get update
sudo apt-get dist-upgrade
```
Hope this will resolve the issue for other users as well!
I am not sure whether it was necessary, but it is probably a good idea to rebuild and resource your catkin workspace afterwards.

kokokoharu, 2019-10-11:
> I just did a fresh installation of ubuntu 16.04 and kinetic and do not encounter the problem there.
> This is clearly a bug at your end.
>
> `libompl.so` is (ttbomk) the only library installed by osrf [...]

This is good. I was having the similar issue and after I included /opt/ros/kinetic/lib/x86_64-linux-gnu in .bashrc, everything becomes fine.

beevabeeva, 2020-02-21:
@0Nel Thanks! This worked for me.

Targetarm-Ankit, 2022-09-19:
Hello, 
I simply update my linux system using the following command and it worked for me 
sudo apt update
sudo apt upgrade
In case it does not work try installing moveit-simple-controller-manager using  sudo apt install ros-<ros-distro>-moveit-simple-controller-manager 
sudo apt update 
sudo apt upgrade
close your terminal and reopen. It should work now