moveit/moveit #477: Release Kinetic 0.9.6 and IJ

130s, 2017-03-24:
- New release requested: https://github.com/ros-planning/moveit/pull/474#issuecomment-288656958
- Last release: [20170308](https://github.com/ros-planning/moveit/releases/tag/0.9.5)
- Num of commits since the last release: [10](https://github.com/ros-planning/moveit/compare/kinetic-devel@%7B03-08-17%7D...kinetic-devel@%7B03-23-17%7D)
- Last public sync on the buildfarm: [20170322](https://discourse.ros.org/t/new-packages-for-indigo-jade-and-kinetic-2017-03-22/1546)

Honestly with the smaller amount of commits, making a release is less stressful. And I personally would like to take release requests, esp. bugfix, seriously.

If no objection I can make release by the end of March (i.e. in a few days), so that we'll probably have ample time to test from shadow-fixed before the next public sync (which is now [back to 2-week? interval routine](https://discourse.ros.org/t/preparing-for-indigo-sync-2017-03-20/1530/1)).

v4hn, 2017-03-24:
> And I personally would like to take release requests, esp. bugfix, seriously.

I fully agree. Please keep in mind though that with each release that is synced to our users in kinetic, they have to recompile their whole moveit-related workspace because we change sonames and linking fails.
So releasing *too* often is also annoying to users. At least this is the case until we allow for patch releases without ABI breakage.

davetcoleman, 2017-03-26:
Thanks @130s, please proceed

130s, 2017-04-03:
@v4hn noted. Thanks for the input.

Prerelease test running locally failed. I'm re-running now in case there was just a temporary hiccup. I'm not sure this error log means:
```
:
[ROSUNIT] Outputting test results to /tmp/catkin_workspace/test_results/moveit_ros_planning_interface/rostest-test_python_move_group.xml
[ INFO] [1491259066.494305427]: Loading robot model 'fanuc'...
[ INFO] [1491259066.564783379]: Loading robot model 'fanuc'...
Starting context monitors...
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

testpython_move_group ... FAILURE!
FAILURE: Test Fixture Nodes ['joint_state_publisher', 'robot_state_publisher'] failed to launch
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 121, in fn
    self.assert_(not failed, "Test Fixture Nodes %s failed to launch"%failed)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

[testpython_move_group][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testpython_move_group

:
[ROSUNIT] Outputting test results to /tmp/catkin_workspace/test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml
[ INFO] [1491259068.081815630]: Loading robot model 'fanuc'...
[ INFO] [1491259068.163542683]: Loading robot model 'fanuc'...
Starting context monitors...
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

testrobot_state_update ... FAILURE!
FAILURE: Test Fixture Nodes ['joint_state_publisher', 'robot_state_publisher'] failed to launch
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 121, in fn
    self.assert_(not failed, "Test Fixture Nodes %s failed to launch"%failed)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

[testrobot_state_update][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testrobot_state_update

:
[ROSUNIT] Outputting test results to /tmp/catkin_workspace/test_results/moveit_ros_planning_interface/rostest-test_cleanup.xml
[ INFO] [1491259069.722681457]: Loading robot model 'fanuc'...
[ INFO] [1491259069.788289848]: Loading robot model 'fanuc'...
Starting context monitors...
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

testcleanup_tests ... FAILURE!
FAILURE: Test Fixture Nodes ['joint_state_publisher', 'robot_state_publisher'] failed to launch
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 121, in fn
    self.assert_(not failed, "Test Fixture Nodes %s failed to launch"%failed)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

[testcleanup_tests][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testcleanup_tests
:
Test results of underlay workspace

Invoking: catkin_test_results /tmp/prtest_moveit_kinetic_xenial/catkin_workspace/test_results --all

:
moveit_ros_planning_interface/rostest-test_cleanup.xml: 1 tests, 0 errors, 1 failures, 0 skipped
moveit_ros_planning_interface/rostest-test_python_move_group.xml: 1 tests, 0 errors, 1 failures, 0 skipped
moveit_ros_planning_interface/rostest-test_robot_state_update.xml: 1 tests, 0 errors, 1 failures, 0 skipped
:
Summary: 88 tests, 0 errors, 3 failures, 0 skipped
```

130s, 2017-04-03:
@v4hn noted. Thanks for the input.

Prerelease test running locally passed for IJ but failed for K. I'm re-running now in case there was just a temporary hiccup. I'm not sure this error log means:
```
:
[ROSUNIT] Outputting test results to /tmp/catkin_workspace/test_results/moveit_ros_planning_interface/rostest-test_python_move_group.xml
[ INFO] [1491259066.494305427]: Loading robot model 'fanuc'...
[ INFO] [1491259066.564783379]: Loading robot model 'fanuc'...
Starting context monitors...
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

testpython_move_group ... FAILURE!
FAILURE: Test Fixture Nodes ['joint_state_publisher', 'robot_state_publisher'] failed to launch
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 121, in fn
    self.assert_(not failed, "Test Fixture Nodes %s failed to launch"%failed)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

[testpython_move_group][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testpython_move_group

:
[ROSUNIT] Outputting test results to /tmp/catkin_workspace/test_results/moveit_ros_planning_interface/rostest-test_robot_state_update.xml
[ INFO] [1491259068.081815630]: Loading robot model 'fanuc'...
[ INFO] [1491259068.163542683]: Loading robot model 'fanuc'...
Starting context monitors...
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

testrobot_state_update ... FAILURE!
[... 52 lines cut ...]

[ROSTEST]-----------------------------------------------------------------------

[testcleanup_tests][failed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 0
 * FAILURES: 1

ERROR: The following tests failed to run:
 * testcleanup_tests
:
Test results of underlay workspace

Invoking: catkin_test_results /tmp/prtest_moveit_kinetic_xenial/catkin_workspace/test_results --all

:
moveit_ros_planning_interface/rostest-test_cleanup.xml: 1 tests, 0 errors, 1 failures, 0 skipped
moveit_ros_planning_interface/rostest-test_python_move_group.xml: 1 tests, 0 errors, 1 failures, 0 skipped
moveit_ros_planning_interface/rostest-test_robot_state_update.xml: 1 tests, 0 errors, 1 failures, 0 skipped
:
Summary: 88 tests, 0 errors, 3 failures, 0 skipped
```

130s, 2017-04-04:
Ok the prerelease test error on Kinetic is consistently happening for the last 2 runs in the last 3 hours.

I found that the binary of `robot_state_publisher` is [failing to build](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=state_publisher), but [prerelease test installs dependency from shadow-fixed](http://wiki.ros.org/regression_tests#What_gets_tested.3F), which is not failing now. So I don't know why this error happens.

130s, 2017-04-04:
- Indigo https://github.com/ros/rosdistro/pull/14449
- Jade https://github.com/ros/rosdistro/pull/14448
- Kinetic This issue https://github.com/ros-planning/moveit/issues/477#issuecomment-291304697 still occurs during prerelease test.

130s, 2017-04-07:
[This](https://discourse.ros.org/t/packages-ros-org-is-out-of-sync/1624/1) might be related to the strange error above. I'll re-run prerelease test later today.

130s, 2017-04-07:
> [This](https://discourse.ros.org/t/packages-ros-org-is-out-of-sync/1624/1) might be related to the strange error above. I'll re-run prerelease test later today.

Nope, still NG. I still have no idea why this happens.

> ```
> FAILURE: Test Fixture Nodes ['joint_state_publisher', 'robot_state_publisher'] failed to launch
> ```

130s, 2017-04-08:
We may need thishttps://github.com/ros-planning/moveit_resources/pull/14 from the released deb. I'll make a release there (even though why that's not the issue on Indigo and Jade isn't known to me).

130s, 2017-04-08:
We may need this https://github.com/ros-planning/moveit_resources/pull/14 from the released deb. I'll make a release there (even though why that's not the issue on Indigo and Jade isn't known to me).

130s, 2017-04-08:
https://github.com/ros-planning/moveit/issues/477#issuecomment-292693992 did the trick. Prerelease test passed.
Bloom said the release artifact kept in the release repo is 0.9.5-1 while I expect 0.9.6. I'll look early next week.

130s, 2017-04-11:
K 0.9.6 https://github.com/ros/rosdistro/pull/14515

closed 2017-04-11 by 130s