moveit/moveit #479: setFromIK results in incorrect joint positions

broesdecat, 2017-03-28:
### Description
I have a rosnode that takes 2 poses as input, which represent goals of the end effectors of the 2 robot arms in my scene. Originally, my rosnode called setPoseTarget for each of the end effectors, in the movegroup that contains both robots. This worked correctly, except that planning failed quite often with OMPL.
Hence, I decided to try STOMP (through industrial-moveit), but STOMP only takes joint values as goals, so I converted the setPoseTarget(pose, link) calls i had into the following code:
`std::map<std::basic_string<char>, double> goal_joint_values;

const std::vector<std::string> &iiwa_joint_names = iiwa_modelgroup->getJointModelNames();
const std::vector<std::string> &ur_joint_names = ur_modelgroup->getJointModelNames();

std::vector<double> iiwa_joint_values;
kinematic_state->setFromIK(iiwa_modelgroup, target_pose1, iiwa_ee_link, 10, 0.1);
kinematic_state->copyJointGroupPositions(iiwa_modelgroup, iiwa_joint_values);
for(std::size_t i = 1; i < iiwa_joint_names.size()-1; ++i) // Indexing hack for now, to skip fixed joints
	goal_joint_values[iiwa_joint_names[i]] = iiwa_joint_values[i];

std::vector<double> ur_joint_values;
kinematic_state->setFromIK(ur_modelgroup, target_pose2, ur_ee_link, 10, 0.1);
kinematic_state->copyJointGroupPositions(ur_modelgroup, ur_joint_values);
for(std::size_t i = 1; i < ur_joint_names.size()-1; ++i) // Indexing hack for now, to skip fixed joints
	goal_joint_values[ur_joint_names[i]] = ur_joint_values[i];

group->setJointValueTarget(goal_joint_values);`

However, it turns out the joint positions I get back from setFromIK do not correspond to the target poses, and the resulting plan is hence wrong.
From the doxygen and documentation, I do not immediately see where I should look for the issue.

For example, if I order an iiwa LBR with base at the root of the coordinate system to move to x=0.2, y=0.2, z=1.0, it ends up at x=0.42, y=-0.39, z=2.0.

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04 through Docker
* Binary build, 0.7.6-Alpha

jrgnicho, 2017-03-30:
You might want to check the the coordinate frame of your kinematic model matches the frame of your cartesian pose.  Plus you should check the the **setFromIK** call succeeded.  See [here](http://docs.ros.org/jade/api/moveit_core/html/classmoveit_1_1core_1_1RobotState.html#ab816880027ef7e63bbdef22a0497cc78) for more info on this method.

broesdecat, 2017-04-04:
Thx for the suggestions.
I investigated them and simplified my scenario further, but am still coming up empty...
I was already checking that setFromIK succeeded.
The coordinate frame of the model (robotmodel->getModelFrame()) matches the one of the pose (both are the world root frame). However, looking at the results, I do get the feeling that some kind of transform is involved, but any suggestions of where it could be are welcome. Below, I'll post my current code, which is a subscriber that takes cartesian pose messages as [x,y,z,rotx,roty,rotz]. In all my tests, even keeping rotational identity results in issues. 

```
std::string iiwa_ee_link = "iiwa_link_7"; // TODO should be iiwa_link_ee, but then results are even worse.
std::string linkgroup = "manipulator";

ros::NodeHandle *node_handle = NULL;
moveit::planning_interface::MoveGroup *group = NULL;
moveit::planning_interface::PlanningSceneInterface *planning_scene_interface = NULL;
robot_state::RobotStatePtr kinematic_state;
robot_state::JointModelGroup* iiwa_modelgroup;
robot_model::RobotModelPtr kinematic_model;
tf::TransformListener *tf_listener = NULL;

void reply(const std_msgs::Float64MultiArray& msg){
	moveit_msgs::DisplayTrajectory display_trajectory;
		
	// Retrieve position from message
	geometry_msgs::Pose target_position1; // Expressed in group->getPlanningFrame()
	target_position1.position.x = msg.data[0];
	target_position1.position.y = msg.data[1];
	target_position1.position.z = msg.data[2];
	target_position1.orientation = tf::createQuaternionMsgFromRollPitchYaw(msg.data[3], msg.data[4], msg.data[5]);
	
	// Compute IK
	Eigen::Affine3d target_eig_pose1;
	tf::poseMsgToEigen(target_position1, target_eig_pose1);	
	kinematic_state->setToIKSolverFrame(target_eig_pose1, group->getPlanningFrame()); // Convert to robotmodel frame if needed.
	bool found_ik = kinematic_state->setFromIK(iiwa_modelgroup, target_eig_pose1, iiwa_ee_link, 10, 0.1);
	if(!found_ik)
		return;
	
	// Retrieve joint values map
	std::map<std::string, double> goal_joint_values;
	std::vector<double> iiwa_joint_values;
	kinematic_state->copyJointGroupPositions(iiwa_modelgroup, iiwa_joint_values);
	const std::vector<std::string> &iiwa_joint_names = iiwa_modelgroup->getJointModelNames();
	for(std::size_t i = 1; i < iiwa_joint_names.size()-1; ++i) { // TODO cleanly remove the fixed joints (first and last) from the list
		goal_joint_values[iiwa_joint_names[i]] = iiwa_joint_values[i];
	}
	
	// Set plan request data
	group->setJointValueTarget(goal_joint_values);
//	group->setPoseTarget(target_position1, iiwa_ee_link); // This works perfectly, but not supported by STOMP...
	
	robot_state::RobotState start_state(*group->getCurrentState());
	group->setStartState(start_state);	
	group->setGoalOrientationTolerance(0.01);
	group->setGoalPositionTolerance(0.01);

	// Plan and execute
	moveit::planning_interface::MoveGroup::Plan my_plan;
	bool success = group->plan(my_plan);
	group->execute(my_plan);
}

int main(int argc, char **argv){
	ros::init(argc, argv, "moveit_iiwa_listening");
	ros::NodeHandle n;
	node_handle = &n;
	
	robot_model_loader::RobotModelLoader robot_model_loader("robot_description");
	kinematic_model = robot_model_loader.getModel();
	kinematic_state.reset(new robot_state::RobotState(kinematic_model));
	kinematic_state->setToDefaultValues();
	iiwa_modelgroup = kinematic_model->getJointModelGroup(linkgroup);
	
	ros::AsyncSpinner spinner(1);
	spinner.start();
	
	moveit::planning_interface::PlanningSceneInterface psi;
	planning_scene_interface = &psi;
	moveit::planning_interface::MoveGroup g(linkgroup);
	group = &g;
	
	ros::Subscriber sub = node_handle->subscribe("/iiwa/targetcartpose", 1000, reply);

	ros::spin();
	
	ros::shutdown();
	return 0;
}
```

v4hn, 2017-04-04:
Well, choosing a different ee link because "the result is better" is not a good idea unless the results are actually *right*.
Also `setToIKSolverFrame` should not be used there because it is already called internally by `setFromIK`. Although I don't know whether this makes a difference in your case.

What IK solver do you use? Is there a moveit config package for that around that I could check?

PS: why would you pass along float multiarrays when you could use `geometry_msgs::PoseStamped` messages instead...

broesdecat, 2017-04-04:
Some quick answers to @v4hn:
- Changing the link was only part of my debugging efforts, I indeed prefer correct results ^^
- `setToIKSolverFrame` makes no difference, but I added it initially because from the docs it was not clear that it was not needed.
- I tried various IK solvers (tracIK, KDL, LMA)
- For config, I am using the one in the [iiwa stack repo](https://github.com/SalvoVirga/iiwa_stack) (only removed the group tag in demo.launch because the RobotModelLoader doesn't like it.
- On the float multiarray messages: because I'm still finding my way around ROS and that was the easiest hack for now :) Should indeed become proper messages afterwards.

jrgnicho, 2017-04-06:
@broesdecat I see that you are calling the methods 	
```
group->setGoalOrientationTolerance(0.01);
group->setGoalPositionTolerance(0.01);
```
which are used to set tolerances when you have a cartesian pose goal but in your case you don't need to do so as your goal is defined in joint space. You can use [setJointGoalTolerances()](http://docs.ros.org/indigo/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#a3672b7f6c7a3f869c5a0458062117d98).
Furthermore, rather than solving the IK first and setting the joint values from the solution you can set a joint pose goal from a cartesian pose with this method [setApproximateJointValueTarget ](http://docs.ros.org/indigo/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#a467ed1d9be5539533ec92c76fb3b5e71)

broesdecat, 2017-04-06:
Thx @jrgnicho for the alternative API!
It indeed does not seem to suffer from the above-reported issue, and a scenario with a single robotarm now works perfectly.

However, my full scenario has multiple robotarms, which gets me into another issue: setApproximateJointValueTarget requires that the moveGroup as a whole has an IK solver. However, with multiple robotarms, I can only assign an IK solver to the subgroups (which are chains), and not the parent group (as far as I am aware). Any idea how to work around this? Someone already hit [the same issue](https://groups.google.com/forum/#!topic/moveit-users/cMlQwTADqXI) some time ago, unanswered...

jrgnicho, 2017-04-06:
@broesdecat I see.  At the moment MoveIt! only supports serial chain manipulator kinematics and motion planning.

v4hn, 2017-04-07:
On Thu, Apr 06, 2017 at 11:08:51AM -0700, Jorge Nicho wrote:
> At the moment MoveIt! only supports serial chain manipulator kinematics and motion planning.

This is not entirely true. There is just no officially released kinematics plugin that deals with
non-chain setups, but MoveIt! + Motion Planning supports them quite nicely if the IK plugin
can work with them. Our group plans to release such a plugin in the next half a year.

But this doesn't resolve the original problem in this thread and if this involves a bug on MoveIt's
side of things, then this should be fixed too :)