moveit/moveit #481: No kinematic solvers

BasB1, 2017-04-05:
### Description

Good day,

I want to implement moveit! on the Kuka Youbot.
To create the package configuration i used the setup assistant.
In the planning groups i cannot select any kind of kinematic solver.
The drop down just says "none".

I tried to make a package on an other PC with the solvers and copied it to the youbots PC.
When i run the trajectory execution it gives a error that it cannot find the specified kinematic solver.

I tried googling this problem, reinstalled moveit through source, git clone etc. but i feel like i'm out of options.

### Your environment
* ROS Distro: indigo
* OS Version: Ubuntu 14.04
* Source/binary, tried them all

### Steps to reproduce
I cant tell you how to reproduce the problem. ROS might be the problem?

### Expected behaviour
I can chose the kinematic solver.

### Actual behaviour
I can not chose the kinematic solver.
Just the option "none".

### Backtrace or Console output
n/a

v4hn, 2017-04-07:
make sure you have the `ros-indigo-moveit-kinematics` and the `ros-indigo-track-ik` packages installed on your system.
The assistant only lists plugins that are installed.

BasB1, 2017-04-07:
Thank you. That did it.

closed 2017-04-07 by v4hn

davetcoleman, 2017-04-10:
This will help future users with this problem: https://github.com/ros-planning/moveit/pull/485