moveit/moveit #486: Release ROS Lunar MoveIt!

davetcoleman, 2017-04-10:
It looks like we are approaching MoveIt!'s turn to release into Lunar, [geometric_shapes](https://github.com/ros-planning/geometric_shapes/issues/65) is now a blocker for URDFDom... @130s do you have cycles to release that soon?

As discussed in our last maintainer meeting, our plan is to release lunar out of the current ``kinetic-devel`` branch for as long as possible.

Subject for discussion: as soon as we release MoveIt! into Lunar, I'd support reducing @130s' effort by no longer releasing updates into the Jade branch, including cherry-picks

130s, 2017-04-13:
+1 for the kindness @davetcoleman :), and for dropping Jade support soon.
Also +1 for being kind for all MoveIt! users as much as possible. I/we can make a release into Lunar soon, hopefully before its official release (I assume that asking around maintainers of all reverse-dependent packages to make releases might take some time).

130s, 2017-04-13:
Asking around some direct dependencies for the release into Lunar. I'm sure there will be more once I look into non-direct dependency:
- RViz https://github.com/ros-visualization/rviz/issues/1090
- octomap_msgs https://github.com/OctoMap/octomap_msgs/issues/8
- object_recognition_msgs https://github.com/wg-perception/object_recognition_msgs/issues/5
- warehouse_ros https://github.com/ros-planning/warehouse_ros/issues/36
- moveit_msgs @130s
- moveit_resources @130s
- ompl https://bitbucket.org/ompl/ompl/issues/318/release-into-ros-lunar
- srdfdom @130s

mikaelarguedas, 2017-04-14:
> +1 for the kindness @davetcoleman :), and for dropping Jade support soon.

Jade will be EOL at the same time as Lunar is released and packages won't be built for it as of end of May.

> Also +1 for being kind for all MoveIt! users as much as possible. I/we can make a release into Lunar soon, hopefully before its official release

If this is possible, that would be really awesome, thanks for pounding on that!

130s, 2017-04-17:
- [x] FCL, which is used to be a ROS package but now standalone, so might be worth being noted.
  - Looks like for all Ubuntu X (0.3.2), [Y](http://packages.ubuntu.com/source/yakkety/fcl), [Z](http://packages.ubuntu.com/source/zesty/fcl) (0.5.0 for both) that [Lunar supports](http://www.ros.org/reps/rep-0003.html#lunar-loggerhead-may-2017-may-2019) it's already released.
  - I can't remember if Debian is now required (REP says "recommendation" but I remember `bloom` wants a released package on Debian)

mikaelarguedas, 2017-04-17:
Debian is a "recommended support", but we try to get as many packages as possible released into debian.

Is there any blocker to get Moveit! working on debian stretch ? looks like libfcl is the same as in Y and Z (0.5.0) on [debian stretch](https://packages.debian.org/stretch/libfcl-dev) so updating the rosdep key should be all you need.

130s, 2017-04-19:
> looks like libfcl is the same as in Y and Z (0.5.0) on [debian stretch](https://packages.debian.org/stretch/libfcl-dev) so updating the rosdep key should be all you need.

@mikaelarguedas Thanks for clarification. Looks like the necessary `fcl` is already available and [registered on rosdistro](https://github.com/ros/rosdistro/blob/89acff9ab83bbf336a25b7559ebf74ac6a49d82c/rosdep/base.yaml#L1457).

> Is there any blocker to get Moveit! working on debian stretch ? 

I'm not sure since I've been holding from running `bloom` for Lunar knowing that all the dependency hasn't been cleared yet.

130s, 2017-04-25:
- [ ] household_objects_database_msgs https://github.com/ros-interactive-manipulation/household_objects_database_msgs/issues/5

130s, 2017-04-25:
- [ ] manipulation_msgs https://github.com/ros-interactive-manipulation/manipulation_msgs/issues/10

130s, 2017-05-10:
- [ ] python-assimp for Ubuntu YZ https://github.com/ros/rosdistro/pull/14963

130s, 2017-05-11:
Running prerelease tests locally:
- Lunar-Xenial passed ([full log](https://drive.google.com/open?id=0Bw_hbVS3wTLydmtDbl8xWlN5Q2s).
- Lunar-Zesty failed for missing rosdep keys ([full log](https://drive.google.com/open?id=0Bw_hbVS3wTLybWJpcUx2ZExCWlk))

130s, 2017-05-11:
- [ ] libglew-dev --> https://github.com/ros/rosdistro/pull/14963

130s, 2017-05-11:
Done https://github.com/ros/rosdistro/pull/14966

mikaelarguedas, 2017-05-11:
FYI: you'll need to rerun bloom once https://github.com/ros/rosdistro/pull/14967 has been merged, for moveit to compile on Yakkety, Zesty and Debian Stretch

130s, 2017-05-11:
> FYI: you'll need to rerun bloom once ros/rosdistro#14967 has been merged, for moveit to compile on Yakkety, Zesty and Debian Stretch

Thanks for the heads-up @mikaelarguedas. I'm curious if there was any issue? If so where can I see it indicated?

I admit that after https://github.com/ros-planning/moveit/issues/486#issuecomment-300648476 where I once ran prerelease test against `Zesty`, I resolved rosdep keys but then forgot to re-run prerelease test but instead I just proceeded and confirmed that `bloom` was able to resolve all `rosdep` keys. Clearly this doesn't mean that I confirmed MoveIt! building on Z (as well as Y).

Also I thought I/we confirmed about FCL situation in a previous post https://github.com/ros-planning/moveit/issues/486#issuecomment-294607434, so if that wasn't sufficient I want to know what else needed to be done back then.

mikaelarguedas, 2017-05-11:
As far as I understand, fcl 0.5.x was required for kinetic. At that time wily and xenial were providing 0.3.x so we had to package libfcl 0.5 ourselves for these Ubuntu distributions http://repositories.ros.org/ubuntu/main/pool/main/f/fcl/. Now that fcl 0.5 is upstream for yakkety, zesty and stretch, we will use the upstream version. To do so we need to update the rosdep key to map to libfcl-dev for these distributions.
> Also I thought I/we confirmed about FCL situation in a previous post #486 (comment), so if that wasn't sufficient I want to know what else needed to be done back then.

The only thing we could have done was to update the rosdep keys as suggested in https://github.com/ros-planning/moveit/issues/486#issuecomment-294629339

> I admit that after #486 (comment) where I once ran prerelease test against Zesty, I resolved rosdep keys but then forgot to re-run prerelease test but instead I just proceeded and confirmed that bloom was able to resolve all rosdep keys. Clearly this doesn't mean that I confirmed MoveIt! building on Z (as well as Y).

Bloom passed because the rosdep entry "exist" in the rosdep database but the jobs fail because the package it refers to doesn't exist. What you did was adding the rosdep keys that did not "exist" in the rosdep database but it didn't mean that the ones that existed were accurate.

> I'm curious if there was any issue? If so where can I see it indicated?

Several jobs failed after the release because of the unability to install libfcl-0.5-dev on these platforms. 
A good way to get notified when a package fails is to be listed as a maintainer of the package. Apparently the recipients of the failure notification email were:
* dave@dav.ee,
* me@v4hn.de,
* mferguson@fetchrobotics.com

Another way (less convenient) is to look at the status pages: e.g. http://repositories.ros.org/status_page/ros_lunar_default.html?q=moveit_core. This shows you that the job passed only on xenial but failed on yakkety and zesty. By clicking on the red box under Y64 you are redirected to the failing job.

Hope this helps

mikaelarguedas, 2017-05-11:
@130s FYI: https://github.com/ros/rosdistro/pull/14967 has been merged

130s, 2017-05-11:
- ompl 
  - [x] https://bitbucket.org/ompl/ompl/issues/318/release-into-ros-lunar --> https://github.com/ros/rosdistro/pull/14958
  - [ ] Build issue on armhf https://github.com/ros-gbp/ompl-release/issues/7

130s, 2017-05-11:
> FYI: ros/rosdistro#14967 has been merged

@mikaelarguedas Does it make sense to `bloom` without OMPL issue https://github.com/ros-gbp/ompl-release/issues/7 not resolved yet? If so I can certainly re-bloom.

mikaelarguedas, 2017-05-11:
I think it does given that armhf users is less that 3% of the user base. Having moveit working for the remaining 97% is sensible IMO.

130s, 2017-05-11:
Ok, thanks for the cheerup @mikaelarguedas :)

Hm, I wanted to be cautious so ran a prerelease test against `Zesty`, then it failed ([full log](https://drive.google.com/open?id=0Bw_hbVS3wTLyNFo0d29FMWM2ZWM)). I'm just re-running it locally now.

```
:
==> cmake /tmp/catkin_workspace/src/moveit/moveit_ros/visualization -DCATKIN_DEVEL_PREFIX=/tmp/catkin_workspace/devel_isolated/moveit_ros_visualization -DCMAKE_INSTALL_PREFIX=/tmp/catkin_workspace/install_isolated -DCATKIN_SKIP_TESTING=1 -G Unix Makefiles in '/tmp/catkin_workspace/build_isolated/moveit_ros_visualization'
:
==> make -j1 in '/tmp/catkin_workspace/build_isolated/moveit_ros_visualization'
Scanning dependencies of target moveit_rviz_plugin_render_tools_automoc
[  2%] Automatic moc for target moveit_rviz_plugin_render_tools
Generating moc source moveit_rviz_plugin_render_tools_automoc.dir/moc_trajectory_vis_NE6OZUMGTDEUX5.cpp
Generating moc compilation moveit_rviz_plugin_render_tools_automoc.cpp
[  2%] Built target moveit_rviz_plugin_render_tools_automoc
Scanning dependencies of target moveit_rviz_plugin_render_tools
[  4%] Building CXX object rviz_plugin_render_tools/CMakeFiles/moveit_rviz_plugin_render_tools.dir/src/render_shapes.cpp.o
In file included from /usr/include/boost/config.hpp:61:0,
                 from /usr/include/boost/detail/workaround.hpp:41,
                 from /usr/include/boost/range/config.hpp:14,
                 from /usr/include/boost/range/begin.hpp:18,
                 from /usr/include/boost/range/functions.hpp:18,
                 from /usr/include/boost/range.hpp:18,
                 from /usr/include/OGRE/OgreIteratorRange.h:48,
                 from /usr/include/OGRE/OgreIteratorWrappers.h:33,
                 from /usr/include/OGRE/OgreRenderQueue.h:32,
                 from /usr/include/OGRE/OgreMovableObject.h:34,
                 from /usr/include/OGRE/OgreSimpleRenderable.h:33,
                 from /opt/ros/lunar/include/rviz/ogre_helpers/point_cloud.h:33,
                 from /tmp/catkin_workspace/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/include/moveit/rviz_plugin_render_tools/octomap_render.h:42,
                 from /tmp/catkin_workspace/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/include/moveit/rviz_plugin_render_tools/render_shapes.h:40,
                 from /tmp/catkin_workspace/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/render_shapes.cpp:37:
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_half<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:265:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(half, 5.000000000000000000000000000000000000e-01, "5.00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:265:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_third<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:266:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(third, 3.333333333333333333333333333333333333e-01, "3.33333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:266:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_twothirds<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:267:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(twothirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666667e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:267:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_two_thirds<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:268:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(two_thirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666667e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:268:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_three_quarters<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:269:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(three_quarters, 7.500000000000000000000000000000000000e-01, "7.50000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
CMakeFiles/Makefile2:3145: recipe for target 'rviz_plugin_render_tools/CMakeFiles/moveit_rviz_plugin_render_tools.dir/all' failed
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'moveit_ros_visualization':
  Command '['/tmp/catkin_workspace/install_isolated/env.sh', 'make', '-j1']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /tmp/catkin_workspace/build_isolated/moveit_ros_visualization && /tmp/catkin_workspace/install_isolated/env.sh make -j1

Command failed, exiting.
# END SUBSECTION
```

mikaelarguedas, 2017-05-12:
Looks like a boost issue to me. Maybe something changed in boost 1.62 that's why the error shows up only on Zesty. I guess you already tried using the suggested flags (`note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes`)?

130s, 2017-05-12:
Hm, I don't have much gears anytime soon to try fixing compilation issues. 

- If someone can take a look, that'll be appreciated.
- Elif no one can, then might as well remove rosdistro entry for now for Lunar until it gets fixed.

Btw just memo (to myself); on buildfarm (not prerelease test) I see the [builds on Y and Z are failing](http://repositories.ros.org/status_page/ros_lunar_default.html?q=moveit) for Lunar (e.g. [moveit_core on May 11](http://build.ros.org/view/Lbin_uZ64/job/Lbin_uZ64__moveit_core__ubuntu_zesty_amd64__binary/2/consoleFull)) but this looks like another issue that was mentioned in https://github.com/ros-planning/moveit/issues/486#issuecomment-300811843 and can be fixed by re-`bloom`ing.

130s, 2017-05-12:
> Hm, I don't have much gears anytime soon to try fixing compilation issues.
>
> -    If someone can take a look, that'll be appreciated.

To raise attention I opened a separate issue https://github.com/ros-planning/moveit/issues/504

130s, 2017-05-12:
Re-`bloom`ed https://github.com/ros/rosdistro/pull/14979, hopefully this fixes one of the issues (regarding FCL https://github.com/ros-planning/moveit/issues/486#issuecomment-300853362).

v4hn, 2017-05-12:
Weird, I didn't hit this problem before, although I use even more current versions of gcc and boost on my system...

It is probably enough to replace [this line](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/visualization/CMakeLists.txt#L4) by `add_compile_options(-std=gnu++11)`, and it is ok to do this here I suppose.
I don't think there are any other packages, but if something else transitively includes the ogre-headers, they will need this too.

Because I can't reproduce the error, I can't test the fix though.

mikaelarguedas, 2017-05-12:
There is a gist [here](https://gist.github.com/mikaelarguedas/7731b4100e7ed78b6aab73f052ffac79) with the prerelease console outputs for Yakkety and Zesty for anyone wanting to investigate the failures. I haven't ran it for Stratch because I expect it to fail exactly the same way as Zesty.
Errors can be reproduced by running the prerelease locally (following instructions of [prerelease.ros.org/lunar](http://prerelease.ros.org/lunar))

v4hn, 2017-05-12:
From the prerelease page:

> you can run in the terminal on a Ubuntu machine

Sorry I'm out, I run Lunar-Linux and don't have access to/time to access a Ubuntu machine in the next couple of days.

I wanted to point out that this could actually be a problem of Ubuntu or Debians packages though.
I run gcc 6.3, boost 1.63 and 1.64 and ogre 1.9.0 over here (pretty much zasty's versions) and don't see the error.

Changing the compile flag should work though.

mikaelarguedas, 2017-05-12:
everything runs inside docker so the platform of your host machine should not matter much as long as it supports docker. I'm testing the fix locally as we speak, I'll update here with the outcome.

de-vri-es, 2017-05-13:
I'm on vacation right now so I only skimmed the issue, but it may be enough
to compile with `-DBOOST_MATH_DISABLE_FLOAT128` (this is from memory). If
it works I prefer that over enabling gcc specific extensions, since that
hinders compatibility with non GNU platforms.

130s, 2017-05-27:
I thought MSA build issue on Ubuntu YZ #506 would block the Lunar release, but for the platforms where binaries were successfully built, it seems we can install, like Ubuntu Xenial.
So, the task in the title "release into Lunar" is basically achieved and we only have a "regression" #506.
I close this ticket, but if that's not desirable don't hesitate to shoot me.
Thanks all who paid attention to this especially @mikaelarguedas

mikaelarguedas, 2017-05-27:
Thanks @130s for following up on this.
I don't think that this ticket should be closed because, IMHO, releasing into a new distro means having binaries building on all platforms for which there is required support (that are [Ubuntu X Y and Z](http://www.ros.org/reps/rep-0003.html#lunar-loggerhead-may-2017-may-2019) on amd64 architecture).

@davetcoleman @de-vri-es @v4hn and moveit maintainers, is there any plan on your end to build on the missing platforms / support the deboostified version of urdfdom? If yes, do you have an idea of where this could fit in your roadmap and how external contributors can help?

---
@130s 
>  it seems we can install, like Ubuntu Xenial.

We usually don't sync packages that don't build on all platforms, I had to sync the day of the official Lunar release despit the fact that moveit was the only repo not compiling at that point, but I think it is still a blocker for the "Lunar release of moveit". I'm not sure what the policy is in the scenario where this still doesnt compile for the next syncs. The possible outcomes are disabling the moveit builds for the other platforms or completely remove moveit from Lunar until it compiles. Maybe @tfoote or @dirk-thomas  can give us more information in that matter.

dirk-thomas, 2017-05-30:
> The possible outcomes are disabling the moveit builds for the other platforms

Since the three Ubuntu distros Xenial, Yakkety, and Zesty are being managed in the same [build file](https://github.com/ros-infrastructure/ros_buildfarm_config/blob/8c60b5f914827a469c88cd273568f8c16a5194b9/lunar/release-build.yaml#L52-L59) they can't be disabled separately. I think we can only highly recommend that the `moveit` teams makes an effort to support all of these distros. If they don't the farm will continue to send notifications for the failing jobs. I don't think there is anything we should do about it.

v4hn, 2017-06-07:
On Fri, May 26, 2017 at 10:14:28PM -0700, Mikael Arguedas wrote:
> is there any plan on your end to build on the missing platforms

Yes. The latest branch should compile on all (reasonable) platforms requested by users.
This clearly includes more current Ubuntu releases and Debian Stretch.

> / support the deboostified version of urdfdom?
> If yes, do you have an idea of where this could fit in your roadmap and how external contributors can help?

I haven't looked into the issue yet, so I can't say anything about the means required.

The standard contributor-guideline "point out the problem and propose a fix" seems to apply here.

davetcoleman, 2017-06-20:
I will address the MSA bug now, sorry for the delay

130s, 2017-07-16:
Looks like all issues are now resolved so finally ready to make full-release into Lunar.

For kinetic-devel, there has only 4 commits since the last release, so making a patch release into Lunar (and Kinetic too?) soon might be good. Or do we want to wait for more commits to be available? Thoughts welcomed.

BryceStevenWilley, 2017-07-19:
I just ran into an issue with compiling [moveit_planners_ompl](https://github.com/ros-planning/moveit/tree/kinetic-devel/moveit_planners/ompl) with OMPL, both from source. 

Specifically, the [constraints library](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/constraints_library.cpp#L308) tries to make a SpaceStorage instance with a const ref, but as of [OMPL 1.3.0](https://github.com/ompl/ompl/commit/51940fb95ba7d3c6b79b75275ed1205a0a96abd4), most of OMPL was changed to pass by value. Not sure why this hasn't been caught by Jenkins yet. My setup is just a catkin wrapper around OMPL and moveit in a single workspace, and the from-source-lib of OMPL is being passed to moveit_planners_ompl. 

This is something that should be addressed on the OMPL side (we should be returning values, not const ref anyway), but it'd require another OMPL release, which would delay Lunar release even more.

v4hn, 2017-07-19:
@BryceStevenWilley Could you open a new issue for this please?

davetcoleman, 2017-07-21:
Since we haven't released a full Lunar release yet, I think we should do that asap now that MSA is fixed

davetcoleman, 2017-07-31:
All checklist items have been completed, I'm going to declare Lunar released and any additional tasks can be added to new issues. Thanks @130s for all your efforts!

closed 2017-07-31 by davetcoleman