moveit/moveit #488: Caching attached collision objects to improve collision checking

Levi-Armstrong, 2017-04-12:
Maintainers,

I have been looking into ways to implement this and it is not as straight forwards as I had hoped. I have came up with two option but before I begin implementing I wanted to get the maintainers input. 

The current caching of the robot collision object occurs during the constructor of the collision robot and the methods that could be used to update the attached object cache are const methods so it is not possible. I describe two option below and I prefer option 1 because it will be minimal changes to the existing code.

Option 1:
Create a non const method like the setPadding (example: cacheAttachedObject & removeCachedAttachedObject) then modify the planning scene class to call this method when object are attached to the robot.

Option 2:
Add similar features that exist in collision world (observer, etc) that would monitor when objects are attached to the robot and update the internal cache.

davetcoleman, 2017-04-15:
I am not familiar enough with that part of the code base to make an informed decision between the two options, but they both sound ok to me. I am reaching out to @isucan to see if he remembers which would be a better place.

Thanks for proceeding with this Levi!

isucan, 2017-04-18:
It has been a while, but Option 2 sounds more robust to me. If you forget to call the new methods from method 1, that likely lead to bugs. That said, if the difference in implementation effort is large, option 1 is ok as well.

v4hn, 2017-04-19:
Sorry @isucan, you probably agree anyway... This code is a mess. -.-
I just spend half a day trying to get my head around it and there are way too many interdependencies.

I guess the basic problem with this is that the attached objects are (and should be) part of the `RobotState`, but collision detection is intentionally left out of this class until now. I assume this is mostly for historical reasons and to keep kinematics and collision checking separate.
The `CollisionRobot` is part of the `PlanningScene` instead and reads the attached objects from a `RobotState` class instance that is passed in for each and every collision check.

Now, you can try to cache attached objects in the `PlanningScene`. But then you will get into trouble when users want to check self-supplied `RobotState`s that are not the current state of the scene.
Actually, this is the way [the current state is handled too](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/src/planning_scene.cpp#L1923).
To me this points towards a (partially shared?) and possibly null_ptr cache for FCL objects in the individual `RobotState` instances, which is neither of your two options @Levi-Armstrong .

v4hn, 2017-10-10:
@Levi-Armstrong mentioned to me that this is on his agenda for WMD. I'll add the appropriate flags :-)

Levi-Armstrong, 2017-10-10:
Possibly, my plan was to add the feature from industrial_moveit collision plugin exposing detailed distance queries within FCL. Afterwards if time permits I will happily address this issue.

v4hn, 2017-10-10:
Cool! 😎

simonschmeisser, 2018-06-05:
Any progress? I just did some profiling and PlanningScene::isStateValid/isStateColliding takes around 1ms in my test scene without attached object and around 25ms with an attached object with 4k faces. 40k faces only lead to around a factor 2 in time, ie ca 50ms

v4hn, 2018-06-12:
Wow, I guess nobody was aware of this cache in quite a while. Thank you for this analysis, Simon!

If I understand you right, this caches the "low level" FCL structures, but not the all the involved objects?

I would be interested in benchmarks that deactivate the cache or replace it by a single cache.

The scenario you describe is rather common by now, so I am not sure the different caches are still justified these days.

simonschmeisser, 2018-06-12:
Well, those classes all have the same name so I'm not 110% sure what the difference between a CollisionGeometrie and a CollisionObject is. I noticed however that we seem to have some sort of congestion in "user code", so the quality of this cache might not be the source of our troubles.

WorldObjects and AttachedObjects seem to differ somehow, that might be the reason for having seperate caches (if it's only the user_pointer_, the pointer to the "MoveIt!"-data structure then we might be able to work around it)

Levi-Armstrong, 2018-06-12:
I remember seeing the caching but something tells me that in the case of attached objects it is not working. Have you by chanced tested to see if it actual finds the attached object in the cache map?

simonschmeisser, 2018-06-12:
I'll try to come up with a way to verify that. I also have some doubts as shape_msgs are used somewhat too often in higher layers of the api and such a serialization would of course destroy the connection.

Levi-Armstrong, 2018-06-12:
Yea that was what I was thinking because the key is the shape pointer so if the shape gets created from a message the key changes. Although for a given planning request you would think it would only need to create the object once and then should be pulling from the cache but I have a suspicion that this is not happening for some reason.

simonschmeisser, 2018-06-13:
Just to add some numbers:

CollisionRobotFCL::constructFCLObject: 25us
 + before "TODO"=robot: 12us
 + after "TODO"=attachedObject: 12us
 +  +  CollisionRobotFCL::getAttachedBodyObjects: 2us
 +  +    + createCollisionGeometry: 1us

so it seems that at least for the checking two states with the same attached object the caching works fine. Still we should investigate having some sort of higher level object cache and passing around mesh-ids in messages instead of whole meshes.

(Side note: it seems the 25ms instead of the usual 0.5-1ms I observe for isStateColliding are valid collision checks, this might be the difference between AABB and actuall mesh to mesh collision tests)

simonschmeisser, 2018-06-25:
Small update: I indentified that the very long times needed for collision checking (isStateValid etc) in our case are due to somehow broad phase collision checking with the octomap not working. So even if the range/volume represented by the octomap (ie the box content) is quite far apart from the attached object or any other mesh it still does a full collision check. This can take several ms depending on mesh size. I will investigate further (and open a new issue or hopefully pr).

v4hn, 2018-07-02:
@simonschmeisser Could you please elaborate a bit more on your explanation? I fail to follow you here...

v4hn, 2019-09-06:
@j-petit Could you please have a quick look at this listing and see whether/what about it is still relevant with your major `CollisionEnv` change?

j-petit, 2019-09-06:
I am pretty sure that this is still relevant as for FCL the new collision
environment is simply wrapping `CollisionRobot` and `CollisionWorld`
together. I will have a closer look when I am back from vacation after the
16th of Sept.