moveit/moveit #490: Segmentation Fault in plan_execution.cpp

johaq, 2017-04-21:
### Description

I am having a segmentation fault in plan_execution.cpp in line 483:

```
00477 void plan_execution::PlanExecution::successfulTrajectorySegmentExecution(const ExecutableMotionPlan* plan,
00478                                                                          std::size_t index)
00479 {
00480   ROS_DEBUG("Completed '%s'", plan->plan_components_[index].description_.c_str());
00481 
00482   // if any side-effects are associated to the trajectory part that just completed, execute them
00483   if (plan->plan_components_[index].effect_on_success_)
00484     if (!plan->plan_components_[index].effect_on_success_(plan))
00485     {
00486       // execution of side-effect failed
00487       ROS_ERROR("Execution of path-completion side-effect failed. Preempting.");
00488       preempt_requested_ = true;
00489       return;
00490     }
00491 
00492   // if there is a next trajectory, check it for validity, before we start execution
00493   std::size_t test_index = index;
00494   while (++test_index < plan->plan_components_.size())
00495     if (plan->plan_components_[test_index].trajectory_ && !plan->plan_components_[test_index].trajectory_->empty())
00496     {
00497       if (!isRemainingPathValid(*plan, std::make_pair<int>(test_index, 0)))
00498         path_became_invalid_ = true;
00499       break;
00500     }
00501 }

```
Debugging the code I found out that plan->size() was 0. 
Does this mean I am using moveit wrong because plan should never be empty at that point? Or is a simple check just missing here?

### My environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04

davetcoleman, 2017-04-25:
Are you suggesting more user feedback be added here for the corner case that a plan of size 0 is provided? Sounds good time.

johaq, 2017-04-25:
Well, moveit crashes when a plan of size zero is provided. So a check is necessary I think. I would suggest something like this: #492

130s, 2017-08-07:
Addressed via #535 (along with #492).

closed 2017-08-07 by 130s