moveit/moveit #493: List available grasp planning packages

130s, 2017-04-25:
Just a placeholder for a potential future task. From https://github.com/ros-planning/moveit/pull/328#issuecomment-257355326

>> add a table with available open source grasp planning packages to that tutorial
>
> I'm pretty interested in such a list myself. I'm only aware of your `moveit_simple_grasps` at the moment. :)

v4hn, 2017-04-25:
Believe it or not, this is on my personal todo list for 5 months now.
I don't see that I will have the time to do that in the near future though, so contributions are more than welcome!

tkelestemur, 2017-06-06:
One package that I've been using : https://github.com/atenpas/gpd and it is pretty good.

davetcoleman, 2017-07-30:
@tkelestemur see https://github.com/ros-planning/moveit/issues/566

yijiangh, 2018-04-03:
I'm wondering what is the grasp planner used in the [kinetic Moveit! pick_place tutorial](https://github.com/ros-planning/moveit_tutorials/blob/3747bd7f05b855c1cde0a5775098749abfd1468d/doc/pr2_tutorials/pick_place/src/pick_place_tutorial.cpp)?

I've read some posts [1] saying that the server for the [grasp planning service](http://docs.ros.org/jade/api/moveit_msgs/html/srv/GraspPlanning.html) is left for users to specify, but there seems to be a "default" grasp planner in this PR2 pick_place_tutorial. I tried to trace down and find this grasp planner but had no success in finding that. Would you mind pointing this out for me?

[1] https://groups.google.com/forum/#!topic/moveit-users/htPXzJXRDK8

v4hn, 2018-04-05:
The grasp pose is hard-coded in the `pick` method.
There is no grasp planner involved in that tutorial.

qyp-robot, 2022-04-26:
Hello, may I ask, I see the message output in the gpd_ros package, the position and attitude, which are in the camera coordinate system; and the transformation to the camera coordinate system is the correct pose information at the end of the robot.
If I grab multiple objects in sequence, and set the final output posture to be 10, I want moveit to execute in sequence, and what format should it be passed to the move_group interface? If you have the relevant code can I refer to it? Thank you so much!
[image]