moveit/moveit #495: Moveit_setup_assistant segmentation fault in libQtGUI.so.4 / libQtTest.so.4

luminize, 2017-04-27:
### Description

Moveit crashes when starting my current robot configuration or `moveit_setup_assistant` for that matter.
Original this question on [answers.ros.org](http://answers.ros.org/question/259490/moveit_setup_assistant-crash-with-segmentation-fault-in-libqtguiso4/)

Up until now I have built Moveit and Rviz locally, but to no avail and I would like to run ROS on my Debian Jessie system. I would appreciate some pointers on how to find the problem and solve this.

### Environment
* ROS Distro: Kinetic
* OS Version: Debian 8.7
* Binary release version: moveit_setup_assistant: 0.9.5-0jessie-20170327-175525-0700
```
machinekit@xw6600:~/catkin_ws/src/moveit$ apt show ros-kinetic-moveit-setup-assistant
Package: ros-kinetic-moveit-setup-assistant
Version: 0.9.5-0jessie-20170327-175525-0700
Maintainer: Dave Coleman <dave@dav.ee>
Installed-Size: 2,794 kB
Depends: libboost-date-time1.55.0, libboost-filesystem1.55.0, libboost-iostreams1.55.0, libboost-program-options1.55.0, libboost-regex1.55.0, libboost-signals1.55.0, libboost-system1.55.0, libboost-thread1.55.0, libc6 (>= 2.14), libconsole-bridge0.2, libgcc1 (>= 1:4.1.1), libgl1-mesa-glx | libgl1, libglu1-mesa | libglu1, liblog4cxx10, libogre-1.9.0, libpocofoundation9, libpython2.7 (>= 2.7), libqt5core5a (>= 5.3.0), libqt5gui5 (>= 5.3.0), libqt5widgets5 (>= 5.2.0~rc1), libstdc++6 (>= 4.9), libtinyxml2.6.2, liburdfdom-model-state0.3, liburdfdom-model0.3, liburdfdom-sensor0.3, liburdfdom-world0.3, libx11-6, libxext6, libyaml-cpp0.5, ros-kinetic-moveit-core, ros-kinetic-moveit-ros-manipulation, ros-kinetic-moveit-ros-move-group, ros-kinetic-moveit-ros-perception, ros-kinetic-moveit-ros-planning, ros-kinetic-moveit-ros-planning-interface, ros-kinetic-moveit-ros-robot-interaction, ros-kinetic-moveit-ros-visualization, ros-kinetic-moveit-ros-warehouse, ros-kinetic-octomap, ros-kinetic-orocos-kdl, libyaml-cpp-dev, ros-kinetic-srdfdom (>> 0.3.1), ros-kinetic-xacro
Homepage: http://moveit.ros.org
Priority: extra
Section: misc
Download-Size: 940 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu/ jessie/main amd64 Packages
Description: Generates a configuration package that makes it easy to use MoveIt!
```
* Source build of Moveit and Rviz after Segfault of mentioned package above
* Source build git commit b0a76b1bec394d28b11d4de5ccccc644ea670d68
* Source build of Rviz: 1f53c35c3a2f21a81b270d92f920d499a82f1c27

### Steps to reproduce
`roslaunch moveit_setup_assistant setup_assistant.launch`

Since my question on answers.ros.org I have [modified (commenting out) the CMakeLists.txt](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_setup_assistant/CMakeLists.txt#L29-L32) so that I know for sure (I hope) that I build `moveit_setup_assistant` for Qt5 since Rviz is also built for Qt5.

### Expected behaviour
startup of the moveit_setup_assistant

### Actual behaviour
crash with creation of core file

### Backtrace and Console output

[terminal output](https://gist.github.com/luminize/fdf8f3613c0a05965237bdcc272a438e)
[gdb backtrace](https://gist.github.com/luminize/3a1075d8ae713ae1544acaba0fdd4893)

v4hn, 2017-05-08:
disRecord proposed on ros answers that this might relate to your opencv linking qt4.
Is this the case?

Did you look at `ldd <relevant library here>` to figure out whether the libraries might transitively link qt4 AND 5?

luminize, 2017-05-09:
I've seen disRecords answer, but for me Croot is no option because I use a Debian installation.
Finding the culprit would mean checking all libraries.

Question 1
If opencv3 is the culprit, then I'd have to build `ros-kinetic-opencv3` and `ros-kinetic-vision-opencv` and finding out about linking against Qt5... Any more packages?

Question 2
The other possibility would be to remove the dependency from moveit. I don't use vision atm.
[a search on github moveit](https://github.com/ros-planning/moveit/search?utf8=%E2%9C%93&q=opencv&type=) shows me that the package introducing the dependency is the perception package.
Would that be a second solution?

Below info on my packages.
I have these packages installed:
```
machinekit@xw6600:~/catkin_ws/src/moveit$ dpkg-query -l | grep opencv
ii  libopencv-calib3d2.4:amd64                   2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Camera Calibration library
ii  libopencv-contrib2.4:amd64                   2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision contrib library
ii  libopencv-core2.4:amd64                      2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision core library
ii  libopencv-features2d2.4:amd64                2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann2.4:amd64                     2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-highgui2.4:amd64                   2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision High-level GUI and Media I/O library
ii  libopencv-imgproc2.4:amd64                   2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Image Processing library
ii  libopencv-legacy2.4:amd64                    2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision legacy library
ii  libopencv-ml2.4:amd64                        2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Machine Learning library
ii  libopencv-objdetect2.4:amd64                 2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Object Detection library
ii  libopencv-photo2.4:amd64                     2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision computational photography library
ii  libopencv-video2.4:amd64                     2.4.9.1+dfsg-1+deb8u1                             amd64        computer vision Video analysis library
ii  python-opencv                                2.4.9.1+dfsg-1+deb8u1                             amd64        Python bindings for the computer vision library
ii  ros-kinetic-opencv3                          3.2.0-4jessie-20170315-013721-0700                amd64        OpenCV 3.x
ii  ros-kinetic-vision-opencv                    1.12.4-0jessie-20170315-024403-0700               amd64        Packages for interfacing ROS with OpenCV, a library of programming functions for real time computer vision.
```

All the libraries:
```
machinekit@xw6600:~/catkin_ws/src/moveit$ sudo find / -name *opencv* | grep so
/opt/ros/kinetic/lib/libopencv_highgui3.so.3.2
/opt/ros/kinetic/lib/libopencv_superres3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_bioinspired3.so
/opt/ros/kinetic/lib/libopencv_xfeatures2d3.so
/opt/ros/kinetic/lib/libopencv_line_descriptor3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_photo3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_ml3.so.3.2
/opt/ros/kinetic/lib/libopencv_text3.so.3.2
/opt/ros/kinetic/lib/libopencv_xphoto3.so
/opt/ros/kinetic/lib/libopencv_optflow3.so.3.2
/opt/ros/kinetic/lib/libopencv_fuzzy3.so.3.2
/opt/ros/kinetic/lib/libopencv_phase_unwrapping3.so
/opt/ros/kinetic/lib/libopencv_photo3.so
/opt/ros/kinetic/lib/libopencv_rgbd3.so.3.2
/opt/ros/kinetic/lib/libopencv_rgbd3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_flann3.so.3.2
/opt/ros/kinetic/lib/libopencv_xobjdetect3.so
/opt/ros/kinetic/lib/libopencv_line_descriptor3.so.3.2
/opt/ros/kinetic/lib/libopencv_calib3d3.so
/opt/ros/kinetic/lib/libopencv_highgui3.so
/opt/ros/kinetic/lib/libopencv_shape3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_highgui3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_video3.so.3.2
/opt/ros/kinetic/lib/libopencv_videostab3.so
/opt/ros/kinetic/lib/libopencv_datasets3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_xfeatures2d3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_flann3.so
/opt/ros/kinetic/lib/libopencv_datasets3.so.3.2
/opt/ros/kinetic/lib/libopencv_saliency3.so
/opt/ros/kinetic/lib/libopencv_bgsegm3.so
/opt/ros/kinetic/lib/libopencv_flann3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_optflow3.so
/opt/ros/kinetic/lib/libopencv_ccalib3.so
/opt/ros/kinetic/lib/libopencv_reg3.so.3.2.0
/opt/ros/kinetic/lib/libopencv_aruco3.so.3.2
[... 85 lines cut ...]
/usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_photo.so.2.4.9
[... cut ...]
/usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_legacy.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4.9
/usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4
```

luminize, 2017-05-09:
just for info, inspecting the first mentionned library shows that it's linked against Qt4:
```
machinekit@xw6600:~/catkin_ws/src$ ldd /opt/ros/kinetic/lib/libopencv_highgui3.so.3.2 | grep Qt
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007fb158f5f000)
	libQtTest.so.4 => /usr/lib/x86_64-linux-gnu/libQtTest.so.4 (0x00007fb158d39000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007fb15884a000)
```

v4hn, 2017-05-09:
So opencv on Debian is the culprit then.

Both your proposed options seem feasible to me.

v4hn, 2017-05-15:
This is not an upstream error and solutions were proposed, so I'll close it.

closed 2017-05-15 by v4hn