moveit/moveit #496: robotStateMsgToRobotState returns success when state message is empty

jrgnicho, 2017-04-30:
### Description

The [robotStateMsgToRobotState](http://docs.ros.org/indigo/api/moveit_core/html/conversions_8cpp_source.html#l00365) returns  **true** (success) even when the robot state message is empty.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Binary build
* If binary, which release version?

### Steps to reproduce
1. Clone Repo and build
This issue can be recreated by cloning this [repo](https://github.com/jrgnicho/industrial_moveit.git) and buildind the **robotMsgToState_issue** branch in a catkin worspace.

2. In a sourced terminal, Run Moveit demo launch file
```
roslaunch stomp_test_kr210_moveit_config demo.launch
```

3. In another sourced terminal, run executable:
```
rosrun test_planning trajectory_node 
```

### Expected behaviour
The terminal from where the demo.launch file was ran will print the received *Start RobotState Message* which isn't populated.  The issue is that the code gets to that [point](https://github.com/jrgnicho/industrial_moveit/blob/robotMsgToState_issue/stomp_moveit/src/stomp_planner.cpp#L564) even though the **robotStateMsgToRobotState** was called to copy the message into a moveit::core::RobotState. 

### Actual behaviour
 If the **robotStateMsgToRobotState** function were to work correctly it should return **false** when it gets and empty RobotState message and the planner code shouldn't be able to reach the point where it prints the message.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

davetcoleman, 2017-05-01:
Seems it could use something like ``if (RobotState.JointState.name.empty()) return false;``

v4hn, 2017-05-08:
Maybe `planning_scene::PlanningScene::isEmpty(RobotState)` instead?
Either way, MultiDOFJointStates should be respected too.

The functions ignore `is_diff`, which makes the state non-empty when set to true.
I'm not sure where this function is used throughout MoveIt and whether there is some use-case that would break if we add the check, but I'm fine with finding this out through a merged pull-request.
All translation functions in this file should behave the same with respect to empty input messages though.

davetcoleman, 2017-06-04:
This can be closed now, correct?

130s, 2017-06-05:
> This can be closed now, correct?

I think so, addressed by #499.

PR #518 (2017-06-07): [Indigo] checks for empty name arrays messages before parsing the robot state message data (cherrypick #499)
Cherry-pick of #499 that fixes #496
closed 2017-06-05 by 130s