moveit/moveit #497: packaging/linking issues due to so-versions

simonschmeisser, 2017-05-03:
We build deb packages to distribute our software to customers and to install it for testing. Since we are still in the testing phase we directly use the swri deb repositories to install current moveit kinetic-devel. Now since a little while moveit includes and increments so-version with every release. So when updating say ros-kinetic-moveit-core we get a unresolved library dependency on

     libmoveit_profiler.so.0.9.5 => not found

as the new so-name is libmoveit_profiler.so.0.9.6

Since our packages somehow do not pick up this strict so-name dependency (they depend on moveit-core, not moveit-core = 0.9.5) , dpkg allows to install these updates even though later on our code does not run any more (before updating to a new build)

I'm wondering if there is anything missing on moveits side, if I understand it correctly it is the responsibility of the library package to inform about  incompatibilities between package/library versions. 

Using an extra entry in the shlibs file, like in this example from /var/lib/dpkg/info/libfontconfig1:amd64.shlibs

      libfontconfig 1 libfontconfig1 (>= 2.11.94)

where I think the version states the oldest version this is still compatible to? and dpkg-build or whatever can then deduce its dependency magic

Also I see that many libraries provide one more symbolic link in the libs folder ie.

     libmoveit_profiler.so
     libmoveit_profiler.so.0.9    <--- this one
     libmoveit_profiler.so.0.9.6

Does anybody have more insight?

I guess I will add a script to add a manual dependency on the current version of moveit-core just to be sure

jspricke, 2017-05-03:
Hi Simon, regarding the dependencies, this should be handled by dpkg-shlibdeps if you add ${shlibs:Depends} to the Dependencies, see 8.6.1 of the Debian policy: https://www.debian.org/doc/debian-policy/ch-sharedlibs.html

simonschmeisser, 2017-05-03:
thanks for the hint but I'm using this already. unfortunately I get (dpkg -I)

     Depends: libassimp3v5, libboost-chrono1.58.0, libboost-filesystem1.58.0, libboost-system1.58.0, libboost-thread1.58.0, libc6 (>= 2.15), libconsole-bridge0.2v5, libgcc1 (>= 1:4.0), libstdc++6 (>= 5.2), libtinyxml2.6.2v5, liburdfdom-world0.4, libyaml-cpp0.5v5, qt58base, qt58declarative, qt58serialbus, ros-kinetic-moveit-core, ros-kinetic-moveit-ros-perception, ros-kinetic-moveit-ros-planning, ros-kinetic-opencv3, ros-kinetic-orocos-kdl, mikado-ros

while it should be (I think):

Depends: libassimp3v5, libboost-chrono1.58.0, libboost-filesystem1.58.0, libboost-system1.58.0, libboost-thread1.58.0, libc6 (>= 2.15), libconsole-bridge0.2v5, libgcc1 (>= 1:4.0), libstdc++6 (>= 5.2), libtinyxml2.6.2v5, liburdfdom-world0.4, libyaml-cpp0.5v5, qt58base, qt58declarative, qt58serialbus, ros-kinetic-moveit-core (=0.9.6), ros-kinetic-moveit-ros-perception (=0.9.6), ros-kinetic-moveit-ros-planning (=0.9.6), ros-kinetic-opencv3, ros-kinetic-orocos-kdl, mikado-ros

jspricke, 2017-05-03:
In Debian this would be achieved by naming the library package name according to the SONAME (policy 8.6.2). As we don't have this with bloom, we could adopt the shlibs file to express the exact dependency. Reading `man dh_makeshlibs`, I think it should be enough to call it with `-V`. If you agree, I can create a pull request for bloom later.

jspricke, 2017-05-03:
..`-V` would not be enough as we would get `(>= 0.9.6)`. Calling dh_makeshlibs correctly seems to be not so easy, maybe we could sed the `>` afterwards.. (I don't like that, but whatever..)

simonschmeisser, 2017-05-03:
Sounds like we're on a good track. 

There is an example in the man page (https://www.mankier.com/1/dh_makeshlibs)

     dh_makeshlibs -V 'libfoobar1 (>= 1.0)'

I'd assume this syntax can also be used with a strict dependency like this

     dh_makeshlibs -V 'libfoobar1 (= 1.0)'

What I don't know yet is on which side to add this? In my package or in moveit-core.

jspricke, 2017-05-03:
in the debian/rules of moveit-core (which is generated by bloom...)

lkj1114889770, 2017-05-04:
I also get the same problem when I use the move_group interface.I just upgrade ros-kinetic-moveit, and the codei write before without error now the following error occurred: 

 **error while loading shared libraries: libmoveit_planning_scene_interface.so.0.9.5: cannot open shared object file: No such file or directory**

the lib version has change from 0.9.5 to 0.9.6

So how to solve the problem??

jspricke, 2017-05-04:
@lkj1114889770 there was an ABI change, you have to recompile.

lkj1114889770, 2017-05-04:
@jspricke Ｉhave catkin_make in my catkin workspace again, it doesn't work. So what's do you mean to recompile , actually I don't really understand, can you explian it more clearly ?  Thanks.

jspricke, 2017-05-04:
Try `rm -r build devel`. Read up on ABI versioning and Sonames.

lkj1114889770, 2017-05-04:
@jspricke It works! I 'rm -r devel', and catkin_make all file in src.It do solve theproblem.

Thanks !

simonschmeisser, 2017-05-04:
@jspricke sounds good, looking forward to get that integrated into bloom

jspricke, 2017-05-08:
@simonschmeisser I've opened a pull request https://github.com/ros-infrastructure/bloom/pull/426, can you give it a try?

ipa-mdl, 2018-06-02:
If moveit increments the sonames with every release, this should be tracked with `version_eq` tags in the package manifests.

Hugal31, 2022-02-01:
Do we have a solution for this? The bloom PR is in a stale, and MoveIt! still not include a correct shlibs file, so it is not possible to use dh_makeshlibdeps correctly with MoveIt!.

Example:  I have a package that links against libmoveit_robot_model.so.1.1.6. When I package it, dh_makeshlibdeps do not specify any version, so when MoveIt! upgrade to 1.1.7, my package is not forced to upgrade and I have a link error.