moveit/moveit #498: [kinetic] Invalid Trajectory: start point deviates from current robot state

lkj1114889770, 2017-05-04:
I use descrtes in moveit. And when i execute a cartesian motion planning follow the descrates tutorial  [http://wiki.ros.org/descartes/Tutorials/Getting%20Started%20with%20Descartes ](url) , The errors comes in the terminal of running ur3_moveit_config demo.launch , the error is:

Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'elbow_joint': expected: -1.43245, current: 1.6088
My ROS version is kinetic., so the ropstopic has no topic named /move_group/trajectory_execution/allowed_start_tolerance, which is a way to solve theproblem described in http://moveit.ros.org/moveit!/ros/2017/01/03/firstIndigoRelease.html for Indigo. So how to sove the problem in kinetic.

gavanderhoorn, 2017-05-04:
ros-industrial-consortium/descartes#196.

v4hn, 2017-05-08:
According to the linked issue, this is resolved.

closed 2017-05-08 by v4hn