moveit/moveit #501: Timing issues with joint state updates in PlanningSceneMonitor

dcconner, 2017-05-10:
### Description

Lines 
[trajectory_execution_manager.cpp L920](https://github.com/ros-planning/moveit/blob/026f9794dd6b09b45debb58415c736acb680a6ab/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L925) and [L1456](https://github.com/ros-planning/moveit/blob/026f9794dd6b09b45debb58415c736acb680a6ab/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1460) are throwing errors regularly due to csm_->waitForCurrentState  returning false.

After not seeing debug output, I built from source and added some debug statements to above methods, and then to [jointStateCallback](https://github.com/ros-planning/moveit/blob/026f9794dd6b09b45debb58415c736acb680a6ab/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L345)  to track the joint state updates.

The updates are coming in at the expected 10 hz, but they appear to be lagging almost 1 second vs. the when I echo the JointState topic to console.

I see that there have been a number of recent updates including some related to these methods #232, and was curious if anyone else has seen this issue.

### Your environment
* ROS Distro:  Kinetic
* OS Version:  Ubuntu 16.04
* Source or Binary build?  Both
* If binary, which release version?  Occurred with latest kinetic release - specific version <later>
* If source, which git commit or tag? 026f9794dd6b09b45debb58415c736acb680a6ab

### Steps to reproduce

Using an internal model of Kinova Mico arm with Kinect-based point cloud for Octomap updating of planning scene.

### Expected behaviour

Joint states are published and updated at 10 hz.  Topic seems to have minimal delay, so I would not expect the execution errors linked above.

### Actual behaviour

Periodic time out due to lagging processing of joint states of around 1 second.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb. <later>

skohlbr, 2017-08-07:
After switching from Indigo to Kinetic we started seeing this error message also every single time when planning in Gazebo simulation (arm still moved though). I'll revisit this in the coming days and provide more info, but it seems very likely that #232 did not properly fix race conditions regarding planning scene updates.

skohlbr, 2017-08-08:
Revisiting this I ran across the ROS_ERROR spam described in #579. I couldn't immediately reproduce the timeouts, but remember seeing them before, possibly when using a custom move_group plugin.

v4hn, 2017-09-11:
@skohlbr @dcconner Is this still relevant?

saurabhbansal90, 2018-01-04:
Hello All, I am also facing this problem (i.e. Empty Jointstates). I am not finding any proper solution for this problem. How can I avoid this issue in my software.
I am not able to move my arm, because of this issue.

Can somebody please let me know how can I resolve this issue.

Thanks in advance,
Saurabh

rhaschke, 2018-01-04:
@saurabhbansal90 Which release you are working on (Indigo or Kinetic)?
In Kinetic we shouldn't have this issue.
For Indigo there is a pending PR: #717. I will implement the pending changes myself.
Could you meanwhile go for a test on Indigo?

saurabhbansal90, 2018-01-05:
Hello Haschke,

Thanks for your response and giving this update. Yes I am working on Indigo only.
I will wait for the update to come.

One more thing, on Kinetic I am not able to use **moveit_core**. Is there any update with which I can use moveit core in Kinetic.

rhaschke, 2018-01-05:
> I will wait for the update to come.

As I don't have a running Indigo system anymore, I'm depending on your feedback to know that the proposed PR #717 indeed solves your issue. So, please build and test if you can.

> One more thing, on Kinetic I am not able to use moveit_core. Is there any update with which I can use moveit core in Kinetic.

This is rather unspecific. MoveIt for Kinectic is released for ages and it works nice for many of us. Please provide a detailed error report in a separate issue.

saurabhbansal90, 2018-02-01:
Hello Haschke,
Thanks for the response. I  was facing the compilation issues with MoveIt on ROS Kinetic. I thought this issue is with MoveIt support in Kinetic.
But I resolved this issue with below link:
https://answers.ros.org/question/216842/ros-using-c-11-how-to-use-with-catkin/

rhaschke, 2018-02-01:
Closed via #752.

closed 2018-02-01 by rhaschke