moveit/moveit #502: Rely on OMPL from launchpad, not ros-*-ompl?

130s, 2017-05-10:
Currently MoveIt! requires `ros-%DISTRO%-ompl`, binary built on ROS buildfarm.
Looking at @mamoll's post [here](https://bitbucket.org/ompl/ompl/issues/318/release-into-ros-lunar#comment-36701382) OMPL binary is also available from `launchpad`, I assume there has been duplicate release task effort.

Do we want to rely on the standalone version of OMPL? I don't have a strong opinion here though.

As @mamoll mentioned, version may not be uptodate of the time, e.g. for Xenial [1.0.0 on launchpad](https://launchpad.net/ubuntu/+source/ompl/1.0.0+ds2-1build1) while [1.2.1 on ROS Kinetic](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=ompl). By aggregating the release effort, is it going to be possible to keep up the version better
(I'm not knowledgeable in release process on launchpad. I don't know either who does the release tasks for launch pad for OMPL)?

I briefly tried on Indigo to use `libompl-dev` as follows but clearly needs update in order to get it work.

```
$ dpkg -p libompl-dev |grep Ver
Version: 0.13.0+git20130920.01d0ca4-1

$ sudo apt-get purge ros-indigo-ompl 
$ sudo apt-get install libompl-dev 
$ sudo apt-get install ros-indigo-baxter-simulator ros-indigo-baxter-moveit-config 
$ rospack profile 

term-1$ roslaunch baxter_gazebo baxter_world.launch 

term-2$ roslaunch baxter_moveit_config demo_baxter.launch 
:
[ WARN] [1494443736.442724788, 49.015000000]: MoveGroup running was unable to load ompl_interface/OMPLPlanner
[ INFO] [1494443737.047532372, 49.611000000]: TrajectoryExecution will use old service capability.
[ INFO] [1494443737.047716068, 49.611000000]: Ready to take MoveGroup commands for group both_arms.
[ INFO] [1494443737.047876174, 49.611000000]: Looking around: no
[ INFO] [1494443737.048265776, 49.612000000]: Replanning: no
[ WARN] [1494443738.232109157, 50.781000000]: Unable to find a random collision free configuration after 100 attempts
[ INFO] [1494443741.399036418, 53.875000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1494443741.399366737, 53.875000000]: Planning attempt 1 of at most 1
[ERROR] [1494443741.399467296, 53.875000000]: No planning plugin loaded. Cannot plan.
[ INFO] [1494443741.401486929, 53.878000000]: ABORTED: Unknown event
```

mamoll, 2017-05-10:
@lepalom is handling the Debian packaging. It seems inevitable that Debian releases will always be somewhat behind ROS releases.

v4hn, 2017-05-10:
In the best case scenario the latest MoveIt release should compile with both. :)

I believe it makes sense to stay with `ros-*-ompl` for our osrf-debs in the forseeable future.
I think it's also in the interest of @mamoll to have OMPL bug reports by MoveIt users that use
more current releases instead of possibly older versions that made it into the respective ubuntu versions.

As the ROS-D (debian ;)) movement gains traction we will see a debian MoveIt at some point anyway.
That one will use Debian's ompl package for sure...

davetcoleman, 2017-06-04:
I'm late to this conversation but just an extra thought about moving towards debian-based OMPL - I believe this would make it more difficult to develop on OMPL and MoveIt! at the same time. [Building OMPL in my Catkin workspace](http://moveit.ros.org/install/source/dependencies/) has been really handy and is easy to overlay over the ROS buildfarm debians

mamoll, 2017-06-04:
I'm also in favor of having ROS package its own (likely newer) version of OMPL. We want people to be able to contribute to OMPL and use their contributions in MoveIt!. They are less likely to do that if they have to wait a year or two for their contributions to appear in the Debian release.

130s, 2017-06-06:
I think we've reached the consensus: MoveIt! will keep using `ros-%DISTRO%-ompl`. Thank you all.

closed 2017-06-06 by 130s