moveit/moveit #504: [Ubuntu Zesty] Build failure around Boost

130s, 2017-05-12:
Please see the error message at https://github.com/ros-planning/moveit/issues/486#issuecomment-300911832

This blocks MoveIt! release into ROS Lunar https://github.com/ros-planning/moveit/issues/486

I'm not sure if similar error occurs on other platforms, but due to the [higher version of `boost`](
http://www.ros.org/reps/rep-0003.html#lunar-loggerhead-may-2017-may-2019) than the one in Xenial, it'd be worth checking Ubuntu `Yakkety`, Debian `Stretch` too (I can run prerelease test on Yakkety soon to see if any issues rise).

If someone can take a look that'll be appreciated. Just let us know by adding a comment then.

mikaelarguedas, 2017-05-12:
I would guess the same issue will appear on Stretch (given that it ships with the same version of boost as Zesty).
The issue doesnt seem to appear on Yakkety. But the build fails on another package for an unrelated reason IMO: moveit_setup_assistant.

There is a gist [here](https://gist.github.com/mikaelarguedas/7731b4100e7ed78b6aab73f052ffac79) with the prerelease console outputs for Yakkety and Zesty for anyone wanting to investigate the failures

closed 2017-05-15 by v4hn

mikaelarguedas, 2017-06-01:
@130s is it possible to make a new release with the fix from https://github.com/ros-planning/moveit/issues/505 so that every moveit package can be released except MSA ?
Thanks!

PR #505 (2017-05-15): add BOOST_MATH_DISABLE_FLOAT128
### Description

This fixes #504 the compilation issue on Ubuntu Zesty and Debian Stretch.
This change is only for lunar and doesn't need to be cherry-picked into other branches 

Note: I settled for this solution because it seemed the least intrusive to me. There are several other fixes that could be considered:
- By changing the `-std=c++11` flag to `-std=gnu++11` in the `add_compile_option()` call but I preferred to keep the flags consistent across all packages of the moveit repository.
- By using (but needs to bump cmake minimum version to 3.0.2):
```
set(CM [... cut]