moveit/moveit #511: MSA: consider using `xacro --in-order` on Indigo as well?

gavanderhoorn, 2017-05-22:
`xacro` supports Jade+-like parsing of xacros on Indigo, but the Setup Assistant doesn't use this. This leads to the situation where xacros that load perfectly fine when `xacro` is invoked / used stand-alone to not load in the MSA.

Would it be an idea to have Indigo MSA add the `--in-order` argument ([here](https://github.com/ros-planning/moveit/blob/fb2ebe3826e8d2b15dc99188343fc0cf2b3dc939/moveit_setup_assistant/src/widgets/start_screen_widget.cpp#L508)) so that we get Jade+ xacro parsing? Should this be an option (ie: check box)?

v4hn, 2017-06-07:
On Mon, May 22, 2017 at 12:26:58PM -0700, G.A. vd. Hoorn wrote:
> `xacro` supports Jade+-like parsing of xacros on Indigo, but the Setup Assistant doesn't use this. This leads to the situation where xacros that load perfectly fine when `xacro` is invoked / used [...]

I believe changing the default in indigo in not an option because in-order parsing can also break xacros that work well in current indigo.
That being said, I would support a check-box and/or an item in `.setup_assistant` to address this.

gavanderhoorn, 2017-06-07:
A checkbox is a good idea. I'll see if I can implement that.

gavanderhoorn, 2017-07-13:
Although #540 is merged, I'll keep this open as the command line interface to the MSA needs to be checked for support of this.

ipa-fxm, 2017-10-23:
Though the new checkbox works great when `Create New Moveit Configuration Package` is selected, unfortunately, it fails to load the xacro (which uses Jade+ features) when `Edit Existing Moveit Configuration Package` is selected.

gavanderhoorn, 2017-10-23:
@ipa-fxm: see the discussion on the #540 for why we didn't add the checkbox to the *Edit existing package* UI.

gavanderhoorn, 2017-10-23:
@bmagyar: what are you `+1`-ing specifically? :)

bmagyar, 2017-10-23:
My bad,  I haven't seen your response ;)

I do agree with the need Felix expressed, I had issues with that years
ago...

v4hn, 2020-03-23:
closing as outdated

closed 2020-03-23 by v4hn