moveit/moveit #512: rviz: displaying attached-object(s) is wrongly triggered by [Show Robot Visual] / [Show Robot Collision]

svenkrause, 2017-05-23:
### Description
Let
 V = Show Robot Visual
 C = Show Robot Collision
be the fields in the rviz-PlanningScenePlugin under "Scene Robot".

I get the following faulty behaviour in rviz. If only one of V & C is <true> and the other is <false>, and I set the <true> one to <false> as well, I get:
 - (V, !C) --> (!V, !C): robot is (correctly) not displayed, but attached object (incorrectly) is
 - (!V, C) --> (!V, !C): robot is (correctly) not displayed, but attached object (incorrectly) is

From this state, if I set either V or C to <true> again, I get:
 - (!V, !C) --> (V, !C): robot is (correctly) displayed, but attached object (incorrectly) is not
 - (!V, !C) --> (!V, C): robot is (correctly) displayed, but attached object (incorrectly) is not

In other words, the attached object is visible (invisible) whenever the robot is invisible (visible).

On the other hand, if I then set the other one to <true> as well, the attached object shows up again
 - (V, !C) --> (V, C): robot and attached object (correcty) are both displayed
 - (!V, C) --> (V, C): robot and attached object (correcty) are both displayed

and setting either one of these to <false> again afterwards does not change that:
 - (V, C) --> (V, !C): robot and attached object (correcty) are both displayed
 - (V, C) --> (!V, C): robot and attached object (correcty) are both displayed

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

### Steps to reproduce
1) create a PlanningScene with an attached object
2) display the scene in rviz
3) toggle "Show Robot Visual" / "Show Robot Collision"

### Expected behaviour
The attached object should be visible (invisible), whenever the robot is visible (invisible).

### Actual behaviour
Toggling off one of "Visual" / "Collision" while the other is off already, creates the opposite behaviour: The attached object is visible (invisible), whenever the robot is invisible (visible).

v4hn, 2017-06-07:
sorry for the delay.

This looks like a reasonably simple bug, did you try to find it yourself?

svenkrause, 2017-06-07:
I had a quick look in the code online, but couldn't find it. I'd have to download the moveit-source and debug it, but I'm short on time at the moment, so I thought, maybe someone who contributed to that part of the code might have an idea, where the bug might lie. If not, I'll investigate further at some point in the (relatively near) future.

v4hn, 2017-06-07:
> but I'm short on time at the moment,

Just as everyone else :)

> so I thought, maybe someone who contributed to that part of the code might have an idea, where the bug might lie.

I would expect to find it around here:
https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/visualization/robot_state_rviz_plugin/src/robot_state_display.cpp
but it could also lie in one of the dependencies of course...

> If not, I'll investigate further at some point in the (relatively near) future.

Thanks for reporting. It would be awesome if you could debug it and provide a patch :)

simonschmeisser, 2017-07-28:
Using the same approach I can make the attached object disappear:
```
void RobotStateVisualization::setVisualVisible(bool visible)
{
  visual_visible_ = visible;
  robot_.setVisualVisible(visible);
  if (!visual_visible_ && !collision_visible_)
    render_shapes_->clear();
}

void RobotStateVisualization::setCollisionVisible(bool visible)
{
  collision_visible_ = visible;
  robot_.setCollisionVisible(visible);
  if (!visual_visible_ && !collision_visible_)
    render_shapes_->clear();
}
```

The attached object however then does not reappear until the next robot state update is received. I think on "real" robots the updates are received at a certain fixed frequency. For industrial_robot_simulator we unfortunately need to move the robot to get an update. Probably the attached object should be stored somewhere to improve this. I will open a PR once #565 has been merged (or should I add this to it?)

Generally speaking we might want to have another checkbox for the visibility of all attached objects?

svenkrause, 2018-10-25:
World MoveIt-Day: I'm working on it.

PR #1199 (2018-11-15): fix several issues in rendering of attached bodies in rviz plugins
Rebase of #1152. Splitted into the basic fix (this PR) and the additional new feature providing a BoolProperty to switch attached bodies on/off (#1200).
This fixes #512.
Also added bug fixes for several other rendering issues I observed during testing the initial commit.
Look into individual commits for details.
closed 2018-11-15 by v4hn