moveit/moveit #522: moveGroup.setEndEffectorLink() errors

AndyZe, 2017-06-08:
### Description

See http://answers.ros.org/question/238536/issue-using-moveitplanning_interfacemovegroupsetposetarget-with-non-empty-end_effector_link/

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build
* 0.9.6 according to package.xml (this is for 'moveit' -- looks like there are several moveit-related packages)

AndyZe, 2017-06-08:
I've noticed that:
`moveGroup.setEndEffector("camera_ee");`
`ROS_INFO_STREAM("MoveGroup EE: " << moveGroup.getEndEffector() );`

Prints out:
`MoveGroup EE: `

AndyZe, 2017-06-15:
OK, I figured out that one of my end effectors wasn't set up properly. Still having trouble with the motion planning of the 2nd end effector, but at least moveGroup.setEndEffector() is changing the value returned by moveGroup.getEndEffector().

AndyZe, 2017-06-16:
I'm going to go ahead and reopen this as I've spent quite a bit of time trying to solve it. It seems now that moveGroup.setEndEffectorLink() is being applied, but then it tries to plan for **2!** end effectors -- the default and the new end effector. Of course, constraints on both can't be satisfied. See attached error, my planning groups, and the code snippet below:

`  // Test
  camera_pose = moveGroup.getCurrentPose();
  camera_pose.pose.position.x += 0.04;

  // Move
  moveGroup.setEndEffectorLink("camera_ee_link");
  moveGroup.setPoseTarget( camera_pose,"camera_ee_link" );
  moveGroup.setPlannerId("ESTkConfigDefault");
  moveGroup.setGoalPositionTolerance(0.25);
  moveGroup.setGoalOrientationTolerance(0.1);
  //br.sendTransform( tf::StampedTransform(transform, ros::Time::now(), "camera_ee_link", "new_cam_frame") );
  ROS_INFO_STREAM("End effector: " << moveGroup.getEndEffectorLink() );
  listener.waitForTransform( "base_link", "new_cam_frame", ros::Time(0), ros::Duration(10.0));
  moveGroup.move();`

[image]

[image]

AndyZe, 2017-06-16:
I realize it's kind of sketchy to include joints and links **and** a chain in my Planning Group. Here's a new setup with just the chain. Same errors when switching EndEffectorLinks.

[image]

willcbaker, 2017-11-09:
Can confirm, it seems we are unable to sample goal states for ee_links that are not the default ee_link

hartmanndennis, 2017-11-15:
Try the source build of moveit. #582 should have fixed this bug.

BrettHemes, 2018-05-23:
@AndyZe did this get resolved?  I am having a similar issue...   I am on version 0.9.11 and after changing the end effector link ONCE it works but on the SECOND change, I see similar behavior where constraints are getting checked for two links (the first changed along with the second)...

BrettHemes, 2018-05-24:
All right! progress!  ... so I played with this a bit more and it turns out the issue was me not understanding fully the `planning_interface`.  It seems that you can set pose target(s) (plural) for a number of end effector(s) (again plural)?  See the relevant API source [at line 1126 here](http://docs.ros.org/kinetic/api/moveit_ros_planning_interface/html/move__group__interface_8cpp_source.html#l01126).  That said, when switching between end effector links using `setEndEffectorLink` the pose target(s) set for the previously set link hang around, thus resulting in the observed multiple constraints checks (and subsequent failure).  A simple call to `clearPoseTargets()` after changing the end effector link solved my issue (and I believe should solve that of @AndyZe's multiple constraint checks).

Sorry for the bother; hopefully this comment helps someone else with similar issues :smile:

TIL the planning interface can do cool stuff for end effector stuff using multiple targets

AndyZe, 2018-05-24:
Cool! I ended up making a new MoveGroup with MoveItSetupAsst for each end effector.

closed 2018-05-24 by AndyZe