moveit/moveit #526: Trajectory slider plays for the previous goal when a plan for the new goal is unavailable

130s, 2017-06-13:
Thanks for the nice feature @Jntzko @v4hn https://github.com/ros-planning/moveit/pull/491!

### Description

Pressing `Play` button on `Trajectory Slider` panel shows older trajectory, when a plan for the current goal is not available.

Without knowing if this behavior is intended, this can be simply confusing.

### Your environment
* ROS Distro: [Indigo]
* OS Version: Ubuntu 14.04
* Build

 ```
$ dpkg -p ros-indigo-moveit-ros-visualization |grep Ver                                                 
Version: 0.7.10-0trusty-20170607-162519-0800
```

### Steps to reproduce

```
roslaunch pr2_moveit_config demo.launch
```

Using right hand eef,

1. Move it anywhere, then `Plan and Execute`.
1. Move eef anywhere. Don't run `Plan`.
1. "`Play`" on Trajectory Slider.

### Expected behaviour

Show waypoints along the trajectory for the current goal.

### Actual behaviour

Explained in `Description` section.

v4hn, 2017-06-13:
Weeeell... I can see how this can be confusing.
Our MoveIt Motion Planning Panel only imports the Trajectory visualization display and uses it "as is".
This display caches the trajectory it received most recently and presents it to the user.
So as long as no new trajectory - produced by successful planning - was received the old one is still there.
I believe the behavior you noticed was the same with the "loop animation" mode before the panel, so this is not actually something new.

Knowing the internal structure of the plugins, the current behavior makes sense to me.
But maybe we want to adjust this?

130s, 2017-06-13:
Would there be any issue when `Play` on Trajectory Slider panel gets pressed then planning internally runs? Assumption: I can't think of any situations where you want to see a trajectory from the goal previously set while a new goal is already set.

v4hn, 2017-06-14:
I don't like that behavior.
We wanted the user to be able to re"Play" the current trajectory over and over again (as a more controlled version of "loop animation"),
so the play should not trigger planning.

I suppose we could remove the current trajectory whenever the user pressed "Plan" or "Plan&Execute", so that there is *no* plan anymore when planning fails.
Would that help?

130s, 2017-06-14:
> We wanted the user to be able to re"Play" the current trajectory over and over again (as a more controlled version of "loop animation"), so the play should not trigger planning.

That makes sense.

> I suppose we could remove the current trajectory whenever the user pressed "Plan" or "Plan&Execute", so that there is *no* plan anymore when planning fails. Would that help?

+1
Is this something that someone on your end can work on (hopefully very quickly since it's you guys that added this feature)? Otherwise we'll see if someone incl. me in the community will have time to tackle.

PR #531 (2017-06-20): make Isaac happier
rviz display: stop trajectory visualization on new plan

It can be confusing to see the old trajectory even though
the user already requested a new plan.

This is especially annoying in case the second planning request fails!

Fixes #526

Should be forward-picked to j/k
commit 03aa7b88 (2017-06-20): rviz display: stop trajectory visualization on new plan

It can be confusing to see the old trajectory even though
the user already requested a new plan.

This is especially annoying in case the second planning request fails!

Fixes #526
commit 46b0ef0e (2017-06-20): rviz display: stop trajectory visualization on new plan (`#531 <https://github.com/ros-planning/moveit/pull/531>`_)

It can be confusing to see the old trajectory even though
the user already requested a new plan.

This is especially annoying in case the second planning request fails!

Fixes #526

disable slider panel state appropriately on 0 waypoint update

[moveit_ros/visualization] C++ format glitch (see https://travis-ci.org/ros-planning/moveit/jobs/244899864#L163).
commit 24074164 (2017-06-20): rviz display: stop trajectory visualization on new plan (`#531 <https://github.com/ros-planning/moveit/pull/531>`_)

It can be confusing to see the old trajectory even though
the user already requested a new plan.

This is especially annoying in case the second planning request fails!

Fixes #526

disable slider panel state appropriately on 0 waypoint update

[moveit_ros/visualization] C++ format glitch (see https://travis-ci.org/ros-planning/moveit/jobs/244899864#L163).
closed 2017-06-20 by 130s