moveit/moveit #528: Display RobotState crashed rviz

cuixiongyi, 2017-06-16:
## Description
When add moveit_ros_visualization-RobotState, if there a link without visual element in URDF, rviz will crash.
When I added the visual element, it worked find and displayed the robot.

**Example link URDF without visual, will crash:**
```
<link
    name="robot_tip">
    <inertial>
      <origin
        xyz="0 0.0 0.0"
        rpy="0 0 0" />
      <mass
        value="0.0" />
      <inertia
        ixx="0.0"
        ixy="0.0"
        ixz="0.0"
        iyy="0.0"
        iyz="0.0"
        izz="0.0" />
    </inertial>
   </link>

```
**Example link URDF with visual, will not crash:**
```
<link
    name="robot_tip">
    <inertial>
      <origin
        xyz="0 0.0 0.0"
        rpy="0 0 0" />
      <mass
        value="0.0" />
      <inertia
        ixx="0.0"
        ixy="0.0"
        ixz="0.0"
        iyy="0.0"
        iyz="0.0"
        izz="0.0" />
    </inertial>
    <visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://kuka_robot_kr6_r1620/meshes/Robot_L6.STL" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.79216 0.81961 0.93333 1" />
      </material>
    </visual>
  </link>
```

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo]
* OS Version: e.g. Ubuntu 14.04 64bit
* Binary build
* If binary, which release version?
  dpkg version of ros-indigo-moveit-ros-visualization        0.7.9-0trusty-20170517-195545-0800

### Steps to reproduce
remove any visual element of a robot URDF
Then launch its moveit_config

   `roslaunch robot_moveit_config demo.launch`
Then open a new rviz

`rosrun rviz rviz`

Add moveit_ros_visualization-RobotState
It will crash (Segmentation fault (core dumped))

### Expected behaviour
It should display the robot in rviz

### Actual behaviour
rviz crashed, Segmentation fault (core dumped)

### Backtrace or Console output
Segmentation fault (core dumped)

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

v4hn, 2017-06-16:
You are right, I can reproduce this on 14.04 with the current ros-indigo branch.
Crap :/

Apparently aac7921666821fa7b38c0228b390dbc8d133d833 is the faulty commit, reverting it fixes the error.
However, the crash is not directly related to the code there but occurs somewhere inside qt/rviz' event system.
I'll have a look.

v4hn, 2017-06-16:
@cuixiongyi could you checkout #530 and verify that it solves the problem?

cuixiongyi, 2017-06-16:
Yes, I tried your branch. It won't crash any more. 
Thank you for responding. You are amazingly fast.
Yeah, rviz isn't the most stable software. Sometimes it will crash upon open.

v4hn, 2017-06-16:
On Fri, Jun 16, 2017 at 02:00:19PM -0700, Xiongyi Cui wrote:
> Yes, I tried your branch. It won't crash any more.
> Thank you for responding. You are amazingly fast.

In this case I'm actually the one responsible for the bug (for a change...)
and knew what to look for. Don't count on it on other occasions. :)

@130s I suppose we want to have this fixed with the next sync, if that's possible...

130s, 2017-06-16:
> @130s I suppose we want to have this fixed with the next sync, if that's possible...

Yeah, critical bug fixes should be prioritized. Started tracking at #533.

PR #530 (2017-06-19): robot display: Don't reload robot model upon topic change
Instead just reset it to the default configuration.
This avoids needless urdf-parsing and retains a valid kstate_.
For changes to the robot description parameter,
there is a separate callback function.

On Ubuntu 14.04 indigo this also fixes a segfault during startup,
that seems to be related to calling loadRobotModel() twice in a row.

This should be forward ported to j/k.

Fixes #528
commit 85a38e1e (2017-06-19): robot display: Don't reload robot model upon topic change

Instead just reset it to the default configuration.
This avoids needless urdf-parsing and retains a valid kstate_.
For changes to the robot description parameter,
there is a separate callback function.

On Ubuntu 14.04 indigo this also fixes a segfault during startup,
that seems to be related to calling loadRobotModel() twice in a row.

Fixes #528
commit 83b167a6 (2017-06-19): robot display: Don't reload robot model upon topic change

Instead just reset it to the default configuration.
This avoids needless urdf-parsing and retains a valid kstate_.
For changes to the robot description parameter,
there is a separate callback function.

On Ubuntu 14.04 indigo this also fixes a segfault during startup,
that seems to be related to calling loadRobotModel() twice in a row.

Fixes #528

Conflicts:
	moveit_ros/visualization/robot_state_rviz_plugin/src/robot_state_display.cpp
commit dd7f0fdd (2017-06-19): robot display: Don't reload robot model upon topic change

Instead just reset it to the default configuration.
This avoids needless urdf-parsing and retains a valid kstate_.
For changes to the robot description parameter,
there is a separate callback function.

On Ubuntu 14.04 indigo this also fixes a segfault during startup,
that seems to be related to calling loadRobotModel() twice in a row.

Fixes #528
closed 2017-06-19 by 130s