moveit/moveit #533: Release 201706

130s, 2017-06-16:
There is/are some bug fixes that seem (kind of) critical:
- https://github.com/ros-planning/moveit/issues/528

[Next Indigo sync may happen soon, sometime within June](https://github.com/ros-planning/moveit/issues/528#issuecomment-309140751), so if there's no other requests I'll make a new release early in the week of 18th for IKL.

130s, 2017-06-19:
I think the minimum required for this patch release is #528 and that is already merged. If there are any other changes that anyone wants to include in the next sync, please let us know here ASAP. Otherwise I'll  make a quick release on Tuesday Pacific.

FYI I see more than 5 PRs that look green on Travis.

v4hn, 2017-06-21:
> FYI I see more than 5 PRs that look green on Travis.

@davetcoleman and I just merged many of them - there are still more than 5 green ones, but
most of them required further investigations before a merge.

Go ahead with a release 👍

130s, 2017-06-21:
I https://github.com/ros/rosdistro/pull/15325
K https://github.com/ros/rosdistro/pull/15326#thread-subscription-status
L https://github.com/ros/rosdistro/pull/15327

Thanks every one esp. @v4hn @davetcoleman

closed 2017-06-21 by 130s