moveit/moveit #538: pr2 description setup tutorial fails on clean install

ryanoldja, 2017-06-27:
### Description

Attempting to walk through the MoveIt [setup tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html) for the PR2. After uploading the pr2.urdf.xacro file, the moveit_setup_assistant errors out & dies on exit code -11.

### Your environment
* ROS Distro: Kinetic
* OS Version: UbuntuMATE 16.04 (Raspberry Pi 3)
* installed by `apt-get ros-kinetic-moveit`
* also tried installing from source, but was unable to complete the build

### Steps to reproduce
`sudo apt-get install ros-kinetic-moveit`
`sudo apt-get install ros-kinetic-pr2-common`
`roslaunch moveit_setup_assistant setup_assistant.launch`
When the GUI launches, select 'Create New MoveIt Configuration Package'.
Navigate to `/opt/ros/kinetic/share/pr2_description/robots/` and open pr2.urdf.xacro.
Select 'Load Files'.

### Backtrace or Console output

```
... logging to /home/rigrade/.ros/log/667d8c06-5adc-11e7-91e6-b827eb02af68/roslaunch-pi2-3512.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pi2:33275/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

auto-starting new master
process[master]: started with pid [3523]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 667d8c06-5adc-11e7-91e6-b827eb02af68
process[rosout-1]: started with pid [3536]
started core service [/rosout]
process[moveit_setup_assistant-2]: started with pid [3539]
libEGL warning: DRI2: failed to authenticate
[rospack] Error: no package given
[librospack]: error while executing command
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /opt/ros/kinetic/share/pr2_description/urdf/common.xacro
included from: /opt/ros/kinetic/share/pr2_description/robots/pr2.urdf.xacro

inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/shoulder_v0/shoulder.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/upper_arm_v0/upper_arm.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/forearm_v0/forearm.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/gripper_v0/gripper.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/head_v0/head.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/tilting_laser_v0/tilting_laser.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/torso_v0/torso.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/base_v0/base.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/sensors/head_sensor_package.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/sensors/wge100_camera.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/urdf/sensors/projector_wg6802418.urdf.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://playerstage.sourceforge.net/gazebo/xmlschema/#interface
 new: http://ros.org/wiki/xacro (/opt/ros/kinetic/share/pr2_description/gazebo/gazebo.urdf.xacro)
[ INFO] [1498528914.146582940]: Loaded pr2 robot model.
[ INFO] [1498528914.147088716]: Setting Param Server with Robot Description
[ INFO] [1498528914.293050690]: Robot semantic model successfully loaded.
[ INFO] [1498528914.293236730]: Setting Param Server with Robot Semantic Description
[ INFO] [1498528914.433447453]: Loading robot model 'pr2'...
[ INFO] [1498528914.433755107]: No root/virtual joint specified in SRDF. Assuming fixed joint
================================================================================REQUIRED process [moveit_setup_assistant-2] has died!
process has died [pid 3539, exit code -11, cmd /opt/ros/kinetic/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/rigrade/.ros/log/667d8c06-5adc-11e7-91e6-b827eb02af68/moveit_setup_assistant-2.log].
log file: /home/rigrade/.ros/log/667d8c06-5adc-11e7-91e6-b827eb02af68/moveit_setup_assistant-2*.log
Initiating shutdown!
================================================================================
[moveit_setup_assistant-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

v4hn, 2017-06-27:
> libEGL warning: DRI2: failed to authenticate

This looks like a bug in your 3d-graphics setup.

ryanoldja, 2017-06-27:
@v4hn thanks for your reply. How can I go about resolving my 3d-graphics setup issue? Is this causing the MoveIt process failure?

v4hn, 2017-06-27:
> UbuntuMATE 16.04 (Raspberry Pi 3)

Oh, I just saw that now. I'm not convinced anyone tried to run the setup assistant on a pi3 before.
Can you run RViz there and visualize a robot model?

It's quite likely that the assistant fails to create the 3d-graphics context and thus dies.

ryanoldja, 2017-06-27:
Yes, RViz runs fine for visualizing the robot model. Tested that first through the `urdf_tutorial` package.
In those cases, I also get the `libEGL warning: DRI2: failed to authenticate`, but it doesn't seem to cause any problems in RViz.

v4hn, 2017-06-27:
graphics are probably not the problem then.
Too bad, that would have been the easy explanation.

I remember some bug with the debian package of fcl compiled for pi3,
but I don't know whether this is still a problem.

Sorry, but we you will have to debug this yourself.
Compile MoveIt in Debug mode, run the assistant with gdb and have
a look at the stacktrace. Where does the segfault occur?

Also, I might state the obvious here, but you could of course
setup the moveit configuration on a different machine and then copy
it over to your pi. I'm not convinced the segfault is limited to the
assistant though, so whatever is causing it might cause your `move_group`
node to fail later on as well...

ryanoldja, 2017-06-27:
Neither `apt-get install` nor any of the commands used with the [source install](moveit.ros.org/install/source) install have an option 'debug mode'. I can't run gdb on roslaunch either as it's looking for an executable. Could you detail how to do this?

Regarding your last comment, that seems to be the case. Launching with the results from the setup assistant works fine on our Ubuntu Xenial 16.04 machines but does not on the RPi3:
```
[move_group-4] process has died [pid 4876, exit code -11, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/rigrade/.ros/log/b4d61fe6-5b57-11e7-991b-b827eb02af68/move_group-4.log].
log file: /home/rigrade/.ros/log/b4d61fe6-5b57-11e7-991b-b827eb02af68/move_group-4*.log
[ INFO] [1498581798.568850818]: rviz version 1.12.10
[ INFO] [1498581798.569073993]: compiled against Qt version 5.5.1
[ INFO] [1498581798.569147221]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1498581799.590699075]: Stereo is NOT SUPPORTED
[ INFO] [1498581799.591329174]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1498581805.808613654]: Loading robot model 'rigrade_robot'...
[ INFO] [1498581805.808974900]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1498581806.205930774]: Loading robot model 'rigrade_robot'...
[ INFO] [1498581806.206098429]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz_pi2_4828_906981654-5] process has died [pid 4877, exit code -11, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/rigrade/Desktop/rigrade_ws/src/rigrade_meta/rigrade_moveit/launch/moveit.rviz __name:=rviz_pi2_4828_906981654 __log:=/home/rigrade/.ros/log/b4d61fe6-5b57-11e7-991b-b827eb02af68/rviz_pi2_4828_906981654-5.log].
log file: /home/rigrade/.ros/log/b4d61fe6-5b57-11e7-991b-b827eb02af68/rviz_pi2_4828_906981654-5*.log
```

Would it make sense to downgrade to ROS Indigo?