moveit/moveit #539: Implmenting a custom action file

TroyCordie, 2017-06-27:
### Description
Segfault when executing trajectory with custom action file.  New to moveit so I'm hoping I might get pointed in the right direction or this might flag to someone in the know.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source 
* 07448175688087a0216e42181d99b3ed1827e588

### Steps to reproduce
I have been attempting to run the simulator located at https://github.com/wilselby/ROS_quadrotor_simulator.

### Expected behaviour
the custom action file should implement a multi-degree of freedom joint and by all accounts has been working.

### Actual behaviour
When executing a planned trajectory the code segfaults resulting in the following backtrace.

### Backtrace or Console output
https://gist.github.com/TroyCordie/55de6da4ec32112981d7ba4c38512298

v4hn, 2017-06-27:
Ouch, the validation was added recently and apparently there is a bug w.r.t. MultiDOF trajectories.

You can skip it (and most likely therefore the segfault) by setting the validation tolerance to 0.0 [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/trajectory_execution.launch.xml#L14) in your moveit config.

It would help us a lot if you could figure out in which line the segfault occurs or give us an example trajectory to debug this!

Very sorry for the inconvenience.

TroyCordie, 2017-06-27:
Thank you for the response. 
This fix does indeed skip the section of code that was segfaulting, I will see if I can dig in the next few days and find the culprit. 

Though I am a bit confused by the topic the trajectory is now being output on /execute_trajectory/... I had assumed the topic /multi_dof_joint_trajectory_action/... would be the topic. This may just be a generic question so  I will also ask it in the google groups.

v4hn, 2017-06-28:
I don't understand the question, actually.
Please raise generic questions on http://answers.ros.org

It would be really great if you could look into the segfault a bit more and report back!
You do not by any chance have a gazebo / moveit demo mode setup ready for testing?

TroyCordie, 2017-06-29:
I do have a gazebo model I am trying to tie into moveit, when I get something happening between the two I'll dig deeper into the segfault.

TroyCordie, 2017-09-12:
I have finaly had a chance to follow this up and the issue is in 
bool TrajectoryExecutionManager::validate(const TrajectoryExecutionContext& context) const located in moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp

specifically if (positions.size() != n) on line ~963.

The issue arises when planning for a multi dof joint with out any other joint as trajectory.joint_trajectory.points.front().positions is empty. c is not a strong point but this is a null pointer?

currently, my fix is to add the check 

if(!trajectory.joint_trajectory.points.empty())
    {
     ......
    }
     else if(!trajectory.multi_dof_joint_trajectory.points.empty())
    {
      ROS_INFO_NAMED("move_group", "multi dof joint trajectory valadation not implmented");
      return true;
    }

PR #691 (2017-12-07): Avoid segfault when validating a multidof-only trajectory
Also warn that MultiDOFJointTrajectory validation is not supported

Fixes #539

Should be backported to indigo.
commit abd54361 (2017-12-07): Avoid segfault when validating a multidof-only trajectory (#691)

Also warn that MultiDOFJointTrajectory validation is not supported

Fixes #539
closed 2017-12-07 by v4hn