moveit/moveit #543: Multi DOF Trajectory only providing translation not velocity or acceleration

TroyCordie, 2017-07-06:
### Description

Multi DOF Trajectory only providing translation not velocity or acceleration

### Your environment
* ROS Distro: Kinetic
* Ubuntu 16.04
* Source
* 0744817

Recently I asked a question on ask ROS regarding the multi DOF joints in moveit and them not publishing velocity or acceleration data see the question at the link below.

http://answers.ros.org/question/264957/moveit-trajectory-only-providing-translation-not-velocity-or-acceleration/ 

I have modified 
void robot_trajectory::RobotTrajectory::getRobotTrajectoryMsg(moveit_msgs::RobotTrajectory& trajectory) const...

by adding at line 284
if (waypoints_[i]->hasVelocities())
        {
            const std::vector<std::string> names =  mdof[j]->getVariableNames();
            const double* velocities =  waypoints_[i]->getJointVelocities(mdof[j]);
             geometry_msgs::Twist point_velocity;

            for(std::size_t ii = 0; ii < names.size(); ++ii){
              if(names[ii].find("/x") != std::string::npos)
              {
                ROS_INFO_STREAM("multi dof joint x velocity: "<<velocities[ii]);
                point_velocity.linear.x = velocities[ii];
              }
              else if(names[ii].find("/y") != std::string::npos)
              {
                ROS_INFO_STREAM("multi dof joint y velocity: "<<velocities[ii]);
                point_velocity.linear.y = velocities[ii];
              }
              else if(names[ii].find("/z") != std::string::npos)
              {
                ROS_INFO_STREAM("multi dof joint z velocity: "<<velocities[ii]);
                point_velocity.linear.z = velocities[ii];
              }
              else if(names[ii].find("/theta") != std::string::npos)
              {
                ROS_INFO_STREAM("multi dof joint theta velocity: "<<velocities[ii]);
                point_velocity.angular.z = velocities[ii];
              }
              else
              {
                ROS_INFO_STREAM("velocity name: "<<names[ii]);
              }
            }
            trajectory.multi_dof_joint_trajectory.points[i].velocities.push_back(point_velocity);
}

The velocities now publish on /execute_trajectory/goal but not the topic /multi_dof_joint_trajectory_action/goal for the action controler I have (bsaicaly a copy of ) https://www.wilselby.com/research/ros-integration/3d-mapping-navigation/

I had assumed both messages would be produced through the same function but maybe they aren't or maybe the way I am adding the velocity is not persisting in momory. As im sure you can c++ is not a strong point any advice would be apreciated.

TroyCordie, 2017-07-07:
sorry, guys I just needed another day at it.
For anyone else looking for pointers I added the code below at line 863 of trajectory_execution_manager.cpp
likely not the best solution but it works
```cpp
if (!trajectory.multi_dof_joint_trajectory.points[j].velocities.empty())
{
  parts[i].multi_dof_joint_trajectory.points[j].velocities.resize(bijection.size());

  parts[i].multi_dof_joint_trajectory.points[j].velocities[0].linear.x =
    trajectory.multi_dof_joint_trajectory.points[j].velocities[0].linear.x * execution_velocity_scaling_;

  parts[i].multi_dof_joint_trajectory.points[j].velocities[0].linear.y =
    trajectory.multi_dof_joint_trajectory.points[j].velocities[0].linear.y * execution_velocity_scaling_;

  parts[i].multi_dof_joint_trajectory.points[j].velocities[0].linear.z =
    trajectory.multi_dof_joint_trajectory.points[j].velocities[0].linear.z * execution_velocity_scaling_;

  parts[i].multi_dof_joint_trajectory.points[j].velocities[0].angular.x =
    trajectory.multi_dof_joint_trajectory.points[j].velocities[0].angular.x * execution_velocity_scaling_;

  parts[i].multi_dof_joint_trajectory.points[j].velocities[0].angular.y =
    trajectory.multi_dof_joint_trajectory.points[j].velocities[0].angular.y * execution_velocity_scaling_;

  parts[i].multi_dof_joint_trajectory.points[j].velocities[0].angular.z =
    trajectory.multi_dof_joint_trajectory.points[j].velocities[0].angular.z * execution_velocity_scaling_;
}
```

davetcoleman, 2017-07-17:
@TroyCordie this seems like a valuable contribution that should be in a pull request instead of an issue - do you mind proposing it as a code change? Thanks!

TroyCordie, 2017-07-19:
Can do give me a few days to tidy up what I have played with in the code

TroyCordie, 2017-09-12:
I believe this is closed with pull request #555

v4hn, 2017-09-13:
Agreed. Closing.

closed 2017-09-13 by v4hn