moveit/moveit #545: Setting the solve_type for Python code issuing movement commands

cvanclee, 2017-07-11:
### Description

Setting the solve_type of the kinematics.yaml of a robot package does not affect the solve_type that MoveIt uses when running a launch file from that robot's package, it will just use the default. Also, running code that connects to MoveIt will also use the incorrect solve_type.

* ROS Distro: Indigo
* OS Version: e.g. Ubuntu 14.04

### Steps to reproduce
I installed trac-ik and set one of my robot's kinematics.yaml to use trac-ik along with solve_type:Distance. Both MoveIt and the console running my movement command code still use solve_type:Speed (the default).

### Expected behaviour
MoveIt (and preferably any code connected to MoveIt) should default to the solve_type specified in the kinematics.yaml of whatever package the launch file came from

### Actual behaviour
MoveIt defaults to the standard default solve_type regardless of what kinematics.yaml specifies

I managed to workaround this issue by editing the launch file I was using to and adding
```
 <group ns="move_group">
    <rosparam command="load" file="$(find fanuc_lrmate200id_moveit_config)/config/kinematics.yaml"/>
  </group>
```
To get my code to use the correct solve_type, I had to edit the Python wrapper source code and edit my launch file. (more detailed fix at http://answers.ros.org/question/265479/using-correct-solve_type-when-running-code/?comment=266023#post-id-266023)

gavanderhoorn, 2017-07-11:
I suggested @cvanclee report this as an issue here on our tracker.

The underlying problem (IK plugins load parameters from private namespaces) is not something that we'll easily fix, but in this case the problem was exacerbated by the fact that the commander wrapper apparently uses the `ros::init_options::AnonymousName` option to the node initialisation.

As the name of the node (and thus the namespace in which the parameters will be searched for) then cannot be predicted, it becomes impossible to load the required parameters in an appropriate namespace and make things work that way.

gavanderhoorn, 2017-07-11:
I think `ros::init_options::AnonymousName` was added to the wrapper to allow users to instantiate multiple wrappers without having to deal with naming them uniquely.

If that is the case, could we perhaps allow initing the wrapper without the `Anonymous` hint somehow?

v4hn, 2018-02-12:
Should be fixed by #701 et al.

closed 2018-02-12 by v4hn