moveit/moveit #548: IkParameterizationType translationyaxisangle4d is not implemented

luminize, 2017-07-14:
### TranslationYaxis4D is not implemented in moveit_kinematics

With the help of @gavanderhoorn his answer on http://answers.ros.org/question/263925/generating-an-ikfast-solution-for-4-dof-arm/ I have made an IkFast plugin for my robot.
When dragging the handles in Moveit the robot goal pose doesn't move, and while trying to drag I get the message:
```
[ERROR] [1500016582.397196842]: IK for this IkParameterizationType not implemented yet.
```

I've looked this up and came to the following answer on ros.answers.org http://answers.ros.org/question/191884/ik-for-this-ikparameterizationtype-not-implemented-yet/

I would be greatly helped to know which actions I should take to (similarly as in the answer above) get these kinematics implemented.

### Environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04

v4hn, 2017-07-17:
I have not seen this IK type before but the basic steps to follow are:

- figure out which parameters you have to pass to `ComputeIk`

- add the appropriate code to compute "the right" parameters and pass them to the function [here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L496)

- File a pull-request! :)

luminize, 2017-07-17:
regarding a PR, no problem. But figuring out the first 2 points I have a question if you don't mind:

There's [the instruction](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L531-L533) that the first value represents the angle. Looking at my IKFast cpp file I guess the first value are we talking about [ is the first element of eerot](https://github.com/luminize/matilda_support/blob/cleaned/urdf/output/kinematics.f01b084b2f20efad296fc7039f2f69b5/ikfast71.TranslationYAxisAngle4D.0_1_2_3.cpp#L344)

So I should be able to use the same `ComputeIK` as in IKP_Transform6D and IKP_Translation3D [like here](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L516) ?

v4hn, 2017-07-17:
> Looking at my IKFast cpp file I guess the first value are we talking about is the first element of eerot

I would guess the same thing.

> So I should be able to use the same ComputeIK as in IKP_Transform6D and IKP_Translation3D like here ?

Possibly? I guess you tried it by now? :)
I would pass the address of one double (representing the angle) as `vals`.

Ping @mintar . You fought with these a lot in the past.
Could you drop a quick hint here?

mintar, 2017-07-18:
> I would pass the address of one double (representing the angle) as vals.

I think so too. In the case of `TranslationXaxis4D`,`TranslationYaxis4D` and `TranslationZaxis4D`, `eerot` is probably only a single `IkReal*`. The best way to find out is to look at the `ComputeIk()` function in your generated `*_ikfast_solver.cpp`.

For example, when I implemented the support for `IKP_TranslationDirection5D`, I looked at these lines:

https://github.com/uos/katana_driver/blob/843b59619e42de37aa4b31cd0505188590cb7fcb/katana_moveit_ikfast_plugin/src/katana_450_6m90a_ikfast_solver.cpp#L271-L273

There you can see that it tries to read a 3-dimensional vector (the direction vector) from `eerot`.

So I computed the `direction` vector from `pose_frame` and put it in there:

https://github.com/ros-planning/moveit/blob/e906c4049c5f049f34cba7a4f88d10f185b93d27/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L521-L527

So you need to do something similar for the three `Translation*axis4D` IK types: compute an angle around the X/Y/Z axis from `pose_frame`, put that angle as a `double*` into `eerot`.

FYI, from the [openrave documentation](http://openrave.org/docs/0.6.6/openravepy/ikfast/): 

> **TranslationXAxisAngle4D, TranslationYAxisAngle4D, TranslationZAxisAngle4D** - end effector origin reaches desired 3D translation, manipulator direction makes a specific angle with x/y/z-axis (defined in the manipulator base link’s coordinate system)

For testing, I found it easier to generate some test poses that I know are reachable by the arm (for example in RViz, generate a new random goal pose), and then call MoveIt's IK service with those poses. Dragging the handle adds further complexity IMO, so I did that as a second step.

If you fix it and could provide a PR, that would be great! I didn't do it back then because I didn't have a `Translation*axis4D` solver file to test with.

luminize, 2017-07-18:
@v4hn @mintar Thanks, I'm walking the learning curve here, So apologies for that.

This is what I did so far:
```C++
    case IKP_TranslationXAxisAngle4D:
    case IKP_TranslationYAxisAngle4D:
    case IKP_TranslationZAxisAngle4D:
      // For **TranslationXAxisAngle4D**, **TranslationYAxisAngle4D**, and **TranslationZAxisAngle4D**, the first value
      // represents the angle.
      mult2 = pose_frame.M;

      double rot[1];
      rot[0] = mult2(0, 1);
      ComputeIk(trans, rot, vfree.size() > 0 ? &vfree[0] : NULL, solutions);
      return solutions.GetNumSolutions();
```

I might be making a mistake in calculating the angle above?

So I will first see that I get the IK to work.

Regarding the IK service I try to do the following:
```
parallels@ubuntu:~$ rosservice call /compute_ik
Usage: rosservice call /service [args...]

rosservice: error: Please specify service arguments
parallels@ubuntu:~$ rosservice type /compute_ik 
moveit_msgs/GetPositionIK
parallels@ubuntu:~$ rosservice call /moveit_msgs/GetPositionIK
ERROR: Service [/moveit_msgs/GetPositionIK] is not available.
```
I did some searching on this but I can't figure out how to set this up. Any pointers?

v4hn, 2017-07-18:
> I might be making a mistake in calculating the angle above?

You do not calculate it at all.
`M` is the 3x3 rotation matrix and you have to extract the Roll/Pitch/Yaw angle yourself.
For example the `tf` package provides methods to extract them.

mintar, 2017-07-19:
What it means is that when you ignore the translation, the rotation around the eef can be expressed as a rotation around one of the eef axes wrt. the robot arm base frame.

For example, a [SCARA](https://en.wikipedia.org/wiki/SCARA) robot has 4 degrees of freedom. The endeffector can rotate. The z axis of the robot's base frame and eef always point up, since the eef misses the degrees of freedom to rotate the z axis out of that plane. Since the two z axes are always parallel, the eef pose can always be expressed as translation (x,y,z) and yaw angle wrt. the base frame, and `TranslationZAxisAngle4D` would be appropriate.

It would help if you could provide the URDF of your robot.

You should be able to get the rotation angle like this:

```cpp
    case IKP_TranslationZAxisAngle4D:
        double roll, pitch, yaw;
        pose_frame.M.GetRPY(roll, pitch, yaw);
        ComputeIk(trans, &yaw, vfree.size() > 0 ? &vfree[0] : NULL, solutions);
        return solutions.GetNumSolutions();
```

If that doesn't work, try `pose_frame.M.GetEulerZYZ` or `pose_frame.M.GetEulerZYX`; see [here](http://www.orocos.org/kdl/usermanual/geometric-primitives#toc64).

luminize, 2017-07-19:
@mintar I see your point with the example of the SCARA robot. My URDF is in the working branch "cleaned" of my robot support package:
https://github.com/luminize/matilda_support/tree/cleaned
Here's also a link of the prototype (non ik-fast):
https://www.youtube.com/watch?v=6Mzok9r9k88

luminize, 2017-07-19:
The reason I'm making an IKfast plugin in the first place is that I'd like to use python with moveit_commander to get the robot to a number of positions. The robot is driven by [Machinekit](http://www.machinekit.io), and that has a python API too.

I use ROS/Moveit as the off-line planning alternative to Machinekits G-code interpreter. So for making an application I'll use a python program which will take actions, depending on hardware inputs/outputs, the Machinetalk remote interface and other application specifics (like production automation machinery).

Because my robot is < 4DOF I've not succeeded in using moveit_commander (cartesian) with the "standard" KDL kinematics or trac_ik plugin.

So in the end I'd like to provide x,y and z (and possibly eef angle) and just let Moveit plan and execute the move.
What I guess will be the easiest/proper workflow is:
- use Translation3D kinematics for cartesian position -> joint angles
- manually set the 4th axis (eef) rotation
- plan and execute

mintar, 2017-07-19:
Ok, your robot indeed doesn't have an axis that's parallel with the base frame, so you can't use TranslationZAxisAngle4D. Your approach to use Translation3D and then manually set the eef rotation sounds reasonable. An even better solution would be to hack your generated `*_ikfast_moveit_plugin.cpp` file (in the `case Translation3D:` block), compute the eef rotation there and put it into `solutions[3]` before returning.

luminize, 2017-07-20:
Thanks, i'll have a look at that.

k-okada, 2018-07-10:
I have same topic (running moveit on yaw-yaw-z-yaw SCARA robot) and the discussion here was so helpful. I have to use Translation3D and then manually set the eef rotation, becauses of ikfast issue.  For the record, here is what I confirmed; 

1. The SCARA robot description https://github.com/k-okada/SCORE/tree/master/src/scara_description need small modification to work with ikfast (https://github.com/k-okada/SCORE/commit/f15ed3e4b629ad48568efab016368d4d195f95e5)
2. ik fast with `TranslationZAxisAngle4D` did not work, see https://github.com/rdiankov/openrave/issues/596 for detail.
3. running ikfast with 'Translation3D' and extend code for last yaw axis https://github.com/k-okada/SCORE/commit/549779a137110cfb1d2d2b6a2d2b550851926f7b#diff-25d902c24283ab8cfbac54dfa101ad31, see https://github.com/k-okada/SCORE/commit/2253b14d2761ab9652a780097ea352f26db7252d for the script
4. fix `ikfast_moveit_plugin.cpp` based on  https://github.com/ros-planning/moveit/issues/548#issuecomment-316298918 comment. https://github.com/ros-planning/moveit/pull/973

[image]

I can confirm this works with moveit_commander sample code
```
#!/usr/bin/env python

import sys
import math
import rospy
from moveit_commander import RobotCommander, roscpp_initialize, roscpp_shutdown
from moveit_msgs.msg import RobotState, Constraints
from geometry_msgs.msg import Pose

if __name__ == '__main__':
    roscpp_initialize(sys.argv)
    rospy.init_node('moveit_py_demo', anonymous=True)

    robot = RobotCommander()
    rospy.sleep(1)
    
    m = robot.manipulator

    # move init
    m.set_pose_target([0.5, 0.0, 0.15,  math.pi, 0, 0])
    m.go()
    
    # move y
    m.set_pose_target([0.3,-0.5, 0.15,  math.pi, 0, 0])
    m.go()

    # move z
    m.set_pose_target([0.3,-0.5, 0.05,  math.pi, 0, 0])    
    m.go()
 
    # rotate z
    m.set_pose_target([0.3,-0.5, 0.05,  math.pi, 0, math.pi/2])
    m.go()
    

```

nxdefiant, 2020-01-01:
Hi,

I have the same issue with an [open manipulator](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/specification/#specification) in its default configuration. From the description I would assume to use OpenRaves TranslationYAxisAngle4D since I want to rotate the end effector around the Y (green) axis.

[image]

But I was unable to get it to work with ikfast. Surprisingly TranslationZAxisAngle4D axis does work with the following change:
```
case IKP_TranslationZAxisAngle4D:
	pose_frame.M.GetRPY(roll, pitch, yaw);
	ComputeIk(trans, &pitch, vfree.size() > 0 ? &vfree[0] : nullptr, solutions);
	return solutions.GetNumSolutions();
```

For some reason unknown to me I can use the pitch angle with TranslationZAxisAngle4D and it does work. I can drag the goal state marker around and the arm does follow. Does someone has an idea here?

mintar, 2020-01-02:
I've posted this snippet from the [openrave documentation](http://openrave.org/docs/0.6.6/openravepy/ikfast/) before:

> **TranslationXAxisAngle4D, TranslationYAxisAngle4D, TranslationZAxisAngle4D** - end effector origin reaches desired 3D translation, manipulator direction makes a specific angle with x/y/z-axis (defined in the **manipulator base link’s coordinate system**)

That explains why TranslationYAxisAngle4D doesn't work, but TranslationZAxisAngle4D does: The "YAxis" or "ZAxis" here is the manipulator's **base link** coordinate frame ("world" in your picture), not the end effector frame. Also, it's not a rotation around that axis, but an angle with that axis - in other words, exactly the pitch angle that you want.

It would be great if you could open a pull request with your change!

mintar, 2020-01-02:
The comment here also tries to explain this:

https://github.com/ros-planning/moveit/blob/d8678597530a8cbdb4af5075cd230fdaf1af6aae/moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp#L96-L98

gavanderhoorn, 2020-01-02:
Might be nice to make that comment even more clear. Perhaps even use the text in https://github.com/ros-planning/moveit/issues/548#issuecomment-570187651? Especially the "not the end effector frame" bit.

mintar, 2020-01-02:
@gavanderhoorn : I don't know how to be more explicit than "Axes are defined in the manipulator base link's coordinate system". My issue comment text was a bit too verbose for a code comment, don't you think?

gavanderhoorn, 2020-01-02:
> @gavanderhoorn : I don't know how to be more explicit than "Axes are defined in the manipulator base link's coordinate system". My issue comment text was a bit too verbose for a code comment, don't you think?

No, I don't feel it's too verbose.

It's 4 lines.

mintar, 2020-01-02:
I've added a PR that adds the missing IK types: #1823

nxdefiant, 2020-01-02:
PR confirmed working with open_manipulator, tested with demo.launch. Thanks.

rhaschke, 2020-01-13:
Fixed in #1823.

PR #1823 (2020-01-13): ikfast: implement Translation*AxisAngle4D IK type
Added an implementation for TranslationXAxisAngle4D TranslationYAxisAngle4D and
TranslationZAxisAngle4D. @nxdefiant tested this with his robot arm for the
TranslationZAxisAngle4D IK type (see #548); the rest should be analogous.

Can be cherry-picked onto kinetic.

Fixes #548 .

- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extended the tutorials / documentation, if necessary [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Include a screens [... cut]
closed 2020-01-13 by rhaschke