moveit/moveit #549: Move Group Interface tutorial OpenRAVE error

oliviao00, 2017-07-17:
### Description

When I attempt to scan my catkin workspace (ws_moveit) for missing packages using rosdep install, I receive a warning that the OpenRAVE package doesn't fit naming conventions.  This prevents OpenRAVE from building when performing the tutorial's next step, catkin build.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build
* Version 0.9.8

### Steps to reproduce
I've installed moveit from source and performed all the prerequisites for the Move Group Interface Tutorial without issue.  I've completed the first steps of the tutorial:
`cd ~/ws_moveit/src`
`git clone https://github.com/ros-planning/moveit_tutorials.git`
`git clone https://github.com/PR2/pr2_common.git -b kinetic-devel`
`git clone https://github.com/davetcoleman/pr2_moveit_config.git`

Then I perform the step:
`rosdep install --from-paths . --ignore-src --rosdistro kinetic`

### Expected behaviour
The output should be:
`#All required rosdeps installed successfully` only, without any warnings.

### Actual behaviour
Instead, I receive: 

`WARNING: Non-catkin package name "OpenRAVE" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.`
`WARNING: Non-catkin package name "OpenRAVE" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.`
`WARNING: Non-catkin package name "OpenRAVE" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.`
`#All required rosdeps installed successfully`

Then, in trying to perform catkin build, the OpenRAVE package fails to build.  This is a section of the output:
`Failed     << OpenRAVE:make                                    [ Exited with code 2 ] `
`Failed    <<< OpenRAVE                                         [ 3.0 seconds ]`

### Backtrace or Console output
https://gist.github.com/oliviao00/15bd4f78aae85ed9dd98e2310a98bb19

I am very new to ROS -- all help is greatly appreciated.

closed 2017-07-18 by oliviao00

davetcoleman, 2017-07-21:
Did you figure this out? Its not clear to me why you were trying to build OpenRave in a catkin workspace

oliviao00, 2017-07-24:
Yes, I didn't need to build OpenRave at all in this instance, and didn't see the error of trying to build it in a catkin workspace.