moveit/moveit #552: /attached_collision_object endless loop

zplizzi, 2017-07-18:
### Description

I am trying to attach an object to a robot arm. I send a message over the `/attached_collision_object` topic. As a result, the object begins being published in the `robot_state: attached_collision_objects:` section of the `/move_group/monitored_planning_scene`, with diff = False. The object shows up in Rviz as expected. However, in the console where I launched my robot launch file (with MoveIt, Rviz, etc), I get an endless stream of messages of the type 

`[ INFO] [1500398010.982460498]: Attached object 'test' to link 'fingertip_link'`

It seems to me that Rviz is trying to apply every message from `/move_group/monitored_planning_scene` as a new attached object, vs recognizing that the object already exists. 

As a result (I think?), collision detection during path planning on that object is broken since the object is deleted faster than the planner can plan a trajectory.

### Your environment
* ROS Distro: Kinetic
* OS Version: e.g. Ubuntu 16.04
* Binary: 1.3.1-0xenial-20170615-105244-0800

zplizzi, 2017-07-18:
I think I understand the problem a bit better. When adding a normal collision object (unattached), there will be a single message on `/move_group/monitored_planning_scene` containing the new object, and diff=True. All future messages also have diff=True, and don't mention the object. This appears to work fine, as it should.

However, when adding an attached collision object, the object is published in every future `/move_group/monitored_planning_scene` message, with diff=False in the `robot_state` part of that message. Then, apparently whoever is listening to that message (Rviz plugin?) doesn't handle the diff=False properly, and re-creates the object each message.

So, it seems like there's two approaches to a solution: either a) treat attached objects like the unattached, and only publish them once with diff=True, or b) fix the Rviz plugin to handle diff=False messages properly.

v4hn, 2017-07-18:
First off, thanks for looking into this. 👍 

At first glance I would think publishing the robot state *always* as `is_diff: false` is a bug in this case, but I didn't investigate this thoroughly.
It could be that some other code behaves wrongly if it receives a "diff" RobotState and/or that it is non-trivial to publish the diffs for the RobotState class.

Still, my preferred solution would work towards your proposal "a)".

> As a result (I think?), collision detection during path planning on that object is broken since the object is deleted faster than the planner can plan a trajectory.

The planner operates on a locked planning scene, so this is most likely not the case.

zplizzi, 2017-07-18:
After some more poking around, it doesn't look like the attached_object code (yet) supports doing diffs. The collision objects have the `collision_detection::WorldDiff` class which handles this, but I couldn't find an analog for the attached collision objects (in the Kinetic branch, at least). That being said, this is the first time I've looked at this code, so I easily could have missed something.

v4hn, 2017-07-19:
On Tue, Jul 18, 2017 at 02:46:30PM -0700, Zack Polizzi wrote:
> After some more poking around, it doesn't look like the attached_object code (yet) supports doing diffs.

This is quite likely and basically what I meant by

> or that it is non-trivial to publish the diffs for the RobotState class.

In this case implementing this in the monitor in general is not as easy, but clearly possible.
Anyways, I believe this would be the correct way forward.

A compromise might handle most cases correctly though.
How about an indicator flag that marks the attached objects of the monitored state as "dirty" whenever *something* changes in there.
In this case we could always send `is_diff: true` messages unless someone attached/detached an object.
I guess that would improve the current state quite a bit.

Vonstafenberg, 2017-08-09:
@zplizzi and @v4hn  I have the same issue. I am doing the pr2 moveit_tutorials planning_scene_ros_api_tutorial. The object appears and gets connected to the gripper. But then it does not get detached. and along the way between the processes of attach and detach I get a stream of messages
.............................................................................

```
[ INFO] [1502275537.591629864]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275537.652572290]: Detaching the object from the robot and returning it to the world.
[ERROR] [1502275537.653146469]: Found empty JointState message ``      **<-- also getting this**
```
```
[ERROR] [1502275537.653223449]: The specified RobotState is not marked as is_diff. The request to modify the object 'box' is not supported. Object is ignored.
[ INFO] [1502275537.815513782]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275538.071848487]: Removing world object with the same name as newly attached object: 'box'
[ INFO] [1502275538.071963773]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275539.831450395]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275540.087312698]: Attached object 'box' to link 'r_wrist_roll_link'
[ INFO] [1502275540.343626912]: Attached object 'box' to link 'r_wrist_roll_link'
```

I have checked the same tutorial in indigo. it works inspite of the error 'Object is ignored'.  and also the stream of messages coming. The box disappears.
Any suggestions...?   please...

v4hn, 2017-08-09:
@Vonstafenberg The repeated `Attached object ...`  info messages can be safely ignored and should not
cause any problem.

> But then it does not get detached. [...] I have checked the same tutorial in indigo. it works

If this is the case, please report this as a bug in `moveit_tutorials` with a more detailed explanation.