moveit/moveit #554: moveit_planners_ompl doesn't compile with OMPL 1.3.0

BryceStevenWilley, 2017-07-19:
### Description
moveit_planners_ompl doesn't compile with the latest version of OMPL (1.3.0).

### Your environment
* ROS Distro: Kinetic
* OS Version: XUbuntu 16.04
* Source: e4a238c0f383349fa2de4cc603f6957594535219

### Steps to reproduce
* Make a new workspace with the following git repos (use kinetic-devel branch if present)
    * [catkin_ompl](https://github.com/KavrakiLab/catkin_ompl.git)
    * [geometric_shapes](https://github.com/ros-planning/geometric_shapes.git)
    * [moveit](https://github.com/ros-planning/moveit.git)
    * [moveit_msg](https://github.com/ros-planning/moveit_msgs.git)
    * [moveit_resources](https://github.com/ros-planning/moveit_resources.git)
    * [srdfdom](https://github.com/ros-planning/srdfdom.git)

### Expected behaviour
Moveit should compile. 

### Actual behaviour
Moveit doesn't compile, citing undefined references in OMPL.

### Backtrace or Console output

https://gist.github.com/BryceStevenWilley/795163dbdf6e0abc5f58a2ecd393dae8

### Possible Cause

The [constraints library](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/constraints_library.cpp#L308) tries to make a SpaceStorage instance with a const ref, but as of [OMPL 1.3.0](https://github.com/ompl/ompl/commit/51940fb95ba7d3c6b79b75275ed1205a0a96abd4), most of OMPL was changed to pass by value.

This is something that should be addressed on the OMPL side (we should be returning values, not const ref anyway), but it'd require another OMPL release, so MoveIt might have to work with OMPL 1.3.0. (see #486).

v4hn, 2017-07-19:
@mamoll Any ideas?

mamoll, 2017-07-20:
@v4hn, I understand the problem: due to C++11 updates in OMPL shared_ptr's are now passed by value instead of const&. However, many functions still return const std::shared_ptr<>&. This becomes a problem if the output of one function is the input for another. I can fix this and release a new patched version of OMPL. I don't understand how Moveit seems ready for release when it doesn't even compile. What's the difference between the build bot setup that Moveit uses and @BryceStevenWilley's setup?

v4hn, 2017-07-20:
> I don't understand how Moveit seems ready for release when it doesn't even compile. What's the difference between the build bot setup that Moveit uses and @BryceStevenWilley's setup?

A very good question indeed. I'll pass that on to @130s .
I am confident he ran pre-release tests for the [pull-requested lunar releases](https://github.com/ros/rosdistro/pull/14966).

BryceStevenWilley, 2017-07-20:
I finally determined that even though my setup above was linking against the correct version of OMPL, it was using the an older version of the OMPL header files ( from Kinetic). It was a quick fix (find ompl as a catkin required competent instead of a separate package) to include the correct header files, and I was hoping that they would give more specific locations where the error is occurring, but the library compiles fine in this case.

I'm pretty sure the problem @mamoll is describing above is present though. Is there some sort of internal compiler copying that ignores it in this case?

130s, 2017-07-20:
I'm on the road so can't take a deep look now so just FYIing for now.

> A very good question indeed. I'll pass that on to @130s .
> I am confident he ran pre-release tests for the [pull-requested lunar releases](https://github.com/ros/rosdistro/pull/14966).

Yes for the prerelease tests (for Lunar though) https://github.com/ros-planning/moveit/issues/486#issuecomment-300648476 and they passed.

> OS Version: XUbuntu 16.04
> Source: e4a238c

I honestly don't know how `XUbuntu` is different from Ubuntu but I assume not much?
And the current HEAD isn't tested using ROS buildfarm setting yet, at least on my side. I only run pre-release tests upon making releases.

(At cases like this running pre-release tests per pull request might be beneficial? `industrial_ci` has that feature. Consolidating CI packages is pending https://github.com/ros-industrial/industrial_ci/issues/98).

> I finally determined that even though my setup above was linking against the correct version of OMPL, it was using the an older version of the OMPL header files ( from Kinetic). It was a quick fix [...]

I'm not exactly sure how you fixed, but we'd appreciate if you could open a pull request for that if necessary.

v4hn, 2017-07-21:
@BryceStevenWilley So the linking bug is not an upstream issue and not a blocker for lunar.

Could you please elaborate on 
> I finally determined that even though my setup above was linking against the correct version of OMPL, it was using the an older version of the OMPL header files ( from Kinetic). It was a quick fix [...]

?

Circular include dependencies have been a problem with ompl for a while (see http://moveit.ros.org/install/source/dependencies/) and the proper way to fix this is to have OMPL install its headers into their own include-subfolder, e.g. `/opt/ros/xyz/include/ompl-1.3/ompl/config.h`. Note that the usual include directive for ompl headers starts with `#include <ompl/...`, so an additional folder is needed to make the ompl include independent of other headers in `/opt/ros/xyz/include`.
I do not understand though, how your proposed fix resolves this problem.

BryceStevenWilley, 2017-07-21:
> So the linking bug is not an upstream issue and not a blocker for lunar.

That is correct. Misjudgment on my part, sorry.

> Circular include dependencies have been a problem with ompl for a while (see http://moveit.ros.org/install/source/dependencies/)

Thank you for linking this, it's exactly the problem I ran into. I didn't realize this was a widespread issue.

> I do not understand though, how your proposed fix resolves this problem.

My fix (which is [here](https://github.com/KavrakiLab/moveit/commit/ba2a8ad266200e940fee60b3f2f38c78fb0e3ccc)), isn't a general purpose fix. It only works if you are using source OMPL in the same catkin workspace as moveit, which is fine for our use case. If we find a more general solution, I'll push fixes upstream. 

Going to close this issue, thanks for y'all's help.

closed 2017-07-21 by BryceStevenWilley