moveit/moveit #556: Bazel build ?

abishekh, 2017-07-21:
### Description

Has anyone successfully built moveit with bazel ? I'm trying to do it but the project directory structure is not very conducive.

###  Environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source build

v4hn, 2017-07-21:
Personally I don't see why you would want to do that, but anyway.
Also I'm not aware of people doing this before.

Please keep us updated. If it's not too much hassle for us and requested, we could also merge support for it upstream.

abishekh, 2017-07-21:
Trying to do it to use moveit APIs in another project that builds with bazel.

davetcoleman, 2017-07-21:
I've learned/used Bazel recently and am excited about this idea! It also seems like a big undertaking since you would probably need to Bazel-convert all MoveIt!'s other ROS dependencies, yea?

abishekh, 2017-07-21:
@davetcoleman correct. Good things don't come easy I guess. A hacky way I've see people use - is to link the pre-generated(catkin) ".so" files and then just stand it up as a reference for future projects. The issue with this is that the headers aren't organized well in the directory structure (there isn't one  central "/include" directory). This means either rewriting a lot of code to account for the directory structure or to set bazel rules for each sub-package that has its own structure.