moveit/moveit #560: REQUIRED process [moveit_setup_assistant-1] has died!

wsycy1994, 2017-07-25:
========================================================
REQUIRED process [moveit_setup_assistant-1] has died!
process has died [pid 9833, exit code -11, cmd /opt/ros/indigo/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/cy/.ros/log/9bb404bc-70db-11e7-906b-c4346b72034a/moveit_setup_assistant-1.log].
log file: /home/cy/.ros/log/9bb404bc-70db-11e7-906b-c4346b72034a/moveit_setup_assistant-1*.log
Initiating shutdown!
===============================================================
[moveit_setup_assistant-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Ubuntu14.04
I installed moveit_setup_assistant package use synaptic package manager.
I have already reinstalled ,but it doesn't worked!
please help me !!!!!!  Thanks!!!!!

de-vri-es, 2017-07-25:
Please provide a clearer description of your problem. As it stands we have very little to work with.

* What were you doing or trying to do when the problem occurred?
* Which exact commands did you execute?
* At which stage of using the setup assistant does the crash occur? Is it triggered by a specific action?
* Can you complete the tutorial [1] successfully?
* Could you give more details of your system? Did you install any moveit components or dependencies from source?

[1] http://docs.ros.org/indigo/api/moveit_tutorials/html/index.html

wsycy1994, 2017-07-26:
1.I just executed "roslaunch moveit_setup_assistant setup_assistant" in terminal and the above error occurred.
2.Not triggered by specific actions, but the beginning of the implementation of the system reported the error
3.I use synaptic package maneger to install package,but when I tried to install ros-indigo-moveit-pr2,I got the error like this:Depends: ros-indigo-pr2-moveit-tutorials  but it is not installable.So I can not complete the tutorial.
4.My system is Ubuntu 14.04--indigo
5.A few days ago I was able to run moveit assistamt successfully, but yesterday , the beginning of the implementation of the system reported the error, I do not remember what I did in that time,but I'm sure I did not install anything big.
6.About the depend's problem ,I don't understand how to check.

Thank you so much for answering my question!! If I did not say clearly please tell me

davetcoleman, 2017-07-26:
> Depends: ros-indigo-pr2-moveit-tutorials but it is not installable.So I can not complete the tutorial.

I just fixed this bug and the new version will be released next ROS sync: https://github.com/ros-planning/moveit_pr2/pull/92

zultron, 2017-11-02:
I'm seeing the same error message with Kinetic packages in Jessie.  Debugger shows a segfault.  I'm happy to debug further if someone wants to walk me through it a little.

```
$ roslaunch moveit_setup_assistant setup_assistant.launch debug:=1
... logging to /home/jman/.ros/log/b93c1e44-c012-11e7-b3da-0242ac110002/roslaunch-docker-ros-mk-7578.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/opt/ros/kinetic/share/moveit_setup_assistant/launch/setup_assistant.launch
started roslaunch server http://docker-ros-mk:35568/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    moveit_setup_assistant (moveit_setup_assistant/moveit_setup_assistant)

ROS_MASTER_URI=http://localhost:11311
]2;/opt/ros/kinetic/share/moveit_setup_assistant/launch/setup_assistant.launch http://localhost:11311
core service [/rosout] found
process[moveit_setup_assistant-1]: started with pid [7596]
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kinetic/lib/moveit_setup_assistant/moveit_setup_assistant...(no debugging symbols found)...done.
Starting program: /opt/ros/kinetic/lib/moveit_setup_assistant/moveit_setup_assistant __name:=moveit_setup_assistant __log:=/home/jman/.ros/log/b93c1e44-c012-11e7-b3da-0242ac110002/moveit_setup_assistant-1.log
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]

Program received signal SIGSEGV, Segmentation fault.
0x00007fffd3720cdc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) bt
#0  0x00007fffd3720cdc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007ffff7deabea in ?? () from /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff7deacd3 in ?? () from /lib64/ld-linux-x86-64.so.2
#3  0x00007ffff7ddd16a in ?? () from /lib64/ld-linux-x86-64.so.2
#4  0x0000000000000003 in ?? ()
#5  0x00007fffffffe963 in ?? ()
#6  0x00007fffffffe9a6 in ?? ()
#7  0x00007fffffffe9c5 in ?? ()
#8  0x0000000000000000 in ?? ()
(gdb) info threads
  Id   Target Id         Frame 
* 1    Thread 0x7ffff7f3b840 (LWP 7598) "moveit_setup_as" 0x00007fffd3720cdc in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) 
```

zultron, 2017-11-03:
Here's the simplest way to reproduce I can find, in a Docker container (run it and get a cup of coffee):
```
docker run -it --rm ros:kinetic-robot-jessie \
    bash -c 'apt-get update &&
        apt-get install -y ros-kinetic-moveit-setup-assistant &&
        roslaunch moveit_setup_assistant setup_assistant.launch'
```

v4hn, 2017-11-17:
@zultron Can you reproduce the problem with a source-build of MoveIt too? Specifically with `-DCMAKE_BUILD_TYPE=Debug`?
The backtrace claims you hit some segfault somewhere in qt which is very hard/useless to debug.

abishekh, 2018-02-06:
I'm seeing this message too.