moveit/moveit #562: Orientation Constraint not being constrained

animez4me, 2017-07-25:
Hi,

I am able to plan an orientation constraint for my end-effector, being parallel to the ground. However, occasionally the path planned results in the end-effector to randomly flip. Is there a method to remove these rotating motion and keep the end effector parallel at all times?

I am currently using RRTConnectkConfigDefault planner and trac_ik solver. I have set the tolerances for the orientation constraint as 2pi and the other two axis as 0.3.

Thanks,

v4hn, 2017-07-26:
Please have a look at https://github.com/ros-planning/moveit/pull/541 for an analysis of the problem and a workaround that might or might not help you.
This pull-request has not been released yet.

animez4me, 2017-07-27:
Hi, I have tried the workaround and I definitely see an improvement. The enforce_joint_model_state_space flag prevented the end effector from going out of the orientation constraint. It also finds the solution reasonably quick (able to use in real time). However, occasionally it would fail the planning (timeout set at 60secs). This happened 4/25 times when I tested it. Is this normal?

Thanks,

v4hn, 2017-07-28:
Well, the patch I linked to is a workaround, it does not address the root of the problem.
As explained in the other thread, enforcing the joint model state space implies rejection sampling and this can be quite slow if the planner fails to find the valid part of the search space.
I've not experienced the problem this bad, but yes, planning *might* fail depending on your overall configuration/setup.

I don't know enough about the internals of OMPL to explain why it fails horribly in few cases while it achieves much better performance in most cases. This sounds like full restarts could improve the planning a lot..
Originally I expected a somewhat unimodal distribution for the planning times when asking for the same plan over and over again. Sometimes, this is not the case though, as with the setting you describe.
Maybe @davetcoleman or @mamoll could give some intuition for that behavior?

Things you can consider:
- pragmatic view: multiple plans with short timeouts
- precompute the valid search space: if you use a small set of constraints, generate state databases that satisfy the constraints (see [here](https://github.com/ros-planning/moveit/blob/4bc43c56ca2a5293f3ef8d55796ec586cfde5f69/moveit_planners/ompl/ompl_interface/src/demo_construct_state_database.cpp) for an example generator
- "ideal" solution: fix the `PoseModelStateSpace` implementation so that non-adjacent states in joint space are not adjacent in pose space.

mamoll, 2017-07-28:
I think @zkingston might know why this is happening. It could be related to some weird Euler angle interpolation issue.

MiroslavKohut, 2018-02-13:
I met with the same problem is there any solution for this ? or issue is actual ? Solution of this is very important for my application.

rcywongaa, 2020-05-26:
> A fix for the particular problem would have to restrict the distance of IK solutions for consecutive points.
> This is no trivial task though and involves looking fairly deep into OMPL and its MoveIt interface and would most likely have to change some established interfaces.

As stated by @v4hn [here](https://github.com/ros-planning/moveit/pull/541)

Is there any plan on addressing this issue?  It sems quite important if orientation constraints are to be used meaningfully.  (In my case, `force_joint_model_state_space` caused the planner to take too long).

Thanks for the great work anyway!

v4hn, 2020-05-26:
Plans, sure. But it's not about plans, it's about incentives and work load.
I planned to implement my discussed fix years ago.

We have a [GSoC project](https://github.com/ros-planning/moveit/issues/2092) this year to leverage OMPL-based constraint motion planning. We will see whether that moves us forward on this front.

davetcoleman, 2020-06-02:
We always welcome external contributions (pull requests) for those who need this fixed!