moveit/moveit #564: move_group process has died , exit code -11

AadityaPatanjali, 2017-07-27:
### Description
**move_group process dies with an exit code 11** on a new installation of moveit while runnning the move_group launch file from crdwolfe's [phantomx_arm](http://github.com/crdwolfe/phantomx_arm) package (connecting a physical robotic arm). It tells me that the error is written to a log file, which is not found.
I even tried running move_group from the main moveit_ros_move_group package. It failed with a **Segmentation Fault**.
I have tried running the same code on another computer running ROS Indigo and it works fine.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu Mate 16.04 running on Raspberry Pi 3
* Binary build latest version as of Jul 27, 2017

### Steps to reproduce
To reproduce this, load a robot description file, eg.: phantomx_arm package from crdwolfe (roslaunch phantomx_arm_bringup arm.launch) and run "roslaunch phantomx_arm_moveit_config move_group.launch"
Or, rosrun moveit_ros_move_group move_group

### Expected behaviour
MoveGroup is supposed to run without any errors if the robotic arm is connected otherwise, should give an error that Robot model parameter is not found

### Actual behaviour
move_group process dies with exit code -11 even if the robotic arm is connected
In the case of running only move_group directly from moveit_ros_move_group, there is a Segmentation fault

### Backtrace or Console output
`ubuntu@alphaduck:~$ roslaunch phantomx_arm_moveit_config move_group.launch 

... logging to /home/ubuntu/.ros/log/5ad7f5d0-730f-11e7-8846-b827eb368427/roslaunch-alphaduck-8679.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.2:34509/

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/controller_list: [{'default': True...
 * /move_group/gripper/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTkConfigDefault/range: 0.0
[... 77 lines cut ...]
... shutting down processing monitor complete
done
`

The gdb backtrace of 'rosrun --prefix 'gdb -ex run --args' moveit_ros_move_group move_group ' is
`GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kinetic/lib/moveit_ros_move_group/move_group...(no debugging symbols found)...done.
Starting program: /opt/ros/kinetic/lib/moveit_ros_move_group/move_group 

**Program received signal SIGSEGV, Segmentation fault.
0x76fd9dde in ?? () from /lib/ld-linux-armhf.so.3**
`

v4hn, 2017-07-28:
I can't reproduce the segfault with MoveIt kinetic on x86_64 over here, but that doesn't mean a lot.

I guess the problem is related to the armhf binaries. Could you try to build the current `kinetic-devel` branch in your workspace (is that even feasible on armhf?) and see whether you can reproduce it there?

richard-a-mck, 2017-10-26:
I've gone through the process of installing `kinetic-devel` from source and building on Ubuntu Mate 16 on a Raspberry Pi 3. Unfortunately, it still segfaults on launch of move-group. 
I modified `src/moveit/moveit_ros/move_group/src/move_group.cpp`  to include the following so it would break for GDB;
```
 ROS_INFO("Press enter");
 std::string output;
 getline(std::cin, output);
 ROS_INFO("Enter pressed");
```
I attempted to run the standard moveit demo which loaded my robot params into the param server. I killed this process. I then started the moveit node with `rosrun moveit_ros_move_group move_group`, it then stalled at `getline()` as expected. Running gdb in another ssh'd terminal with `gdb run -p 14181` produced the following result:

```
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
run: No such file or directory.
Attaching to process 14181
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_move_group/lib/moveit_ros_move_group/move_group...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libarmmem.so...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_move_group/lib/libmoveit_move_group_capabilities_base.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libclass_loader.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroslib.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libtinyxml.so.2.6.2...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libtf.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroscpp.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.58.0...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/librosconsole.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/librostime.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.58.0...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libpthread.so.0...(no debugging symbols found)...done.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Reading symbols from /usr/lib/arm-linux-gnueabihf/libconsole_bridge.so.0.2...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libgcc_s.so.1...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libc.so.6...(no debugging symbols found)...done.
Reading symbols from /lib/ld-linux-armhf.so.3...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_trajectory_execution_manager.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_plan_execution.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_state.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_trajectory.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_planning_interface.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libm.so.6...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_robot_model_loader.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_collision_plugin_loader.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_perception/lib/libmoveit_occupancy_map_monitor.so.0.9.9...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_model.so.0.9.9...(no debugging symbols found)...done.
[... cut ...]
Unable to fetch general registers.: No such process.
(gdb) [Thread 0x6feff400 (LWP 14219) exited]
[Thread 0x706ff400 (LWP 14206) exited]
[Thread 0x710ff400 (LWP 14205) exited]
[Thread 0x718ff400 (LWP 14200) exited]
[Thread 0x720ff400 (LWP 14199) exited]
[Thread 0x72aa5400 (LWP 14198) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
bt
No stack.
(gdb) quit
```
Does this provide any useful info for debugging? Let me know if there's any other data I can try and get to assist.

gavanderhoorn, 2017-10-27:
My first recommendation would be to build MoveIt with the `CMAKE_BUILD_TYPE` set to `Debug` (or `RelWithDebInfo`). Your stacktrace as-is is not very informative (notice the `no debugging symbols found` for just about every library and binary).

gavanderhoorn, 2017-10-27:
Additionally: if you start `demo.launch` with the `debug:=true` argument, it should already load `move_group` in GDB. That removes the need for your edit of `move_group.cpp`.

richard-a-mck, 2017-10-29:
Great, thanks for the pointer. I've edited .catkin_tools/proiles/default/config.yaml and changed this line - "-DCMAKE_BUILD_TYPE=RelWithDebInfo". I've also removed my blocking line from move_group.cpp. 
It'll take a few hours to build I think so I'll post again once I have further info.

richard-a-mck, 2017-10-30:
Bizarrely enough, movegroup seems to work now, I can't tell what I've changed. When I attempt to run gdb via debug:=true I get the following error, which I believe is a GDB error specific to Raspberry Pi:
```
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]

Program received signal SIGSEGV, Segmentation fault.
0x76fd9dde in ?? () from /lib/ld-linux-armhf.so.3
(gdb) bt
#0  0x76fd9dde in ?? () from /lib/ld-linux-armhf.so.3
#1  0x76fd9df6 in ?? () from /lib/ld-linux-armhf.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) quit
```
I readded the input request section, launched GDB from another window and found no segfault. 
Rerunning demo.launch, it launches as normal but with a long wait (5-10 seconds) at this point in the process;
```
Listening to '/planning_scene_world' for planning scene world geometry
[Long wait occurs here]
[ WARN] [1509331424.451051076]: It is likely there are too many vertices in collision geometry
```

bmagyar, 2017-10-30:
You could be running out of ram... Just a wild guess.

I'm terrible at remembering things,  so forgive me if the question is
silly: have you seen anyone else getting moveit to work on an armv7?  AFAIK
the RobotState class has some bit magic to allocate random poses, maybe
there is a breakage there?

gavanderhoorn, 2017-10-30:
@bmagyar: that is a good suggestion. `RobotState` does indeed do some funky things with memory. Could very well be that it exploits some x86/amd64 vagueries that make it work there, but fail on ARM (unaligned access fi).

gavanderhoorn, 2017-10-30:
@richard-a-mck: re: gdb crash: the IOCTL failure is afaik indeed a gdb thing. You could try to add `xterm -e` to the `launch-prefix` in `move_group.launch` to see whether that works around it.

As to 

> It is likely there are too many vertices in collision geometry

that's really just a warning, but it's probably a good idea to have proper collision meshes in any case. Do you have `collision` geometry? And is it subsampled, or are you using the same meshes as for the `visual` element?

richard-a-mck, 2017-10-30:
I was originally using the same meshes as visual but I've down sampled them to 150 faces each for collision and now the demo launches as normal with no delays. I'll try modifying the launch prefix and rerunning but the segfault seems to have disappeared at this point. Any other tests I can run to provide more info on your side?

gavanderhoorn, 2017-10-31:
Well without a way to reproduce the crash it's going to be difficult to diagnose it.

Does it still not crash if you rebuild your workspace with `CMAKE_BUILD_TYPE=Release`?

richard-a-mck, 2017-10-31:
Ok, now we're into really interesting territory. I recompiled with `-DCMAKE_BUILD_TYPE=Release` and it segfaults again during the robot model load. Specifically:
```
core service [/rosout] found
process[joint_state_publisher-1]: started with pid [15334]
process[robot_state_publisher-2]: started with pid [15335]
process[move_group-3]: started with pid [15336]
[ INFO] [1509489805.779383521]: Loading robot model 'northcon_boom'...
[ INFO] [1509489806.256443687]: Loading robot model 'northcon_boom'...
[move_group-3] process has died [pid 15336, exit code -11, cmd /home/ncboom/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/ncboom/.ros/log/047c3154-bdc1-11e7-96f0-59cb6320fe74/move_group-3.log].
log file: /home/ncboom/.ros/log/047c3154-bdc1-11e7-96f0-59cb6320fe74/move_group-3*.log

```
This is definitely using the lower res collision models, updated moveit files and so on. The only change is that I recompiled it as Release. I'm recompiling again with Debug set to be doubly sure.

gavanderhoorn, 2017-11-01:
@richard-a-mck wrote:
> Ok, now we're into really interesting territory. 

I think the better word is "annoying", not "interesting".

> I recompiled with `-DCMAKE_BUILD_TYPE=Release` and it segfaults again during the robot model load.

that would seem to point to some code that when optimised on ARM does things it isn't supposed to.

> The only change is that I recompiled it as Release. I'm recompiling again with Debug set to be doubly sure.

I'd try `RelWithDebInfo` and then try to trigger the `SEGFAULT` in a GDB session so we can have *some* idea of where to look.

But it could be that this is going to be difficult, as these type of problems tend to be hard to find.

richard-a-mck, 2017-11-03:
I recompiled with debug enabled as mentioned and it started crashing again.  Here's some output from GDB. I'm not very familiar with GDB so I've googled around a bit and hopefully have run some useful commands. 
```
gdb -p 8377
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 8377
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_move_group/lib/moveit_ros_move_group/move_group...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libarmmem.so...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_move_group/lib/libmoveit_move_group_capabilities_base.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.0.9.9...done.
Reading symbols from /opt/ros/kinetic/lib/libclass_loader.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroslib.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libtinyxml.so.2.6.2...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libtf.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroscpp.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.58.0...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/librosconsole.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/librostime.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.58.0...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libpthread.so.0...(no debugging symbols found)...done.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Reading symbols from /usr/lib/arm-linux-gnueabihf/libconsole_bridge.so.0.2...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libgcc_s.so.1...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libc.so.6...(no debugging symbols found)...done.
Reading symbols from /lib/ld-linux-armhf.so.3...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_trajectory_execution_manager.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_plan_execution.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_state.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_trajectory.so.0.9.9...done.
Reading symbols from /lib/arm-linux-gnueabihf/libm.so.6...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_robot_model_loader.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_collision_plugin_loader.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_perception/lib/libmoveit_occupancy_map_monitor.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_model.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_transforms.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_collision_detection.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_planning_scene.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_profiler.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/geometric_shapes/lib/libgeometric_shapes.so...done.
Reading symbols from /opt/ros/kinetic/lib/libdynamic_reconfigure_config_init_mutex.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libtf_conversions.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libmessage_filters.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libxmlrpcpp.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroscpp_serialization.so...(no debugging symbols found)...done.
[... cut ...]
  4    Thread 0x705ff400 (LWP 8386) "move_group" 0x768f7454 in __libc_do_syscall () from /lib/arm-linux-gnueabihf/libpthread.so.0
  5    Thread 0x6fbff400 (LWP 8391) "move_group" 0x768f7454 in __libc_do_syscall () from /lib/arm-linux-gnueabihf/libpthread.so.0
  6    Thread 0x6f3ff400 (LWP 8392) "move_group" 0x768f7454 in __libc_do_syscall () from /lib/arm-linux-gnueabihf/libpthread.so.0
  7    Thread 0x6e9ff400 (LWP 8405) "move_group" 0x768f7454 in __libc_do_syscall () from /lib/arm-linux-gnueabihf/libpthread.so.0
(gdb) 

```
As mentioned, I can't run the standard debug:=true at launch due to issues with GDB on RPi.

richard-a-mck, 2017-11-03:
I should mention, the preceding comment was created by running the move_group node from the devel folder of my ws_moveit workspace.
Here's the GDB output from running the full demo:
```
 gdb -p 8896
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 8896
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_move_group/lib/moveit_ros_move_group/move_group...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libarmmem.so...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_move_group/lib/libmoveit_move_group_capabilities_base.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.0.9.9...done.
Reading symbols from /opt/ros/kinetic/lib/libclass_loader.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroslib.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libtinyxml.so.2.6.2...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libtf.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroscpp.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.58.0...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/librosconsole.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/librostime.so...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.58.0...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libpthread.so.0...(no debugging symbols found)...done.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Reading symbols from /usr/lib/arm-linux-gnueabihf/libconsole_bridge.so.0.2...(no debugging symbols found)...done.
Reading symbols from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libgcc_s.so.1...(no debugging symbols found)...done.
Reading symbols from /lib/arm-linux-gnueabihf/libc.so.6...(no debugging symbols found)...done.
Reading symbols from /lib/ld-linux-armhf.so.3...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_trajectory_execution_manager.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_plan_execution.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_state.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_trajectory.so.0.9.9...done.
Reading symbols from /lib/arm-linux-gnueabihf/libm.so.6...(no debugging symbols found)...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_robot_model_loader.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_planning/lib/libmoveit_collision_plugin_loader.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_ros_perception/lib/libmoveit_occupancy_map_monitor.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_robot_model.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_transforms.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_collision_detection.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_planning_scene.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/moveit_core/lib/libmoveit_profiler.so.0.9.9...done.
Reading symbols from /home/ncboom/ws_moveit/devel/.private/geometric_shapes/lib/libgeometric_shapes.so...done.
Reading symbols from /opt/ros/kinetic/lib/libdynamic_reconfigure_config_init_mutex.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libtf_conversions.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libmessage_filters.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libxmlrpcpp.so...(no debugging symbols found)...done.
Reading symbols from /opt/ros/kinetic/lib/libroscpp_serialization.so...(no debugging symbols found)...done.
[... cut ...]
156	    for (std::size_t i = 0; i < capabilities_.size(); ++i)
157	      ss << "*     - " << capabilities_[i]->getName() << std::endl;
158	    ss << "********************************************************" << std::endl;
159	    ROS_INFO_STREAM(ss.str());
160	  }
161	
162	  ros::NodeHandle node_handle_;
163	  MoveGroupContextPtr context_;
164	  std::shared_ptr<pluginlib::ClassLoader<MoveGroupCapability> > capability_plugin_loader_;
165	  std::vector<MoveGroupCapabilityPtr> capabilities_;
(gdb) 
```

v4hn, 2017-11-17:
Looking at this I would say it's more likely that you hit a compiler bug for your architecture
than that MoveIt does something architecture-specific that fails on arm... (Although the latter is not impossible of course)

You might want to try to compile with llvm to see whether it's a gcc bug.

jschleicher, 2019-03-15:
I had experienced a similar issue with ubuntu 16.04 on a rapberry pi. The arm packages provided in 18.04 work fine though. So I'd suggest to close this as "compiler bug" as v4hn suggested.

Maybe it could be related to [gcc bug 77728](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77728)?

v4hn, 2019-03-29:
Thanks for your comment @jschleicher .

closed 2019-03-29 by v4hn