moveit/moveit #567: 201708 Release

130s, 2017-07-31:
- Lunar needs new release to complete its release https://github.com/ros-planning/moveit/issues/486#issuecomment-319132241
- If there are any issues/PRs that should be included, please let us know in this ticket by 4th Fri.

v4hn, 2017-08-03:
I would very much like to see the segfault addressed by #572 & #565 fixed and #553 merged before the next release.

The earlier is a rather severe bug, the latter a easily understandable edge-case that is waiting for someone to click "Merge" for the last two weeks. :)

130s, 2017-08-05:
Maintainers: please hold new merge. 

TODOs from @v4hn are all clear now. I'm running pre-release tests locally for IKL.

130s, 2017-08-07:
- [x] Indigo -> https://github.com/ros/rosdistro/pull/15712
- [x] Kinetic -> https://github.com/ros/rosdistro/pull/15715
- [x] Lunar -> https://github.com/ros/rosdistro/pull/15716

closed 2017-08-07 by 130s

gavanderhoorn, 2017-08-07:
Thanks for all the work @130s 👍.

Btw: is there something special about the Indigo changelogs? Fi the changes to `moveit_setup_assistant` are listed in the `moveit`, `moveit_ros` (with a `[moveit_core][moveit_planners]` prefix?) and finally `moveit_setup_assistant`. I don't know for certain, but it could be that this happened to more packages.

I've seen your PR on `catkin_pkg`. Are the changelogs as they are now a result of that work?

130s, 2017-08-07:
Thanks for pointing that out. I overlooked those wrong changelogs in indigo. In the next release I'll fix (I also fixed this time the wrong log from 2 versions earlier).

Indeed that was a result of a not-accepted PR https://github.com/ros-infrastructure/catkin_pkg/pull/188 that proved to be insufficient like this time.