moveit/moveit #568: OMPL with LBKPIECE1 crashes frequently after planning request

christian-rauch, 2017-08-02:
### Description

MoveIt crashes sporadically during planning using OMPL with LBKPIECE1.
I am using the `move_group.launch` launch file from the ([moveit plugin](https://github.com/christian-rauch/kuka_lwr_sdh2_moveit)) generated by the setup assistant to start the `move_group`. I am then using the rviz plugin to set the end-effector poses and plan the joint trajectory with OMPL.

The crash also occurs when just using the `demo.launch` with the default planner (LBKPIECE1).
A workaround is currently to use e.g. LBKPIECEkConfigDefault.

### Your environment
* Ubuntu 14.04, indigo
* binary, build: 0.7.11

### Steps to reproduce
I can reproduce the same backtrace by launching the `demo.launch` and leaving the planner unspecified. The segfault occurs after hitting `Plan` a couple of times (less than 20) without moving the end-effector.

### Expected behaviour
valid joint trajectory

### Actual behaviour
SIGSEGV

### Backtrace or Console output

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa37fe700 (LWP 9865)]
0x00007fffc9f82c10 in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors(std::vector<int, std::allocator<int> >&, std::vector<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >&) const
    () from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10

#0  0x00007fffc9f82c10 in ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors(std::vector<int, std::allocator<int> >&, std::vector<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::Grid<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >&) const () from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10
#1  0x00007fffc9f82e60 in ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::neighbors(std::vector<int, std::allocator<int> >&, std::vector<ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >&) const () from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10
#2  0x00007fffc9f82fd5 in ompl::GridB<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*, ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::OrderCellsByImportance, ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::OrderCellsByImportance>::createCell(std::vector<int, std::allocator<int> > const&, std::vector<ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*, std::allocator<ompl::GridN<ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::CellData*>::Cell*> >*) ()
   from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10
#3  0x00007fffc9f8512d in ompl::geometric::Discretization<ompl::geometric::LBKPIECE1::Motion>::addMotion(ompl::geometric::LBKPIECE1::Motion*, std::vector<int, std::allocator<int> > const&, double) ()
   from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10
#4  0x00007fffc9f7ffc3 in ompl::geometric::LBKPIECE1::solve(ompl::base::PlannerTerminationCondition const&) ()
   from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10
#5  0x00007fffca02a054 in ompl::tools::ParallelPlan::solveMore(ompl::base::Planner*, unsigned long, unsigned long, ompl::base::PlannerTerminationCondition const*) () from /opt/ros/indigo/lib/x86_64-linux-gnu/libompl.so.10
#6  0x00007ffff28c3a4a in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#7  0x00007ffff627b184 in start_thread (arg=0x7fffa37fe700)
    at pthread_create.c:312
#8  0x00007ffff5883ffd in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

v4hn, 2017-08-03:
Thanks for reporting this (again) @christian-rauch .
It's hair-raising this bug apparently still exists...
I can't reproduce it with our UR5 setup though, although the resulting trajectories look horrible with this planner configuration and a different planner should be selected anyway.

In ROS kinetic we moved away from "unspecified" and always set a default planner from within MoveIt,
so the problem should not exist for (almost all?) users there.

My understanding is that you have to specify an explicit planner configuration for *each* `MoveGroup` C++ class you use to avoid this problem, is this right?

christian-rauch, 2017-08-03:
I am specifying the default planner within my launch file:
```
<launch>
  <param name="/move_group/arm/default_planner_config" type="str" value="LBKPIECEkConfigDefault" />
  <include file="$(find kuka_lwr_sdh2_moveit)/launch/move_group.launch" />
</launch>
```
This is just a workaround and does not fix the crashes in the default (unspecified) planner `LBKPIECE1`. You can achieve the same effect by setting `default_planner_config: LBKPIECEkConfigDefault` in your `ompl_planning.yaml`, e.g.:
```
arm:
  default_planner_config: LBKPIECEkConfigDefault
  planner_configs:
    - SBLkConfigDefault
    - ESTkConfigDefault
    - LBKPIECEkConfigDefault
    - BKPIECEkConfigDefault
    - KPIECEkConfigDefault
    - RRTkConfigDefault
    - RRTConnectkConfigDefault
    - RRTstarkConfigDefault
    - TRRTkConfigDefault
    - PRMkConfigDefault
    - PRMstarkConfigDefault
```

I guess many users start with moveit by using the setup assistant (which is a great tool by the way). At least in indigo, it does not set `default_planner_config`. Setting this to some reasonable planner (at least not known to crash) would save some people some debugging time :-)

v4hn, 2017-08-03:
On Thu, Aug 03, 2017 at 01:09:11PM -0700, Christian Rauch wrote:
> I am specifying the default planner within my launch file:
> You can achieve the same effect by setting `default_planner_config: LBKPIECEkConfigDefault` in your `ompl_planning.yaml`, e.g.:

Usually the idea was to put it in `config/ompl_planning.yaml`. But sure, you can set it in the launch file too.

Are you sure this decides the default planner when planning through the `MoveGroup` C++ interface in `indigo-devel`?
I think I remember a problem there. If it doesn't, then I would like to fix this soon! (That's a somewhat independent issue though...)

> This is just a workaround and does not fix the crashes in the default (unspecified) planner `LBKPIECE1`.

Yes. I know. Up to now everyone circumvented this by specifying a different (and most of the cases also more useful) planner instead of the default configuration in indigo.
MoveIt! as well as OMPL would be happy if someone could debug the `LBKPIECE1` segfault, but the default changed in kinetic, the generated trajectories of the old default are far from optimal in many cases, it's easy to work around it, and last but not least the code involved is rather complex.
So everyone refrained from debugging this...

> At least in indigo, it does not set `default_planner_config`. Setting this to some reasonable planner (at least not known to crash) would save some people some debugging time :-)

That is quite a good idea actually. :)

christian-rauch, 2017-08-04:
> Are you sure this decides the default planner when planning through the `MoveGroup` C++ interface in `indigo-devel`?

I am not using the C++ interface directly. I only tested with the `move_group` node in the `moveit_ros_move_group` package (binary package 0.7.11).
If I call `rosparam set /move_group/arm/default_planner_config LBKPIECEkConfigDefault` before starting the move_group, the rviz plugin will show me this planner and you see console output from this planner.

v4hn, 2017-08-04:
Ah, in this case please ignore my comment for the moment. :)

Anyway, I hope I gave enough background to explain why the LBKPIECE1 segfault is still there, although it is very annoying to beginners...

Things to do:

- [ ] add a `default_planner_config` selection list option for each planning group to the setup assistant

- [ ] fix the segfault.

Contributions to address either are very welcome!