moveit/moveit #570: computeCartesianPath minimum steps with joint space jump testing

hartmanndennis, 2017-08-03:
https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/robot_state/src/robot_state.cpp#L1923
`unsigned int steps = (test_joint_space_jump ? 5 : 1) + (unsigned int)floor(distance / max_step);`

Why are the minimum steps between waypoints fixed at 5 when joint space jumps detection is active?
If you have a dense trajectory, your output trajectory is 5 times as big and can slow the time parametrization a lot.

Can the tertiary operator be removed?

v4hn, 2017-08-03:
I am not sure where exactly the 5 comes from, but 1 is not enough if you want to use joint space jump detection.
The detection compares distances between consecutive states with the *average* distance.
This doesn't make sense if you only have 1 step to begin with.

Do you use the multi-waypoint interface and encounter this problem?
As a fix for this use-case we could test for jumps in this function and disable jump detection in the inner (single-waypoint) call to `computeCartesianPath`.
Somebody has to implement this though :)

hartmanndennis, 2017-08-03:
Yes, I use the cartesian path service via the move group interface.
I'll see if I can make a PR with your proposed fix.