moveit/moveit #571: Interest in Pull request for generating valid solutions outside -PI, PI range in ikfast plugin?

hartmanndennis, 2017-08-03:
In the discussion of #537 this issue was partially addressed.
I've worked with the KUKA KR16 whose axes 4 and 6 have (software) joint limits of -350° and 350°. IKfast only returns solution in the range -PI to PI. This is a problem because the robot driver considers PI and -PI to be different things. If you are at PI and command position -PI, the joint will rotate 360° before stopping. To be able to use the whole joint range and therefore be more flexible in executing tasks, I made a custom extension to the IKfast plugin which generates solutions for axis 4 and 6 outside the -PI and PI range.
Is there interest in a pull request of a general solution, which uses the joint limits of the robot to decide if extra solutions should be generated?
I don't know what the best solution for continuous joints would be.

v4hn, 2017-08-03:
> Is there interest in a pull request of a general solution, which uses the joint limits of the robot to decide if extra solutions should be generated?

I believe it would be a great addition.
However, for robots like the UR family where each joint is [-2pi;+2pi] this can blow up the solution space quite a lot, so I would propose to leave it optional.

Continuous joints should always be treated as [-pi;pi] and unrolled as required by the user/trajectory processing tools.

davetcoleman, 2017-08-15:
+1 to setting the range based on the URDF joint limits!

> However, for robots like the UR family where each joint is [-2pi;+2pi] this can blow up the solution space quite a lot, so I would propose to leave it optional.

I usually see UR robots with the joint_limited tag enabled, to avoid this issue.

v4hn, 2017-08-15:
On Tue, Aug 15, 2017 at 07:31:06AM -0700, Dave Coleman wrote:
> I usually see UR robots with the joint_limited tag enabled, to avoid this issue.

There is more than one issue there, but yes. :)

hartmanndennis, 2017-08-18:
I was thinking about how to avoid the exploding solution space issue. For the case where we want to get the closest solution to a seed that is not in collision, we don't have to compute every extra possibility.
First, rotating a joint 360° doesn't change collision status so we can ignore solutions that are in collisions.
Since our distance metric is the sum of absolute differences we also know that a solution with distance < PI can't get better by rotating joints. If the best solution has distance > PI we only have to compute solutions for joints with indidvidual distance > PI. This should reduce the solution space a lot.
I'll start working on a pull request.

hartmanndennis, 2017-08-21:
Unfortunately not all assumptions above are true in every case.
I'm currently testing a solution where for every IKFast solution I check every joint if its distance to the seed is greater than 180°. If that is the case I add/subtract(depending of the sign of the distance) 360° and check if the result is still in joint limits. This new solution is always closer to the seed. I don't need to generate the cartesian product because I'm only interested in the best solutions. In this way the number of solutions doesn't change.
I currently only implemented it for the no free parameter case of searchPositionIK because it's simple to implement there. For the other cases it would probably be best to add a function that works on a `IkSolutionList<IkReal> ` and call it where it is needed.
There is the question what to do about the multi-solution version of getPositionIK because it implies that it returns every valid solution.

v4hn, 2017-08-21:
> There is the question what to do about the multi-solution version of getPositionIK because it implies that it returns every valid solution.

Yes. Especially w.r.t. 7dof chains this can be a lot...
I would like to see this feature in MoveIt but most likely it should be optional.
If we want to break ABI for this, we could also add it to the `KinematicsQueryOptions`.